espressif/esp-idf #1285: Potentially confusing change of dhcps default behaviour in cf7ae81

Eroc33, 2017-11-18:
Commit cf7ae81 changed dhcps to allow setting a custom dns server, however this seems to have changed the default behaviour for assigning dns ips over dhcp.
Previously the default behaviour was to use the ip of the device itself as the dns server, whereas now the default is to use the ip 0.0.0.0.
I suspect this may have broken captive portal code for many people (particularly if they are using the pattern common in tutorials/examples for arduino-esp32).
While the fix to user code would be relatively minor it seems like changing the default here in the idf is probably as simple as changing https://github.com/espressif/esp-idf/blob/master/components/lwip/apps/dhcpserver.c#L97 to be `static dhcps_offer_t dhcps_dns = 0x00;`

liuzfesp, 2017-11-26:
Hi @Eroc33 we will check it and make sure the default behavior is not changed @zhangyanjiaoesp

zhangyanjiaoesp, 2017-11-28:
@Eroc33  Thanks for your good suggestion. 
I have modify the code ,after it is merged ,I will post the commit ID

closed 2017-12-01 by igrr