espressif/esp-idf #1287: Task Watchdog is triggered after WIFI reconnect

akomakom, 2017-11-18:
I'm now able to consistently reproduce the following intermittent (as in how long it takes to happen) scenario:

> I (195205) wifi: bcn_timout,ap_probe_send_start
> I (197715) wifi: ap_probe_send over, resett wifi status to disassoc
> I (197715) wifi: state: run -> init (1)
> I (197715) wifi: pm stop, total sleep time: 0/192525694                                                                  
>                                                              
> I (197715) wifi: n:11 0, o:11 0, ap:255 255, sta:11 0, prof:1
> I (197725) network: WIFI Disconnect, reconnecting                                            
> I (197725) network: WIFI ReConnected
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:     
> CPU 0: wifi
> CPU 1: IDLE                                                   
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)                   
> Tasks currently running:             
> CPU 0: wifi                                                                                                              
> CPU 1: IDLE  

(This continues forever).  

Code (https://github.com/akomakom/esp32-deathwatch/blob/master/main/network.c) looks like this:
```

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        ESP_LOGI(TAG, "Starting WIFI connect");
    	ESP_ERROR_CHECK(esp_wifi_connect());
    	ESP_LOGI(TAG, "WIFI Connected");
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
        ESP_LOGI(TAG, "IP: "IPSTR"/"IPSTR"/"IPSTR,
            IP2STR(&event->event_info.got_ip.ip_info.ip),
            IP2STR(&event->event_info.got_ip.ip_info.netmask),
            IP2STR(&event->event_info.got_ip.ip_info.gw)
        );

        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        /* This is a workaround as ESP32 WiFi libs don't currently
           auto-reassociate. */
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        ESP_LOGI(TAG, "WIFI Disconnect, reconnecting");
        ESP_ERROR_CHECK(esp_wifi_connect());
        ESP_LOGI(TAG, "WIFI ReConnected");
        break;
    default:
        break;
    }
    return ESP_OK;
}
```
It doesn't look as if SYSTEM_EVENT_STA_GOT_IP occurs after reconnect.

Is this an issue internal to wlip or am I using things incorrectly?  I've also tried to stop/start wifi in the DISCONNECTED handler, same result.  

What I don't understand is:
 * What is causing the network disconnects
 * What task is not resetting the watchdog (apparently none of my tasks)

The issue usually arises fairly quickly (although the program is making HTTPS requests and serving HTTP requests so it is under some load - one request/response every 10 seconds).

akomakom, 2017-11-19:
Sometimes I get a stacktrace as well:

> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)                                         
> Tasks currently running:                      
> CPU 0: wifi                             
> CPU 1: IDLE                                                                   
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)                                                          
> Tasks currently running:                            
> CPU 0: wifi                                                                             
> CPU 1: IDLE                                                                               
> /home/akom/esp/esp-idf/components/freertos/./queue.c:719 (xQueueGenericSend)- assert failed!
> abort() was called at PC 0x400848d8 on core 1                                                
> 0x400848d8: xQueueGenericSend at /home/akom/esp/esp-idf/components/freertos/./queue.c:2037
>                                                                                           
>                                                                                          
> Backtrace: 0x40087cf8:0x3ffbe210 0x40087df7:0x3ffbe230 0x400848d8:0x3ffbe250 0x400fb3b6:0x3ffbe290 0x401070d1:0x3ffbe2b0 0x401011a2:0x3ffbe2d0 0x4010167f:0x3ffbe2f0 0x4010230b:0x3ffbe310 0x40102485:0x3ffbe330 0x
> 400fb878:0x3ffbe350
> 0x40087cf8: invoke_abort at /home/akom/esp/esp-idf/components/esp32/./panic.c:572
> 
> 0x40087df7: abort at /home/akom/esp/esp-idf/components/esp32/./panic.c:572
> 
> 0x400848d8: xQueueGenericSend at /home/akom/esp/esp-idf/components/freertos/./queue.c:2037
> 
> 0x400fb3b6: sys_sem_signal at /home/akom/esp/esp-idf/components/lwip/port/freertos/sys_arch.c:548
> 
> 0x401070d1: err_tcp at /home/akom/esp/esp-idf/components/lwip/api/api_msg.c:1881 (discriminator 6)
> 
> 0x401011a2: tcp_slowtmr at /home/akom/esp/esp-idf/components/lwip/core/tcp.c:1638
> 
> 0x4010167f: tcp_tmr at /home/akom/esp/esp-idf/components/lwip/core/tcp.c:1638
> 
> 0x4010230b: tcpip_tcp_timer at /home/akom/esp/esp-idf/components/lwip/core/timers.c:428 (discriminator 1)
> 
> 0x40102485: sys_timeouts_mbox_fetch at /home/akom/esp/esp-idf/components/lwip/core/timers.c:575
> 
> 0x400fb878: tcpip_thread at /home/akom/esp/esp-idf/components/lwip/api/tcpip.c:474
>

akomakom, 2017-11-30:
In my testing I was able to reduce (but not eliminate) this issue by:
1) Switching to single core (Run FreeRTOS only on first core)
2) Dropping cpu frequency to 80 (although I doubt that this had much impact)

Now, instead of minutes the board runs the example code for up to a day or so, but then still crashes the same way:

> 
> Task watchdog got triggered. The following tasks did not reset the watchdog in time:
>  - IDLE (CPU 0)
> Tasks currently running:
> CPU 0: wifi

(loops forever and does nothing else, suggesting that the wifi task is hung)

While the code worked I noticed that during the SSL connection I often saw warnings that task watchdog took too long (set to default 5 seconds) for the https_get_task.   My current workaround is to increase this timeout just in case and enable panic handler, at least the board will restart and keep working.

zhang-jun-hao, 2017-12-14:
Hi @akomakom,

I am sorry to tell you that have tested and have not reproduced this problem.

Here is the log on weak or unreliable WIFI signal:
[uart.log](https://github.com/espressif/esp-idf/files/1558446/uart.log)

akomakom, 2017-12-14:
Sorry, I meant to update this, I bought a different brand of ESP32 (HiLetgo) a couple of days ago and it did not experience the same issues, so the problem was a defective unit.  The Geekworm device was defective.

closed 2017-12-14 by akomakom