espressif/esp-idf #1288: feature: return old function from esp_log_set_vprintf()

nkolban, 2017-11-19:
We can change the function used to perform logging by calling esp_log_set_vprintf().  However what if I want to "set it back to what it was?" ... would it be possible to change this function to return a pointer to the existing function that is in place?  This shouldn't affect existing code as today the function returns void and if it now returns a vprintf_like_t* that shouldn't affect existing apps.

negativekelvin, 2017-11-19:
https://github.com/espressif/esp-idf/pull/1286

nkolban, 2017-11-19:
It appears that this issue may have been implemented by @phonec and submitted as a pull request as referenced by mr @negativekelvin above.  Let's use this Github issue as a tracker for either that Pull Request or for a place holder should the pull request not be merged.

phonec, 2017-11-20:
[@iggr](https://github.com/igrr) also suggested to leave everything in one function esp_log_set_vprintf and make it thread safe to exclude wrong behavior when simultaneously calling from different components
[#1286](https://github.com/espressif/esp-idf/pull/1286)

closed 2017-11-24 by nkolban