espressif/esp-idf #1289: Fatfs f_wirte doesnt work in the newest update IDF 3.0

abcdcadc, 2017-11-20:
I used f_write to write data to SD card (fatfs library) and it works in IDF 2.0, but in IDF 3.0 it writes nothing to sd card (even it returns FR_OK) (I open card using reader and see the content of file still is the same)
p/S: I tried to check ff_conf.h and notice that if FS_LOCK = 0 as in the newest IDF, it will returns FR_OK and nothing, but if I enable FS_LOCK to 10 or 20 , it will return error 16 (locked file)

It seems that the newest IDF 3.0 has a problem with file handling.

vanphuc44, 2017-11-20:
why don't you try stdio.h (newlib). I am aware that it also has fwrite

igrr, 2017-12-08:
In the latest version of IDF, FS_LOCK option can be set in menuconfig, see http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#fatfs-fs-lock.

If you enable this option and get an error `FR_LOCKED` (16), this means that you are trying to perform some operation on an open file, like renaming it, deleting, copying, etc.

If you enable this option and get an error `FR_TOO_MANY_OPEN_FILES` (18), this means that the number of files you have tried to open is larger than the value of FS_LOCK option. You need to either open less files or increase the value of FS_LOCK.

If your situation doesn't fit these two cases, please attach some code which reproduces the issue you see.

projectgus, 2020-07-14:
Closing due to lack of update

closed 2020-07-14 by projectgus