espressif/esp-idf #1291: Bluetooth Classic (BR/ER) and Wifi cannot be used together

mringwal, 2017-11-20:
According to posts on the esp32 forum and an engineer I've talked to, any Bluetooth Classic connection, e.g. for A2DP audio or just SPP, causes high packet loss over Wifi up to the point where TCP/Wifi connection breaks - even if there is no real data on the Bluetooth connection.

Getting Bluetooth Classic and Wifi to work together requires the Bluetooth and Wifi schedulers to cooperate. From the outside, it looks like this is either not implemented or not working correctly.

This is a real 'no-go' for commercial Bluetooth & Wifi products (e.g. an audio player that supports A2DP and AirPlay) and an alternative architecture e.g. using a Bluetooth Dual-Mode/Wifi chipset is  needed.

Any idea if/when this will be addressed? Thanks!

TianHao-Espressif, 2018-01-05:
Hi, it's in working. In internal test, BR/EDR and WIFI can work together with  A2DP and WIFI TCP. We will release the coexistence version ASAP.

mringwal, 2018-01-05:
Thanks! Sounds great. Looking forward to the update(s).

pavestru, 2018-01-18:
Hi @TianHao-Espressif, fyi, I was testing today both **v3.1-dev** branch and **v3.0-rc1** that should contain [WiFi/Bluetooth coexistence fixes](https://github.com/espressif/esp-idf/releases/tag/v3.0-rc1). I still encounter the issues mentioned by @mringwal. I suppose, these fixes are a work in progress...? Thank you.

ifitshow, 2018-01-23:
@TianHao-Espressif 18天过去了，很是期待贵司推出WIFI与经典蓝牙共存的IDF。

TianHao-Espressif, 2018-02-27:
@pavestru , sorry, as the previous plan, the coexistence will release in V3.1, but there's a little problem and meet Chinese Spring Festival. We will release a version ASAP(If quickly, maybe in 1~2week).

pavestru, 2018-03-16:
Thank you, @TianHao-Espressif for the update ... I hope the development goes well... looking forward.

TianHao-Espressif, 2018-04-03:
@pavestru ， bluetooth & wifi coexistence can work in latest master.

ifitshow, 2018-04-03:
@TianHao-Espressif 已更新到最新，但测试发现还是一样：

ifitshow, 2018-04-03:
@TianHao-Espressif 已更新到最新，但测试发现还是一样：
1. iOS 可以工作，但wifi会断开，如果开启 Software controls WIFI/Bluetooth coexistence，则WIFI不会断，但蓝牙丢包严重；
2. Android 则不管怎样，开始播放后芯片就重启。

TianHao-Espressif, 2018-04-03:
@ifitshow, you should enable "Software controls WIFI/Bluetooth coexistence" and it is enabled default.
If you want play A2DP audio, you should choose "prefer to bluetooth".

And could you provide the reboot log?

TianHao-Espressif, 2018-04-03:
@ifitshow , if use both wifi and bluetooth as coexist, the performance is not 1+1=2 . The competition will cause wifi and bluetooth cannot reach the performance which run by standalong.

Besides, if run both wifi and bluetooth ,the memory may be lacked. You should check the internal memory use map and better to use psram.

In internal test, if you choose "prefer to bluetooth", it can run A2DP to play music smoothly and wifi run tcp transmit simultaneously.

ifitshow, 2018-04-03:
@TianHao-Espressif 十分感谢，上午你发消息时我确认也经是最新的，我说怎么找不到"prefer to bluetooth"相关选项，esp_coexist.h 里也没有相关定义。刚刚再次 git pull 又有不少更新终于在 esp_coexist.h 里相关定义了，同时make menuconfig也有配置了。make monitor flash ...

ifitshow, 2018-04-03:
一切正常了，android也能正常播放了，不过丢包还是比较严重的，声音时不时断一下。还好我的板上已经设计了 8Mbytes 的PSRAM。再次感谢您！@TianHao-Espressif

TianHao-Espressif, 2018-04-03:
@ifitshow, thank you very much for feedback. We will continue to optimize the coexistence parameter to improve performance.

ifitshow, 2018-04-04:
@TianHao-Espressif 经过昨晚一晚的测试，发现比较遗憾的是：WIFI+A2DP 时音频过一会会断一下（已使用 prefer to BT），即使这时WIFI没任何事务（包括断开WIFI连接也不行），个人认为RF时分操作造成A2DP数据流断流。同时外加WIFI数据时，WIFI质量受影响也是很严重（即使使用 prefer to WIFI）。综合而言 1+1<1，期待贵司团队早日更新

TianHao-Espressif, 2018-04-04:
@ifitshow , what's the meaning of "过一会会断一下"? Is it means that A2DP link disconnect or music cannot play smoothly? If A2DP disconnect while wifi do nothing (event have no connection), it's may not coexist's problem. 

For now, wifi's performance with coexistence enable, it's really not good as wifi standalone. When SW_COEXIST_ENABLE is y, the RX AMPDU is disabled and TX AMPDU buffer number is 2 as default, this may effect wifi's performance. But we do this to keep bluetooth's performance.

 In our internal test, Wifi's RX throughput is about 1~5Mbps when bluetooth is running A2DP smoothly.

Coexistence performance will update continuously。

ifitshow, 2018-04-04:
@TianHao-Espressif 原谅我描述不清！A2DP并不会断开连接，在音频上表现就是隔一小会声音中断一下（正常应该是连续不断的），好比距离太远出现断续一样（只是很轻微而以）。（代码运行时不初始化WIFI则不会有这种现象）

TianHao-Espressif, 2018-05-17:
@ifitshow , the coexist is updated for make A2DP fluent even PREFER TO BALANCE after the commit ID 139a8a2745375f164ba45850c1886762835dc4b0. The performance is obviously better than previous version.

ifitshow, 2018-05-22:
@TianHao-Espressif，对你们的努力表示感谢，我更新后遇到了新问题，不管是PREFER TO BALANCE，还是PREFER TO BLUETOOTH，都会造成蓝牙很难连接，即使连接上也会在几秒后自行断开（不开WIFI则是正常的）。功耗到是降低了很多。不清楚是不是我哪里没有配置好？

TianHao-Espressif, 2018-06-04:
@ifitshow , please provide your commit ID of ESP-IDF. In internal test, the bluetooth connection is not hard to connect. I need to check if there's other commit effect it or not.

UaHeadHunter, 2018-06-14:
With latest master WiFi and BLE can work until WiFi will get connected to the AP.
After that BLE is not advertising and nothing can connect to BLE.

Sorry, if it's a wrong issue.

P.S.: params with coexisting WiFi and Bluetooth do nothing in this case.

TianHao-Espressif, 2018-12-03:
In July and Augest 2018, we fixed some coexist problems about performance and stability. So could anyone tell me if it can work for you? Thanks.

chegewara, 2018-12-03:
@TianHao-Espressif I know this is about BT classic and wifi coex but i can confirm BLE + wifi coex works great. With SW coex set to wifi i can run infinite ble scan even with interval = window and at the same time have esp http server working without problem, at least works in my simple test.

TianHao-Espressif, 2018-12-15:
@chegewara , thanks for your comment. We'll continue to improve coex performance.

jack0c, 2022-01-05:
I know it is too late, but we have done the following things since ESP-IDFv3.3(already have MP in smart speaker):
1. implement a template based coexistence mechanism
2. improve the performance of important scenarios, for instance, A2DP + Wi-Fi
3. improve the compatiblity of A2DP with peer devices

closed 2022-05-17 by jack0c