espressif/esp-idf #1292: wifi cannot connect to windows hostednetwork/slow AP

sportybj, 2017-11-20:
ESP-IDF V2.1 downloaded on Nov 18, 2017,
http_request example
connect to WIFI router OK, while cannot connect to AP based on windows 7 hostednetwork, which can be connected via moible phone but slow to establish connection

 The serial port print:
I (111) wifi: wifi firmware version: 59e5fab
I (111) wifi: config NVS flash: enabled
I (111) wifi: config nano formating: disabled
[0;32mI (111) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (121) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (141) wifi: Init dynamic tx buffer num: 32
I (141) wifi: Init data frame dynamic rx buffer num: 32
I (141) wifi: Init management frame dynamic rx buffer num: 32
I (151) wifi: wifi driver task: 3ffbc31c, prio:23, stack:4096
I (151) wifi: Init static rx buffer num: 10
I (161) wifi: Init dynamic rx buffer num: 32
I (161) wifi: wifi power manager task: 0x3ffc4f98 prio: 21 stack: 2560
[0;32mI (171) example: Setting WiFi configuration SSID TestSSID...[0m
[0;32mI (201) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0[0m
I (201) wifi: mode : sta (30:ae:a4:01:68:a8)
I (1411) wifi: n:10 0, o:1 0, ap:255 255, sta:10 0, prof:1
I (2061) wifi: state: init -> auth (b0)
I (2071) wifi: state: auth -> assoc (0)
E (2071) wifi: Set status to INIT.

I (2071) wifi: state: assoc -> init (c)
I (2071) wifi: n:10 0, o:10 0, ap:255 255, sta:10 0, prof:1
I (2201) wifi: n:10 0, o:10 0, ap:255 255, sta:10 0, prof:1
I (2201) wifi: state: init -> auth (b0)
I (2201) wifi: state: auth -> assoc (0)
E (2201) wifi: Set status to INIT.

I (2201) wifi: state: assoc -> init (c)
I (2211) wifi: n:10 0, o:10 0, ap:255 255, sta:10 0, prof:1
I (2331) wifi: n:10 0, o:10 0, ap:255 255, sta:10 0, prof:1
I (2331) wifi: state: init -> auth (b0)
I (2341) wifi: state: auth -> assoc (0)
E (2341) wifi: Set status to INIT.

.......

zhangyanjiaoesp, 2017-12-12:
@sportybj 
I have tested with three windows hostednetwork/slow AP, both of them worked well. Sometimes it will print `I (2331) wifi: state: init -> auth (b0) I (2341) wifi: state: auth -> assoc (0)` several times, but finnally it will connect to  the hostednetwork AP. 
Did you fail to connect to  the hostednetwork AP every time you test? By the way, you can post the exact commit ID.

jack0c, 2018-01-02:
What's board do you use?

FayeY, 2018-01-11:
Closing as can't reproduce. Please feel free to reopen it if the issue persists and you can supply more details. Thanks.

closed 2018-01-11 by FayeY