espressif/esp-idf #1293: AES128 cmac used in bluetooth classic

pufstudio, 2017-11-20:
Hi.

I'm using a2dp bluetooth classic profile to stream music to a ESP32 based media player
we are manufacturing.
I've turned off all BLE features in menuconfig but AES encryption is still enabled.

According to spec AES should not be used in classic mode:
> Bluetooth uses Secure And Fast Encryption Routine + (SAFER+) [6] with a
> 128-bit key as an algorithm for authentication and key generation in Bluetooth versions
> up to 3.0+HS (High Speed), while Bluetooth 4.0 (i.e. Bluetooth Low Energy)
> replaces SAFER+ with the more secure 128-bit Advanced Encryption Standard
> (AES)

This is a log I get from ESP32 when connecting with a Samsung Galaxy S5 (android 6):
```
E (952593) BT: btm_sec_conn_req

E (952613) BT: btm_sec_connected

E (955263) BT: for SMP over BR max_key_size: 0x10,                        local_i_key: 0x07, local_r_key: 0x07

E (955263) BT: AES128_CMAC started, allocate buffer size = 16
E (955273) BT: flag = 0 round = 1
E (955273) BT: AES128_CMAC started, allocate buffer size = 16
E (955283) BT: flag = 0 round = 1
E (961113) BT: bta_av_rc_create ACP handle exist for shdl:0

```
That would not be an issue in itself but when AES encryption is used communication
is incredibly slow and pairing takes significantly longer... Once it is paired then maybe one out of ten times music actually starts playing, usually connection is dropped after some time.

Other smartphones like zte blade III (android 4.4.4) or nokia 3 (android 7) do not use
AES when connected and it's all working as intended.
```
E (2991883) BT: btm_sec_conn_req

E (2991893) BT: btm_sec_connected

E (2992133) BT: bta_av_rc_create ACP handle exist for shdl:0

```
Samsung Galaxy works fine with other bluetooth modules (like ovc3860 or rn52)
so the issue must be on the ESP32 side.

Is there a way to disable support for AES when in classic mode?

mringwal, 2017-11-21:
hi. it looks like your Android phones supports LE Secure Connections over Bluetooth Classic. (for SMP over BR max_key_size: 0x10 - key size is only used in LE Security Manager).

I don't think LE pairing via BR is causing an issue for audio streaming.
The BTstack A2DP Sink Demo doesn't support I2S on the ESP32 for output, but you could try to see if audio streaming is solid. Alternatively, if you set the remote device, you can try the a2dp_source_demo that connects to a Bluetooth speaker and plays a song.

In both cases, please comment the call to hci_dump_open in esp-idf/components/btstack/ after running ./integrate_btstack.py

Getting started here: https://github.com/bluekitchen/btstack/tree/master/port/esp32

pufstudio, 2017-11-26:
Hi, with BTstack it looks like it's working, at least streaming is uninterrupted.
AVRCP seems to be a hit and miss, connection often could not be established and
playing status and element attributes are sometimes not reported properly
(eg. song state is reported as paused while it's actually playing).

I'm not sure what exactly is causing the issue with bluedroid but I would like to disable BLE altogether because it's not needed in this case.
I set BLE_INCLUDED directive to FALSE but that generated a long list of errors and warnings
so I decided to not pursue that route.

Wireshark shows that ESP32 reports LMP version 8, aka Bluetooth 4.2
[image]
I tried to find a way to degrade a reported version (to 3.0+HS or 2.1+EDR) but it looks like it's not handled by the stack but by precompiled bt library so that's also a no go.

I understand that it could be an issue on the phone's side (it's LMP version 7, Bluetooth 4.1) but it's working with other non LE capable audio modules so I believe there should be a way to achieve the same with ESP32.

mringwal, 2017-11-27:
HI. If you like you can post a full packet log that shows the AVRCP connection errors with BTstack, so we can have a look.

pufstudio, 2017-11-28:
Hi.  Here are the logs.
Second one is with hci_dump_open.
1: https://gist.github.com/pufstudio/e2248785ad8886f8ac474038250cdc62
2: https://gist.github.com/pufstudio/edd349b9230224a72a905bdf3cf506b8

Thanks.

pufstudio, 2018-01-03:
This commit fixes the issue, https://github.com/espressif/esp-idf/commit/c401a74bfb2eecb6ebb407c6a2494cf696b3cfeb Working great now.

closed 2018-02-03 by FayeY