espressif/esp-idf #1294: getting a flash error

assegid, 2017-11-20:
I am testing the esp-idf to run the Blink code on a Sparkfun esp32 thing board, but I am unable to flash the board. I am on Windows 7 enterprise and followed the Windows installation of the IDF. make menuconfig and make app work but fails to flash. I have also tried pressing both rst and gpio0 buttons in various combinations. The mingw32 output is below. 

HIDA-4069519+student@HIDA-4069519 MINGW32 ~/esp-idf/blink
$ make flash
/bin/sh: -c: line 0: syntax error near unexpected token `('
/bin/sh: -c: line 0: `echo "Flashing binaries to serial port "USB Serial Port(COM21)" (app at offset 0x10000)..."'
make: *** [/home/student/esp-idf/components/esptool_py/Makefile.projbuild:54: flash] Error 1

assegid, 2017-11-20:
Just an update. Hopefully this will help other folks. Just selecting the correct port in menuconfig  does not seem to work. make flash ESPPORT=COM21 and then pressing GPIO0 button when it shows connecting..... works for flashing.

While writing this update I was testing another one of the examples and I am suspecting the problem may have been the way the serial port was entered in the menuconfig. In device manager in windows the port is listed as USB Serial Port(COM21), so that was how I entered it. I believe mingw32 does not like spaces and that may have been the issue. Perhaps another developer can confirm.

negativekelvin, 2017-11-21:
Just COM21

closed 2018-01-08 by igrr