espressif/esp-idf #1295: Update to 2.1 Causes Loss of Range

BrettGoldbach, 2017-11-20:
Hi All,
Currently I am using esp-idf v2.0 release, but I tried updating to v2.1 to fix some known issues for which it was reported to fix (Device unable to connect to certain routers, and device UART periodically printing out gibberish). The update was easily applied and did fix the two issues mentioned above, however the update seemed to effectively reduce the range of the device.

The device is still able to connect to the routers at t a distance, and use the connection effectively for about 30 seconds, but then it just begins to fail to establish a connection to the remote server. I am using mongoose the handle http connections and have tried updating from 6.7.1 to the most recent 6.10 with no success.

Here are my logs from a successful connection to a AP within 15 ft (with ip addresses obscured):
```
mg_socket_if_init    0x3ffe1280 using select()
mg_mgr_init_opt      ==================================
mg_mgr_init_opt      init mgr=0x3ffe1280
mg_connect_http_base http://IPADDRESS/index.php/v1/device use_ssl? 0 tcp://IPADDRESS:80
mg_do_connect        0x3ffeeb30 tcp://IPADDRESS:80
mg_socket_if_connect 0x3ffeeb30 sock 3 rc -1 errno 119 err 0
mg_add_conn          0x3ffe1280 0x3ffeeb30
mg_mgr_handle_conn   0x3ffeeb30 fd=3 fd_flags=2 nc_flags=8 rmbl=0 smbl=19530
mg_if_connect_cb     0x3ffeeb30 connect, err=0
mg_call              0x3ffeeb30 proto ev=2 ev_data=0x3ffe1190 flags=0 rmbl=0 smbl=19530
mg_call              0x3ffeeb30 user ev=2 ev_data=0x3ffe1190 flags=0 rmbl=0 smbl=19530
mg_call              0x3ffeeb30 after user flags=0 rmbl=0 smbl=19530
mg_call              0x3ffeeb30 after proto flags=0 rmbl=0 smbl=19530
mg_write_to_socket   0x3ffeeb30 5840 bytes -> 3
mg_if_sent_cb        0x3ffeeb30 5840
mg_call              0x3ffeeb30 proto ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=13690
mg_call              0x3ffeeb30 user ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=13690
mg_call              0x3ffeeb30 after user flags=0 rmbl=0 smbl=13690
mg_call              0x3ffeeb30 after proto flags=0 rmbl=0 smbl=13690
mg_mgr_handle_conn   0x3ffeeb30 after fd=3 nc_flags=0 rmbl=0 smbl=13690
mg_mgr_handle_conn   0x3ffeeb30 fd=3 fd_flags=2 nc_flags=0 rmbl=0 smbl=13690
mg_write_to_socket   0x3ffeeb30 3720 bytes -> 3
mg_if_sent_cb        0x3ffeeb30 3720
mg_call              0x3ffeeb30 proto ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=9970
mg_call              0x3ffeeb30 user ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=9970
mg_call              0x3ffeeb30 after user flags=0 rmbl=0 smbl=9970
mg_call              0x3ffeeb30 after proto flags=0 rmbl=0 smbl=9970
mg_mgr_handle_conn   0x3ffeeb30 after fd=3 nc_flags=0 rmbl=0 smbl=9970
mg_mgr_handle_conn   0x3ffeeb30 fd=3 fd_flags=2 nc_flags=0 rmbl=0 smbl=9970
mg_write_to_socket   0x3ffeeb30 5840 bytes -> 3
mg_if_sent_cb        0x3ffeeb30 5840
mg_call              0x3ffeeb30 proto ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=4130
mg_call              0x3ffeeb30 user ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=4130
mg_call              0x3ffeeb30 after user flags=0 rmbl=0 smbl=4130
mg_call              0x3ffeeb30 after proto flags=0 rmbl=0 smbl=4130
mg_mgr_handle_conn   0x3ffeeb30 after fd=3 nc_flags=0 rmbl=0 smbl=4130
mg_mgr_handle_conn   0x3ffeeb30 fd=3 fd_flags=2 nc_flags=0 rmbl=0 smbl=4130
mg_write_to_socket   0x3ffeeb30 4130 bytes -> 3
mg_if_sent_cb        0x3ffeeb30 4130
mg_call              0x3ffeeb30 proto ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=0
mg_call              0x3ffeeb30 user ev=4 ev_data=0x3ffe1160 flags=0 rmbl=0 smbl=0
mg_call              0x3ffeeb30 after user flags=0 rmbl=0 smbl=0
mg_call              0x3ffeeb30 after proto flags=0 rmbl=0 smbl=0
mg_mgr_handle_conn   0x3ffeeb30 after fd=3 nc_flags=0 rmbl=0 smbl=0
mg_mgr_handle_conn   0x3ffeeb30 fd=3 fd_flags=1 nc_flags=0 rmbl=0 smbl=0
mg_handle_tcp_read   0x3ffeeb30 257 bytes (PLAIN) <- 3
mg_recv_common       0x3ffeeb30 257 0
mg_call              0x3ffeeb30 proto ev=3 ev_data=0x3ffe1150 flags=0 rmbl=257 smbl=0
mg_call              0x3ffeeb30 user ev=3 ev_data=0x3ffe1150 flags=0 rmbl=257 smbl=0
mg_call              0x3ffeeb30 after user flags=0 rmbl=257 smbl=0
mg_http_handler2     0x3ffeeb30 IPADDRESS:80
mg_call              0x3ffeeb30 user ev=102 ev_data=0x3ffe0f80 flags=0 rmbl=257 smbl=0
mg_call              0x3ffeeb30 after user flags=0 rmbl=257 smbl=0
mg_call              0x3ffeeb30 user ev=101 ev_data=0x3ffe0f80 flags=0 rmbl=257 smbl=0
mg_call              0x3ffeeb30 after user flags=2048 rmbl=257 smbl=0
mg_call              0x3ffeeb30 after proto flags=2048 rmbl=0 smbl=0
mg_mgr_handle_conn   0x3ffeeb30 after fd=3 nc_flags=2048 rmbl=0 smbl=0
mg_close_conn        0x3ffeeb30 2048 3
mg_call              0x3ffeeb30 proto ev=5 ev_data=0x0 flags=2048 rmbl=0 smbl=0
mg_call              0x3ffeeb30 user ev=5 ev_data=0x0 flags=2048 rmbl=0 smbl=0
mg_call              0x3ffeeb30 after user flags=2048 rmbl=0 smbl=0
mg_call              0x3ffeeb30 after proto flags=2048 rmbl=0 smbl=0
mg_mgr_free          0x3ffe1280
```

And here is a log of the failed connect to a AP about 30 ft away:
```
mg_socket_if_init    0x3ffe1280 using select()
mg_mgr_init_opt      ==================================
mg_mgr_init_opt      init mgr=0x3ffe1280
mg_connect_http_base http://IPADDRESS/index.php/v1/device use_ssl? 0 tcp://52.207.116.53:80
mg_do_connect        0x3ffeeb58 tcp://IPADDRESS:80
mg_socket_if_connect 0x3ffeeb58 sock 1 rc -1 errno 119 err 0
mg_add_conn          0x3ffe1280 0x3ffeeb58
// about 8 seconds pass until it times out and closes
mg_mgr_free          0x3ffe1280
mg_close_conn        0x3ffeeb58 2056 1
mg_call              0x3ffeeb58 proto ev=5 ev_data=0x0 flags=2056 rmbl=0 smbl=19530
mg_call              0x3ffeeb58 user ev=5 ev_data=0x0 flags=2056 rmbl=0 smbl=19530
mg_call              0x3ffeeb58 after user flags=2056 rmbl=0 smbl=19530
mg_call              0x3ffeeb58 after proto flags=2056 rmbl=0 smbl=19530
```

The whole time the wifi handler never reports any disconnects like I would normally see at the edge of reception.

This may be more suited for a mongoose forum, but I know that mongoose is used by a lot of people here, so I wanted to see if anyone has come across this as well.

When I switch back to v2.0 release I am able to go to at least 60 ft without any issues. And I have checked the make menu config, making sure that TX power and similar settings havent changed.

Any help is greatly appreciated.
Thanks,
- Brett

chegewara, 2017-11-20:
@BrettGoldbach You should have obscure IPADDRESS in 2nd log too.

BrettGoldbach, 2017-11-20:
DOH, oh well, it is being a bit overly cautious already.

zhangyanjiaoesp, 2017-12-11:
@BrettGoldbach You can use the latest esp-idf version to test

jack0c, 2018-01-02:
@BrettGoldbach Maybe you can update to 2.1.1 or idf3.0, you can check your menuconfig, about the PHY calibration configure.

FayeY, 2018-01-05:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-05 by FayeY