espressif/esp-idf #1296: menuconfig on Windows doesn't work if mingw32 gcc is installed

dschaefer, 2017-11-21:
I have the msys32 installed as per the install script. I also have mingw32's gcc installed. If you run the mingw32 shell, that gcc runs. However it is not capable of compiling the output of flex. I get the following:

In file included from zconf.tab.c:2448:0:
zconf.lex.c:1076:10: fatal error: wordexp.h: No such file or directory
 #include <wordexp.h>
          ^~~~~~~~~~~
compilation terminated.

Not sure what can be done with that since there is no flex for mingw32, only msys. And that only builds with the msys gcc. The workaround is to do the menuconfig with the msys shell and the rest from mingw32.

projectgus, 2017-11-22:
Hi @dschaefer,

I see. I think you're right that there's currently no way to support "wordexp" in a pure mingw32 environment. (For what it's worth, the problem is less with flex and more with kconfig itself, the input zconf.l file already has wordexp.h hardcoded into it.)

The only other workarounds I can think of are to keep a separate MSYS root for IDF use only, or to override CC & LD via make's environment. ie:

```
CC=/usr/bin/gcc LD=/usr/bin/ld make menuconfig
```

I'm going to close this because I don't think there's anything we can do to fix it properly in IDF, but hopefully anyone who bumps into this error will find this issue when they search. Please reopen if you think we should address it in some other way.

dschaefer, 2017-11-22:
OK. Thanks! The workarounds are fine. But they should probably be in the documentation in case someone else runs into it..

projectgus, 2017-11-22:
Fair enough. Reopening until we add a note to the docs.

projectgus, 2017-12-07:
Ended up being easier to work around this transparently in the Makefile than add a note to the docs.

We have just discovered one remaining known issue though, Eclipse on Windows will still use the wrong compiler - Python is a mingw32 Python and it prepends C:\msys32\mingw32\bin to the PATH which causes cc & ld to still be the mingw32 versions.

Reopening until we have this fixed in Eclipse as well...

dschaefer, 2017-12-07:
BTW, I will be publishing an article on how to use CDT's new Core Build with esp-idf. It's a lot simpler than the older managed build and shows off how to add support for a platform and toolchain without writing an Eclipse plugin.

I still have to figure out how to run make menuconfig. Maybe in the terminal view, maybe in an external msys2 shell. A Kconfig editor would be really helpful ;) but I'll leave that to the community.

projectgus, 2017-12-07:
> BTW, I will be publishing an article on how to use CDT's new Core Build with esp-idf. It's a lot simpler than the older managed build and shows off how to add support for a platform and toolchain without writing an Eclipse plugin.

This is very interesting!

> I still have to figure out how to run make menuconfig. Maybe in the terminal view, maybe in an external msys2 shell. A Kconfig editor would be really helpful ;) but I'll leave that to the community.

We have had some discussion about this internally as well, but no plans yet.

igrr, 2018-01-08:
Should be resolved in https://github.com/espressif/esp-idf/commit/ccea4a0f8f688abc0ab66d340212074864fd3229.

closed 2018-01-08 by igrr