espressif/esp-idf #1297: Unable to use wpa2_enterprise example on ESP32

mduanaa, 2017-11-21:
I try to flash the wpa2_enterprise example to my ESP32 board but facing endless failure. The debug info. is as follow.
 //debug info. begins
I (246) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (246) wifi: mode : sta (24:0a:c4:0e:56:74)
I (376) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (1346) wifi: state: init -> auth (b0)
I (1356) wifi: state: auth -> assoc (0)
I (1366) wifi: state: assoc -> run (10)
I (1366) wpa: wpa2_task prio:2, stack:6656

I (1376) wpa: TLSv1: tlsv1_add_cert_der - failed to parse certificate
I (1376) wpa: TLS: Failed to configure trusted CA certificates
I (1376) wpa: TLS: Failed to set TLS connection parameters
I (1386) wpa: EAP-PEAP: Failed to initialize SSL.
E (1386) wpa: Method private structure allocated failure

I (2396) wpa: >>>>>wpa2 FIALED

I (2406) wpa: wpa2 task delete

I (4246) example: ~~~~~~~~~~~
I (4246) example: IP:0.0.0.0
I (4246) example: MASK:0.0.0.0
I (4246) example: GW:0.0.0.0
I (4246) example: ~~~~~~~~~~~
I (4366) wifi: pm start, type:0
//debug info. ends

There is no logging info. in the Radius Server which means this program fails at the self-configuration stage. The issues seems to be the certification verification. 

//////////////////////////////////////////////////////////////////////////////////////////////////////////////
I tried to comment the certification command in the wpa2_enterprise_main.c as someone says the certification is optional for EAP+PEAP. However, the connection is still unsuccessful. 
The corresponding error info. is : Guru meditation error: core 1 panic'ed(LoadProhibited). Exception was unhandled. 

//debug info. begeins
Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40129a71  PS      : 0x00060330  A0      : 0x8012af39  A1      : 0x3ffc83f0
0x40129a71: bignum_set_unsigned_bin at ??:?

A2      : 0x3ffba464  A3      : 0x00000001  A4      : 0x3ffca378  A5      : 0x3ffca379
A6      : 0x3ffc84c4  A7      : 0x00000001  A8      : 0x80129a6c  A9      : 0x3ffc83d0
A10     : 0x3ffba478  A11     : 0x000000ff  A12     : 0x3ffba464  A13     : 0x92492493
A14     : 0x00000014  A15     : 0x00000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001  LBEG    : 0x40129a59  LEND    : 0x40129a5d  LCOUNT  : 0x00000000
0x40129a59: bignum_set_unsigned_bin at ??:?

0x40129a5d: bignum_set_unsigned_bin at ??:?

Backtrace: 0x40129a71:0x3ffc83f0 0x4012af36:0x3ffc8410 0x400fd81d:0x3ffc8440 0x400fda17:0x3ffc84f0 0x400fa742:0x3ffc85c0 0x400fab31:0x3f
fc8600 0x400f9dc1:0x3ffc8680 0x400f9844:0x3ffc86d0 0x400f989d:0x3ffc8710 0x400f7ea8:0x3ffc8730 0x400f75fe:0x3ffc8770 0x400f5c1d:0x3ffc87
d0 0x400f5f75:0x3ffc8810
//debug info ends
///////////////////////////////////////////////////////////////////////////////////////////////
Supplement Information: 1) latest esp-idf version; 2) wifi try to be reached eduroam; 3) development enviroment / tools: windows7-64bit  msys32 / mingw32

Can anyone give a hint? Thanks a lot.

mduanaa, 2017-11-24:
@XinDeng11 Can you please help ?

XinDeng11, 2017-11-26:
@mduanaa Hi, the issue may caused by certificate not match, in the main folder in the example, you can find some certificates, you need copy the certificates relative to server to the radius server, and renamed them as server.crt, server.key and server.pem, after this, this issue will solved. What's more. there is a simple way to verify whether is due to the certificate, just mark the certificates check function in the example, esp_wifi_sta_wpa2_ent_set_ca_cert() and esp_wifi_sta_wpa2_ent_set_cert_key(), if it work, it should be the reason

mduanaa, 2017-11-27:
@XinDeng11 Thanks for your suggestions. However, I have no access to the Radius Server and I don't think the manager of the Radius Server of our university would let me do that. The debug info. shown in my first trial (failed to parse the certificate) indicates  even the certificate in the example folder is not recognized by your example code. I don't know how many people has tried this example and how many successful cases ?

Yoshi20, 2017-12-01:
Faced lots and lots of problem with this example too.
Try to use the exact same kind of certs as in the example. Others doesn't seem to work.
E.g. SHA256 is/was not working (use SHA1) and everything other than 2048 Bit is/was also not working. (Last tested one or two months ago)

XinDeng11, 2017-12-05:
@mduanaa @Yoshi Dearing, the wpa2_enterprise example and its certificates are surely 100% OK, many newers failed because they didn't have matched certificates between client and server, certificates verify based on RSA algorithm. It needs matched public key and private key. That's why we add the both server and client's certificates in the main floder, but you need build an server on linux at first. Then put the ca.pem, server.pem, sever.crt and server.pem. in the server. @mduanaa, What's more, if you try with your school's server, you need use the client's certificates offered by school server. Also make sure the EAP method is PEAP or TLS, because we noly support those two methods in the example now.

negativekelvin, 2017-12-05:
@XinDeng11 I think you need more example certificates with different key lengths and better error messages to figure out what the problem is when it doesn't work

XinDeng11, 2017-12-05:
@negativekelvin We have tried with 1024 bit, 2048bit and 4096 bit key, all of them are work, what you mentioned add more error logs and log switch, that what we plan to do in the next version. After it finished, I will let you know, thanks.

nubiator123, 2018-01-21:
@XinDeng11 I have exactly the same problems that @mduanaa mentioned with 4096 bit certificates.

Backtrace: 0x4013cd85:0x3ffdd740 0x4013dae9:0x3ffdd760 0x401398ca:0x3ffdd780 0x400eab05:0x3ffdd7b0 0x400eaceb:0x3ffdd860 0x400e7a7e:0x3ffdd930 0x400e7e6d:0x3ffdd970 0x400e70e9:0x3ffdd9f0 0x400e6b56:0x3ffdda40 0x400e6bb9:0x3ffdda80 0x400e5274:0x3ffddaa0 0x400e5865:0x3ffddae0 0x400e3029:0x3ffddb30 0x400e3389:0x3ffddb70

0x4013cd85: mp_read_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/libtommath.h line 990
0x4013dae9: bignum_set_unsigned_bin at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/bignum.c line 114
0x401398ca: crypto_mod_exp at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/wpa_supplicant/src/crypto/crypto_internal-modexp.c line 40
0x400eab05: x509_certificate_check_signature at ?? line ?
0x400eaceb: x509_certificate_chain_validate at ?? line ?
0x400e7a7e: tlsv1_client_set_cred at ?? line ?
0x400e7e6d: tlsv1_client_process_handshake at ?? line ?
0x400e70e9: tlsv1_client_handshake at ?? line ?
0x400e6b56: tls_connection_handshake2 at ?? line ?
0x400e6bb9: tls_connection_handshake at ?? line ?
0x400e5274: eap_peer_tls_process_helper at ?? line ?
0x400e5865: eap_peer_tls_phase2_nak at ?? line ?
0x400e3029: eap_sm_process_request at ?? line ?
0x400e3389: wpa2Task at ?? line ?

Tested with all available examples. Only worked until August, since than only Guru Mediation Error

Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4013cd85  PS      : 0x00060b30  A0      : 0x8013daec  A1      : 0x3ffdd740  
A2      : 0x3ffce8d4  A3      : 0x00000001  A4      : 0x3ffe30a0  A5      : 0x3ffe309f  
A6      : 0x00000002  A7      : 0x00000005  A8      : 0x8013cd83  A9      : 0x3ffdd720  
A10     : 0x00000000  A11     : 0x00000008  A12     : 0x3ffce8d4  A13     : 0x00000000  
A14     : 0x00000032  A15     : 0x00000020  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000

ToBeSmavoo, 2018-01-21:
@XinDeng11 Hi, we are running into the same Guru Mediation error with a properly configured PKI since the end of August. Before that, the same binary and certificates are running without any issues. The difference to the certificates provided by the example seams to be the size of the CA certificate and the used signing algorithm. The example is using a 1024bit CA certificate and SHA-1 to sign the client/server certificates. Both leads to smaller signatures, which seams to be the cause, why all other configurations fail.

A test implementation with a fresh CA and the same certificate parameters works perfectly. If we than change the size of the CA certificate to 2048bit and use a state-of-the-art SHA-265 signing algorithm, we run again into the same Guru Mediation error. In both cases we exchange the ca.pem, client.crt and client.key on the client-side and the ca.pem, server.crt and server.key on the radius-server-side. 

I've attached the certificates for both cases. It would be a big help if you could test both configurations and analyse, why 2048bit certificates (CA and client/server) in combination with SHA-256 are causing this issue. From my point of view, this configuration is pretty standard in enterprise environments and should be supported by the esp-idf.

Thanks in advance.

[ca-1024.zip](https://github.com/espressif/esp-idf/files/1650292/ca-1024.zip)
[ca-2048.zip](https://github.com/espressif/esp-idf/files/1650293/ca-2048.zip)

nubiator123, 2018-02-09:
@igrr Do you have an idea, what line is causing this / how to use a workaround? Currently the WPA2 support only works with ca with 1024. However eduroam is using 4096bit certificates. The WPA2 example currently only works with commits prior to May - August.

nubiator123, 2018-03-04:
@XinDeng11: Are there any Updates for this issue? Could you find the rootcause?

Many thanks

XinDeng11, 2018-03-20:
@nubiator123 @mduanaa @ToBeSmavoo Hi, guys.We have reproduced the phenmenon and found the root casue, the issue had fixed already, maybe you can check the commit bc97b0f3 for detail, thanks.

negativekelvin, 2018-03-20:
@XinDeng11 does it also fix this tls alert issue https://github.com/espressif/esp-idf/issues/248#issuecomment-360338336

mtowara, 2018-03-20:
@XinDeng11 there doesn't seem to be a commit bc97b0f3 in the repo, can you please link to it?

igrr, 2018-03-20:
@mtowara Apologies, the fix (along with a few other changes) did not make it to Github yet. It may take an hour or two for the integration tests to complete, then the change will be deployed to Github.

mtowara, 2018-03-21:
With the new commits I see connection attempts in my eduroam logs, which is an improvement.
However I'm now stuck in an boot loop with the following dump:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x4000c2e0  PS      : 0x00060630  A0      : 0x80104e0c  A1      : 0x3ffca040  
A2      : 0x3ffb66c8  A3      : 0x00000000  A4      : 0x00000020  A5      : 0x3ffb66c8  
A6      : 0x3ffbacd6  A7      : 0x00000002  A8      : 0x00000000  A9      : 0x3ffca000  
A10     : 0x00000003  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x3ffca078  
A14     : 0x3ffc73d0  A15     : 0x00000000  SAR     : 0x0000001f  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000001  

Backtrace: 0x4000c2e0:0x3ffca040 0x40104e09:0x3ffca050 0x40107296:0x3ffca070
0x40104e09: wpa_set_pmk at ??:?
0x40107296: wpa2Task at ??:?

I'm running the wpa2_enterprise example adapted for username / pw instead of certificates.

averri, 2018-05-04:
I have never had success connecting the ESP32 to a WPA2 Enterprise network. This is very frustrating. 
The other scenario that does not work: https://github.com/espressif/arduino-esp32/issues/1381

pfe-aicoe, 2018-06-20:
Hi, Can someone please help me with the following error:
IP address: 
0.0.0.0

E (3905) wpa: Method private structure allocated failure

E (33913) wpa: Method private structure allocated failure

[D][WiFiGeneric.cpp:304] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:309] _eventCallback(): Reason: 4 - ASSOC_EXPIRE

I am trying to connect my ESP32 with a WPA2 -Enterprise network.

ybuyankin, 2018-08-20:
Hi @igrr sorry I've posted this in the arduino-esp32 first but supposedly it's related here more. wpa2_enterprise example is struggling with connecting to WS2008 / Cisco-based corporate PEAP-MSCHAPv2 network. I'm in a no way a wpa2 expert but it seems that it depends on the certificates configuration (or maybe negotiation scenario) on the network and the wpa2 supplicant code somehow did not ignore optional certificate checking results so it's either failing on checking server self-signed certificate when ca cert is supplied like this:

wpa: X509: Did not find any of the issuers from the list of trusted certificates
wpa: TLSv1: Server certificate chain validation failed (reason=6)

or failing to provide a proper certificate to server when it was not set up (by commenting out these lines):

ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_ca_cert(ca_pem_start, ca_pem_bytes) ); ESP_ERROR_CHECK( esp_wifi_sta_wpa2_ent_set_cert_key(client_crt_start, client_crt_bytes, client_key_start, client_key_bytes, NULL, 0) );

It fails this time on server side after this diagnostic appears locally:

wpa: TLSv1: Full client certificate chain not configured - validation may fail

What is still not quite clear to me is why is the local decision to test for remote server certificate validity is based on local one being supplied or not.

Android phones are connecting to the network just fine without any certificates being installed.

All I know about network configuration so far is that it has been set years ago (and still working just fine) up by admin with this Cisco how-to (so this setup expected to be also very common):

[https://www.cisco.com/c/en/us/support/docs/wireless/5500-series-wireless-controllers/115988-nps-wlc-config-000.html
](url)

There is a discussion here as well, and there is at least one guy with exactly the same problem.

[https://www.esp32.com/viewtopic.php?f=2&t=3108&p=29227#p29227](url)

Something is still wrong with PEAP/MSCHAPv2 in esp32 and this is gating us from finally going to production. Help please, thanks!

sagb2015, 2021-02-03:
We now have mbedtls port for supplicant TLS connection as well as option to enable/disable server certificate validation. Please help to check with latest build. If not seen  we will close this issue.

WilliamAtek, 2021-02-03:
Please tell us how to test even with master pull and phase 2 selected to MSCHAPV2 it doesn't work ! Do we have to change something in menuconfig ? I also saw that LwIP can do mschap ???? I tried to enable it no success either.

ayoubazz, 2021-02-04:
I'm trying to connect to our wpa2 enterprise, but  it no success 
The debug info. is as follow :
E (370) WPA: Method private structure allocated failure
->
->

WilliamAtek, 2021-02-08:
Still not working logs : 

`D (19:00:09.514) WiFi: Received Event base WIFI_EVENT

I (19:00:09.516) wpa: WPA2: already enabled
D (10455) wifi:Start wifi connect
D (10456) wifi:connect status 3 -> 0
D (10468) wifi:connect chan=0
D (10469) wifi:first chan=11
D (10470) wifi:connect status 0 -> 1
D (10471) wifi:filter: set rx policy=3
D (10473) wifi:clear scan ap list
D (10484) wifi:start scan: type=0x50f, priority=2, cb=0x401ad65c, arg=0x0, ss_state=0x1, time=5253113, index=0
0x401ad65c: cnx_start_handoff_cb at ??:?

D (10487) wifi:perform scan: ss_state=0x9, chan<11,0>, dur<0,120>
I (19:00:09.553) WiFi: retry to connect to the AP
D (10493) wifi:rsn valid: gcipher=3 ucipher=3 akm=4

D (10505) wifi:profile match: ss_state=0x7
D (10610) wifi:scan end: arg=0x0, status=0, ss_state=0x7
D (10612) wifi:find first mathched ssid, scan done
D (10614) wifi:filter: set rx policy=4
D (10615) wifi:first chan=1
D (10616) wifi:handoff_cb: status=0
D (10628) wifi:ap found, mac=a8:40:41:1d:2b:e8
D (10629) wifi:new_bss=0x3ffb7790, cur_bss=0x0, new_chan=<11,0>, cur_chan=11
D (10632) wifi:filter: set rx policy=5
I (10633) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
D (10646) wifi:connect_op: status=0, auth=4, cipher=3
D (10647) wifi:auth mode is not none
D (19:00:09.712) wpa: RSN: PMKSA cache search - network_ctx=0x0 try_opportunistic=0
D (19:00:09.725) wpa: RSN: Search for BSSID a8:40:41:1d:2b:e8
D (19:00:09.727) wpa: RSN: No PMKSA cache entry found
D (10666) wifi:connect_bss: auth=1, reconnect=0
I (10678) wifi:state: init -> auth (b0)
D (10680) wifi:start 1s AUTH timer
D (10681) wifi:clear scan ap list
D (10686) wifi:recv auth: seq=2, status=0
I (10688) wifi:state: auth -> assoc (0)
D (10689) wifi:restart connect 1s timer for assoc
D (10704) wifi:recv assoc: type=0x10
D (10706) wifi:filter: set rx policy=6
I (10707) wifi:state: assoc -> run (10)
I (19:00:09.772) wpa: wpa2_task prio:2, stack:6656

D (10711) wifi:start 30s connect timer for 4 way handshake
D (19:00:09.787) wpa: WPA2: wifi->wpa2 api completed sig(1073574784)
D (19:00:09.790) wpa: WPA2: wpa2 api return, sm->state(1)
D (19:00:09.792) wpa: IEEE 802.1X RX: version=2 type=0 length=5

D (19:00:09.805) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=9):
D (19:00:09.807) wpa: 02 00 00 05 01 14 00 05 01
D (19:00:09.809) wpa: WPA2: wifi->wpa2 api completed sig(1073574848)
D (19:00:09.822) wpa: WPA2: wpa2 api return, sm->state(1)
D (19:00:09.828) wpa: IEEE 802.1X RX: version=2 type=0 length=5

D (19:00:09.831) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=9):
D (19:00:09.833) wpa: 02 00 00 05 01 d2 00 05 01
D (19:00:09.845) wpa: WPA2: wifi->wpa2 api completed sig(1073574688)
D (19:00:09.848) wpa: WPA2: wpa2 api return, sm->state(1)
D (19:00:09.862) wpa: IEEE 802.1X RX: version=2 type=0 length=6

D (19:00:09.865) wpa: WPA2: RX EAPOL-EAP PACKET - hexdump(len=10):
D (19:00:09.867) wpa: 02 00 00 06 01 d3 00 06 19 20
D (19:00:09.869) wpa: TLS: using phase1 config options
D (19:00:09.884) wpa: SSL: Received packet(len=6) - Flags 0x20
D (19:00:09.886) wpa: EAP-PEAP: Start (server ver=0, own ver=1)
D (19:00:09.888) wpa: EAP-PEAP: Using PEAP version 0
D (19:00:09.902) wpa: SSL: 280 bytes left to be sent out (of total 280 bytes)
[... 118 lines cut ...]
D (11573) wifi:remove 00:00:00:00:00:00 from rc list
I (11575) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1
D (11578) wifi:filter: set rx policy=8
D (11579) wifi:Send disconnect event, reason=23, AP number=0
D (19:00:10.654) event: running post WIFI_EVENT:5 with handler 0x40159a58 and context 0x3ffceaf0 on loop 0x3ffcd83c
0x40159a58: wifi_default_action_sta_disconnected at C:/atek/dev/esp32/esp-idf/components/esp_wifi/src/wifi_default.c:107

D (19:00:10.658) esp_netif_handlers: esp_netif action disconnected with netif0x3ffce7f8 from event_id=5
D (19:00:10.672) esp_netif_lwip: check: remote, if=0x3ffce7f8 fn=0x4014301c
0x4014301c: esp_netif_down_api at C:/atek/dev/esp32/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:1154

D (19:00:10.674) esp_netif_lwip: esp_netif_down_api esp_netif:0x3ffce7f8
D (19:00:10.687) esp_netif_lwip: esp_netif_start_ip_lost_timer esp_netif:0x3ffce7f8
D (19:00:10.690) esp_netif_lwip: if0x3ffce7f8 start ip lost tmr: no need start because netif=0x3ffce85c interval=60 ip=0
D (19:00:10.704) esp_netif_lwip: check: local, if=0x3ffce7f8 fn=0x40143798
0x40143798: esp_netif_update_default_netif_lwip at C:/atek/dev/esp32/esp-idf/components/esp_netif/lwip/esp_netif_lwip.c:163

D (19:00:10.707) esp_netif_lwip: esp_netif_update_default_netif_lwip 0x3ffce7f8
D (19:00:10.720) esp_netif_lwip: call api in lwip: ret=0x0, give sem
D (19:00:10.722) event: running post WIFI_EVENT:5 with handler 0x4011eb9c and context 0x3ffd4c68 on loop 0x3ffcd83c

D (19:00:10.736) WiFi: Received Event base WIFI_EVENT`

Server side log :
`The supplied message is incomplete. The signature was not verified.`

kapilkedawat, 2021-02-10:
Hi @WilliamAtek ,

Is it possible to provide whole server logs? 
Please run the authentication server in debug mode to capture all the logs..

Please also provide following along with that:

1. IDF version used(commit id).
2. Client certificates
3. Server certificates
4. Sniffer captures during the issue.
5. sdkconfig
6. Authentication server and system info(freeradius/hostapd? ubuntu/other version)

In case you have recently pulled the IDF, please make sure you have updated the submodules.

WilliamAtek, 2021-02-10:
I tried with 2 idf version latest master at the time v4.3-dev-2940-g6e776946d and current latest stable release v4.2-47-g2532ddd9f

I see that there is some commit related to wpa fixes recently I will try with those one.

I don't have much more log with the server as you can see in esp logs it just stops the process in the middle of it.

So the server just receive : The supplied message is incomplete. The signature was not verified.

The server is Windows Server 2016 everything is up to date. The access point is openwrt based. Wifi does work on windows and linux devices.

Client certificates : 
wifi_ca.pem :
-----BEGIN CERTIFICATE-----
MIICRDCCAa2gAwIBAgIJAIutIKrV1gxNMA0GCSqGSIb3DQEBCwUAMFYxCzAJBgNV
BAYTAkNBMQswCQYDVQQIDAJRQzEMMAoGA1UEBwwDTVRMMQ0wCwYDVQQKDARBVEVL
MQ0wCwYDVQQLDARBVEVLMQ4wDAYDVQQDDAVBVEVLMjAeFw0yMTAyMDQxNDU3MDBa
Fw0yNjAyMDQxNDU3MDBaMFYxCzAJBgNVBAYTAkNBMQswCQYDVQQIDAJRQzEMMAoG
A1UEBwwDTVRMMQ0wCwYDVQQKDARBVEVLMQ0wCwYDVQQLDARBVEVLMQ4wDAYDVQQD
DAVBVEVLMjCBnzANBgkqhkiG9w0BAQEFAAOBjQAwgYkCgYEAra2AHgOFFnXnEtAR
0Gcw0BNlxzr+PtgwL49geKgrNywC8W06Tlz/2cTwe/1e1d5X3XWqRfvew5O7qA2z
UcB88q7nyM7DLzg5ZfiSN/x30rq3izxZonROS2qre4bx5oXr9tgHOaWTGqvIWWUU
HZjAMBQzHKBkcTAAtPEVPYIxQAMCAwEAAaMaMBgwCQYDVR0TBAIwADALBgNVHQ8E
BAMCBeAwDQYJKoZIhvcNAQELBQADgYEAlbvr8aUi2jUPC1ArTX7vFbEOh07e28yf
gBLDsgibBU5RCf2BARXrq6Ma0RtMO/8A58+E2WFgEOYnDA9hi6t0Wz/+vvw4ZbEL
rJ6+21V4INTZncrFoZPlmnU0mAI6zIXcK7YWWeDbzmJ5QWcra3Z8w9W76NGo0Mxd
ZOmwDHsbMT0=
-----END CERTIFICATE-----

Server certificate : (self signed for test purposes only)
-----BEGIN RSA PRIVATE KEY-----
MIICWwIBAAKBgQCtrYAeA4UWdecS0BHQZzDQE2XHOv4+2DAvj2B4qCs3LALxbTpO
XP/ZxPB7/V7V3lfddapF+97Dk7uoDbNRwHzyrufIzsMvODll+JI3/HfSureLPFmi
dE5Laqt7hvHmhev22Ac5pZMaq8hZZRQdmMAwFDMcoGRxMAC08RU9gjFAAwIDAQAB
AoGAa9eN9BUPcLDpOL9GMHfC0ZEZ46n38SHgWcfXiVQy8WE8upiuWbZZj+k/wl3D
o10ahgLoJeiNnDeBkIphTGQPsMN4sSDTCCqkSUbh3OFRWPJhbacYxB6v+EzSbjsW
KVOrJYIcrfd27DWca4lP7CSYVG4mMyf8WxMq+7yDl/i/DHECQQDZlFRBSpaJgJlh
pyKaOeqhofSx8fbQ8ZEWO8SpvsvnZWFlX5nwGKhou0+BWzzw+RuYn2g2zW+/TQbs
Z5TYmNfNAkEAzFiY8SRxl9up4b9DT+ip9WmvGModWuQZzrNo7GVXE05ft0/dr3vb
h5rNVeEhPyHzhAC0VfcpM5EmXY9Lzq4HDwJAPGczZNTiKdYrvrJ5prtBahHeLrBv
xt3pW3TISh+P1/OLMGtz/vW5eynVDqJXrsZZZOuMMPqjdUZbD7JJ3j/AWQJAeOOV
H1uzA+bE+rM+7CA1o+Tw3FZ9xU6aYelLkkZlnBtKjtTlzjIU7oVLIpo8I/A+8wHD
rfO5YTNmogTrEQttSwJAKc/Uqe95dQmvL7Qm0RptFKbXTRhR/bnoOIJgLK12J27F
9P/Wq8n3mizWOx+GeOeWJ0VZ2jgxBfZ8+IBPioORmQ==
-----END RSA PRIVATE KEY-----

-----BEGIN CERTIFICATE-----
MIICRDCCAa2gAwIBAgIJAIutIKrV1gxNMA0GCSqGSIb3DQEBCwUAMFYxCzAJBgNV
BAYTAkNBMQswCQYDVQQIDAJRQzEMMAoGA1UEBwwDTVRMMQ0wCwYDVQQKDARBVEVL
MQ0wCwYDVQQLDARBVEVLMQ4wDAYDVQQDDAVBVEVLMjAeFw0yMTAyMDQxNDU3MDBa
Fw0yNjAyMDQxNDU3MDBaMFYxCzAJBgNVBAYTAkNBMQswCQYDVQQIDAJRQzEMMAoG
A1UEBwwDTVRMMQ0wCwYDVQQKDARBVEVLMQ0wCwYDVQQLDARBVEVLMQ4wDAYDVQQD
DAVBVEVLMjCBnzANBgkqhkiG9w0BAQEFAAOBjQAwgYkCgYEAra2AHgOFFnXnEtAR
0Gcw0BNlxzr+PtgwL49geKgrNywC8W06Tlz/2cTwe/1e1d5X3XWqRfvew5O7qA2z
UcB88q7nyM7DLzg5ZfiSN/x30rq3izxZonROS2qre4bx5oXr9tgHOaWTGqvIWWUU
HZjAMBQzHKBkcTAAtPEVPYIxQAMCAwEAAaMaMBgwCQYDVR0TBAIwADALBgNVHQ8E
BAMCBeAwDQYJKoZIhvcNAQELBQADgYEAlbvr8aUi2jUPC1ArTX7vFbEOh07e28yf
gBLDsgibBU5RCf2BARXrq6Ma0RtMO/8A58+E2WFgEOYnDA9hi6t0Wz/+vvw4ZbEL
rJ6+21V4INTZncrFoZPlmnU0mAI6zIXcK7YWWeDbzmJ5QWcra3Z8w9W76NGo0Mxd
ZOmwDHsbMT0=
-----END CERTIFICATE-----

sdkconfig
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) Project Configuration
#
CONFIG_IDF_CMAKE=y
CONFIG_IDF_TARGET_ARCH_XTENSA=y
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_TARGET_ESP32=y
[... 1278 lines cut ...]
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options

I did not capture the packets yet I will capture the transaction and post it in next message

Thank you,
William

WilliamAtek, 2021-02-10:
I updated to last master idf and commented line 93 of the example to prevent multiple connection attemps.

I then started the esp32 and wireshark capture here is esp32 log : 

```
�ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6956
ho 0 tail 12 room 4
load:0x40078000,len:14336
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (29) boot: ESP-IDF v4.4-dev-4-g73db14240 2nd stage bootloader
I (29) boot: compile time 09:52:22
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 3, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=24ae8h (150248) map
I (166) esp_image: segment 1: paddr=00034b10 vaddr=3ffb0000 size=043c8h ( 17352) load
I (172) esp_image: segment 2: paddr=00038ee0 vaddr=40080000 size=07138h ( 28984) load
I (184) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=90398h (590744) map
I (393) esp_image: segment 4: paddr=000d03c0 vaddr=40087138 size=0e9b4h ( 59828) load
I (429) boot: Loaded app from partition at offset 0x10000
I (429) boot: Disabling RNG early entropy source...
I (440) cpu_start: Pro cpu up.
I (440) cpu_start: Starting app cpu, entry point is 0x4008125c
0x4008125c: call_start_cpu1 at C:/atek/dev/esp32/esp-idf/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
D (448) clk: RTC_SLOW_CLK calibration value: 3130470
I (460) cpu_start: Pro cpu start user code
I (460) cpu_start: cpu freq: 160000000
I (460) cpu_start: Application information:
I (464) cpu_start: Project name:     wpa2-enterprise
I (470) cpu_start: App version:      1
I (474) cpu_start: Compile time:     Feb 10 2021 10:20:04
I (480) cpu_start: ELF file SHA256:  f788d15051c973e2...
I (486) cpu_start: ESP-IDF:          v4.4-dev-4-g73db14240
V (493) memory_layout: reserved range is 0x3f424ad0 - 0x3f424b08
D (499) memory_layout: Checking 7 reserved memory ranges:
D (504) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (510) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb8238
D (517) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (523) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (530) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (536) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at C:/atek/dev/esp32/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (543) memory_layout: Reserved memory range 0x40080000 - 0x40095aec
0x40080000: _WindowOverflow4 at C:/atek/dev/esp32/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1730

D (549) memory_layout: Building list of available memory regions:
V (555) memory_layout: Examining memory region 0x3ffae000 - 0x3ffb0000
V (562) memory_layout: Start of region 0x3ffae000 - 0x3ffb0000 overlaps reserved 0x3ffae000 - 0x3ffae6e0
[... 1865 lines cut ...]
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
# CONFIG_TCPIP_TASK_AFFINITY_CPU1 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
CONFIG_SEMIHOSTFS_HOST_PATH_MAX_LEN=128
# End of deprecated options
```

Wireshark capture : https://1fichier.com/?8q6n2keztdbq1u1lcr0x

WilliamAtek, 2021-02-23:
For everyone having the same problem, **the problem comes from MbedTLS that doesn't support PEAPv0.**

To fix it go to **menuconfig > components config > Wpa supplicant > Unchek "Use mbedTLS crypto API"**

Then it will work with Windows, Linux and everything else like it should.

Please note that this problem seems purely related to mbedTLS as I had the same problem on openWRT and when I switched to openSSL the problem disappeared.

PunishedSnakePr, 2021-08-18:
@WilliamAtek : Thank u so much! That really did the trick!

closed 2021-09-01 by espressif-bot