espressif/esp-idf #1300: Reset but not reset?

chegewara, 2017-11-21:
I just found that this kind of reset does not reset internal tick counter. After reset time since start was not cleared. 
```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4456
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:12988
entry 0x40078d8c
W (91) boot: PRO CPU has been reset by WDT.
W (91) boot: WDT reset info: PRO CPU PC=0x40089204
0x40089204: rf_init at ??:?

W (91) boot: WDT reset info: APP CPU PC=0x4008b28a
0x4008b28a: noise_init at ??:?

```

igrr, 2017-12-08:
What was the last printed log timestamp before TG0WDT reset happened, was it less than 91?

The timestamp you see is somewhat higher than what i see, trying to reproduce this, but still much lower than the timestamp you should see after the application has started up.

For example, with flash configured as 40MHz DIO mode (which is the default), i get the following:

```
ets Jun  8 2016 00:22:57

rst:0x7 (TG0WDT_SYS_RESET),boot:0x3e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5552
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13712
entry 0x40079020
W (30) boot: PRO CPU has been reset by WDT.
W (30) boot: WDT reset info: PRO CPU PC=0x40008544
W (31) boot: WDT reset info: APP CPU PC=0x400d1a1a (waiti mode)
0x400d1a1a: esp_vApplicationIdleHook at /Users/ivan/e/esp-idf/components/esp32/./freertos_hooks.c:62
```

And then, when the application starts up:
```
I (272) cpu_start: Starting scheduler on PRO CPU.
```

So if the tick counter was preserved after reset, you would get a value higher than 272 (ms).

chegewara, 2017-12-08:
Timestamp was higher for sure because app has been running for at least few seconds. I dont remember exactly what triggered it but if i remeber it was something like that:
- application is REST web server,
- from web interface has been send some request to web server,
- for some reason, dont remember what was that, esp32 got reset

I opened this issue because web interface got information with time since app start and it was weird to me when i saw after reset this time value was not reseted. It was only one time when i observed this, so i cant guarantee it can be reproduced.

chegewara, 2017-12-08:
Im thinking right now, but this maybe is stupid because i dont know if it is possibleat all, is it possible that crash was just on one core and the other core has been not reset? Then gettimeofday() after reset was invoked on "health" core?

igrr, 2017-12-08:
`TG0WDT_SYS_RESET` is a system reset, i.e. it resets both CPUs and all digital peripherals. Only the RTC is not reset.

Log timestamps are obtained from two sources: before FreeRTOS is running, we use CPU internal cycle count register (CCOUNT), once FreeRTOS has started, we use FreeRTOS tick count. So log timestamps should definitely be reset after TG0WDT_SYS_RESET. There's always a room for a bug, but based on your log it doesn't seem like the reset has not happened.

You have also mentioned the time shown in the web interface of the app. As mentioned above, RTC is not reset by TG0WDT_SYS_RESET, so if you have RTC timekeeping enabled (this is the default), and the application is using gettimeofday call, then the time obtained from gettimeofday will be preserved after reset.

chegewara, 2017-12-08:
Then it is possible its not a bug at all, just time was preserved. If my understanding is accurate then feel free to close this and thanks for investigate and explain it.

closed 2017-12-29 by chegewara