espressif/esp-idf #1301: DPORT_UART0_CLK_EN not defined in dport_reg.h

CharlesSeartech, 2017-11-22:
Hi,

I'm using UART1 for the console output so that I can use UART0 for my application. But when I change the setting in Component config → ESP32-specific → UART peripheral to use for console output (0-1) (UART1) to UART1 and rebuild the code I get an error 'DPORT_UART0_CLK_EN' undeclared:
esp-idf/components/esp32/clk.c:190:31: error: 'DPORT_UART0_CLK_EN' undeclared (first use in this function)

The function responsible is esp_perip_clk_init(). I can't find the setting for DPORT_UART0_CLK_EN in the dport_reg.h file or "ESP32 Technical Reference Manual".
Can you please indicate what the bit definition should be or must it be removed?

Regards,
Charles

negativekelvin, 2017-11-22:
https://github.com/espressif/esp-idf/blob/80eb3b6ddbc218bcce830771730f902480007578/components/soc/esp32/include/soc/dport_reg.h#L975

CharlesSeartech, 2017-11-22:
Sorry but the current code is as follows:
        common_perip_clk = DPORT_WDG_CLK_EN |
                              DPORT_I2S0_CLK_EN |
#if CONFIG_CONSOLE_UART_NUM != 0
                              DPORT_UART0_CLK_EN |
#endif
#if CONFIG_CONSOLE_UART_NUM != 1
                              DPORT_UART1_CLK_EN |
#endif
#if CONFIG_CONSOLE_UART_NUM != 2
                              DPORT_UART2_CLK_EN |
#endif
so it wants to disable the clocks of the UART's not used for console output.

igrr, 2017-11-22:
Sorry, missed that while reviewing #1217. Submitted the fix internally.

closed 2017-12-14 by igrr