espressif/esp-idf #1302: error: implicit declaration of function "clock_gettime"

Gfast2, 2017-11-22:
Hi ESP-IDF,

Here I am! Right now I'm writing some code that wanna use this POSIX function. But my ESP-IDF compiler won't let my code go through. 

I've checked out, in the whole repo of EPS-IDF, component `coap` is using this function. [`components/coap/libcoap/src/coap_time.c`](https://github.com/obgm/libcoap/blob/6468887a12666f88b8704d797fc176cd4f40ee4c/src/coap_time.c).
And this function is do mentioned in `components/newlib/include/time.h`:
[`int _EXFUN(clock_gettime, (clockid_t clock_id, struct timespec *tp));`](https://github.com/espressif/esp-idf/blob/master/components/newlib/include/time.h#L160).
But I just can not let my holy code base pass through. 📟 

Here I backed a minimal example:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"

#include <time.h>
#include <sys/time.h>
#include <unistd.h>

void app_main()
{   
    struct timespec ts;
    clock_gettime(CLOCK_REALTIME, &ts);
    printf("Finished.");
}
```

Error Log:

```
CC build/main/hello_world_main.o
/home/gfast2/workspace/clock_gettime/main/./hello_world_main.c: In function 'app_main':
/home/gfast2/workspace/clock_gettime/main/./hello_world_main.c:37:5: error: implicit declaration of function 'clock_gettime' [-Werror=implicit-function-declaration]
     clock_gettime(CLOCK_REALTIME, &ts);
     ^
cc1: some warnings being treated as errors
/home/gfast2/esp/esp-idf/make/component_wrapper.mk:242: recipe for target 'hello_world_main.o' failed
make[1]: *** [hello_world_main.o] Error 1
/home/gfast2/esp/esp-idf/make/project.mk:435: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

```

igrr, 2017-11-22:
This function is declared only if _POSIX_TIMERS is set to 1. IDF doesn't
have support for POSIX timers, so the function isn't implemented anywhere.
Coap is using gettimeofday instead. Perhaps you can also use gettimeofday
in your code.

Gfast2, 2017-11-22:
Hi @igrr ,

Thanks for your declearation! I'll give `gettimeofday` a try.

But Before I close this issue, I'd very pleased to know what is file `components/newlib/include/sys/feature.h` about? because in this file, `_POSIX_TIMERS` is mentioned and seems be set to `1`

Cheers

Gfast2

Gfast2, 2017-11-22:
If someone like me need time in `timespec` structure, but have to get the time from ESP32 dure to the situation here mentioned (by `gettimeofday`). Check out the conversion in between [**here**](https://gist.github.com/enjoylife/4384268)

closed 2017-11-30 by Gfast2