espressif/esp-idf #1304: BLE r_rw_schedule crash

jkhax0r, 2017-11-22:
I've been having the issue that was posted by someone else over on the forums: https://www.esp32.com/viewtopic.php?f=13&t=3513&p=16263&hilit=rw_schedule#p16263.  I too am on a fairly recent esp-idf copy within the last 2 weeks or so.  

Seems I can add to this post though since I have been able to figure out how to reproduce this 100% of the time.   As it turns out if I set the connection parameters on my device to min/max/latency/timeout=6/6/0/1000, I can get it to crash virtually every time when the 2nd or 3rd BLE peripheral connects.  

I suspect the issue is with the connection interval being set to 6 on multiple devices.  Previously I had it at 10 on 2 of my devices and 50 on the others, and it would happen rarely.  Now with them all being set to 6, it is almost 100%.

I have worked around by extending the connection interval on all my devices, but this is not ideal and I have a suspicion the bug may still happen (just 1 in 10000 times now).  It seemed to get progressively better the more I extended the parameters.   

Please let me know if I can provide any other information to help reproduce the issue.

TianHao-Espressif, 2017-11-23:
@jkhax0r , thanks for your information. We'll debug it. 
So it is related to the connection interval? Larger connection interval will reduce the problem happen?

Yulong-espressif, 2017-12-15:
@jkhax0r We have fix this bug, please update the master, thanks.

closed 2017-12-21 by FayeY