espressif/esp-idf #1308: Brownout during wifi startup

tehKaiN, 2017-11-23:


projectgus, 2017-11-23:
Depending on what version of IDF the AT firmware & your other test firmware is compiled with, there was a bug fixed in 27b52dfd18dbc489b012848fef549916e4f58272 which reduces the peak RF calibration current. I believe Arduino-ESP32 was only updated to this IDF version last night (espressif/arduino-esp32@81a9c45a1ef5833e80c9235ff393a7641c669a94) so if your test firmware was using Arduino then it would have had a higher calibration current.

One other difference which may be significant is that the AT firmware runs at 80MHz, which has lower overall power consumption than the higher CPU clock speeds.

Despite the bug situation, I would encourage you to check if your module design has enough capacitance and add more if it does not.

projectgus, 2017-11-30:
Hi @tehKaiN , I'm going to close this issue but please reopen if you think the above explanation doesn't cover the brownout you were seeing.

Angus

closed 2017-11-30 by projectgus