espressif/esp-idf #1311: smart config security

tdesmet, 2017-11-24:
I am looking at smart config and wondering how secure it is.

As it seems to me the ssid and password is send cleartext 
Is it possible to encrypt the data using an aes key?

jack0c, 2018-01-02:
It unidirectional communication, you can use fix Key to encrypt the SSID and Password. Smart config is not a standard protocol, blufi will be better than smart config, you can find an example for blufi.

b1gtang, 2018-02-24:
I suggest that the official can add AES encryption in the smart config library, the user can call the API, in order to avoid code redundancy, this part of the code should not be achieved by the users.

jack0c, 2018-03-22:
@b1gtang Yes, we are considering make more part of code in smart config open source.

SteveOfTheStow, 2018-03-23:
+1 for baking encryption into SmartConfig

kyrulkamal, 2018-05-16:
Any update regarding this issue?

thefat32, 2020-09-16:
As far as I understand, being an unidirectional comm. The best security you could get is a hardcoded key in your mobile app and your ESP firmware. This is not secure at all. It's a bit harder to crack than clear text. But not secure. To make a protocol secure you need to implement a proper handshake between involved parts.

zhangyanjiaoesp, 2020-10-27:
@tdesmet @b1gtang @SteveOfTheStow @kyrulkamal @thefat32 

We will add aes encryption for smartconfig in v4.3, which will called esp-touch-v2

closed 2021-01-15 by espressif-bot

Olofforsberg, 2022-01-28:
Hello, the esp-touch-v2 only works for esp32-c3, will you also enable it for esp32, esp32-s2?