espressif/esp-idf #1313: Errors on latest Toolchain and ESP-IDF

michal037, 2017-11-25:
Hello,
I follow step by step this guide:
https://esp-idf.readthedocs.io/en/latest/get-started/index.html

I downloaded:
https://dl.espressif.com/dl/esp32_win32_msys2_environment_and_toolchain-20171123.zip
https://github.com/espressif/esp-idf/releases/download/v2.1.1/esp-idf-v2.1.1.zip

This is my ESP-IDF PATH:
"C:\msys32\home\Michal\esp\esp-idf" <- created: "export_idf_path.sh" in "C:/msys32/etc/profile.d/"

Then I started the mingw32.exe app.

cd ~/esp
cp -r $IDF_PATH/examples/get-started/hello_world .
cd ~/esp/hello_world
make menuconfig  -> "Change Serial flasher config" -> "Default serial port" -> to "COM8"
Save -> Exit

make flash

**At this point I have errors with _gthr-default.h_**

log file:
https://raw.githubusercontent.com/michal037/VariousDestiny/master/esp-idf-logs

Am I doing something wrong? Is there a problem with this file?

igrr, 2017-11-25:
For v2.1.1, please use the corresponding version of  documentation: https://esp-idf.readthedocs.io/en/v2.1.1/get-started/index.html

'latest' corresponds to the master branch of ESP-IDF.

michal037, 2017-11-25:
I notice differences, I will download older toolchain. 
Thanks @igrr

closed 2017-11-25 by michal037