espressif/esp-idf #1314: Bad documentation

Arthedian, 2017-11-25:
Hello, 
I was reading through your documentation[http://esp-idf.readthedocs.io/en/v2.1.1/api-reference/bluetooth/esp_gattc.html](documentation). and function `esp_ble_gattc_get_characteristic` does not longer exist. Function `esp_ble_gattc_get_char_by_uuid` is existing instead. Can you fix it please?

Yulong-espressif, 2017-11-25:
@Arthedian Thank you very much, this documentation is write for the v2.1.1 version, we will release a version of documentation for v3.0 later.

igrr, 2017-11-25:
Please see http://esp-idf.readthedocs.io/en/latest/api-reference/bluetooth/esp_gattc.html#_CPPv230esp_ble_gattc_get_char_by_uuid13esp_gatt_if_t8uint16_t8uint16_t8uint16_t13esp_bt_uuid_tP21esp_gattc_char_elem_tP8uint16_t

(v2.1.1 in the documentation URL indicates that this is documentation for release 2.1.1. If you are using master branch, check 'latest' documentation version).

closed 2017-11-25 by igrr