espressif/esp-idf #1316: Enhancement: heap_caps_check_integrity could return failed addresses

nkolban, 2017-11-25:
When we call heap_caps_check_integrity() we are returned a boolean to indicate that an integrity check failed and we have the opportunity to log diagnostics to the console.  While this is great, this issue suggest an enhancement.

In a recent bug, I found that some code I had was writing after free to storage it didn't own.  This was caught nicely with the `heap_caps_check_integrity()` call (thank you for that).  However, the problem I then faced was that the I was told the following:

```
CORRUPT HEAP: Invalid data at 0x3ffd0a88. Expected 0xfefefefe got 0xfefefefd
```
What I now needed to do was determine what storage was once `0x3ffd0a88` and had been released.  I have heap trace on and was capturing allocate and release data but in my logic which called `heap_caps_check_integrity()`  I don't know that address (i.e. it was logged on the screen and not returned to the caller).  As such, I couldn't now algorithmically walk through my heap trace to determine who allocated and who released it.  What I had to do was log ALL my heap trace entries, add them to a spreadsheet and then sort them and manually look for my records.  This resulted in:

```
0x3ffd0a2c | 0x3ffd0ba0 | 372 | 40086c46 40129630 | 400865f4 40086efb
```

And then I had to decode the allocation and release addresses.  While all of this is of course manually do-able, if I'd had the address of the error (`0x3ffd0a88`) I could have automated this in my diagnostics.

projectgus, 2021-01-22:
Sorry noone ever commented on this before, @nkolban . I think I misread this request originally when first requested, but you want is:

* If heap tracing is on and heap corruption is detected, try to walk the current trace and dump to the console the allocation responsible

We could indeed do this by adding an API that returns when the heap is corrupt. The problem is that if the heap is corrupt we usually are about to crash.

SoucheSouche, 2022-11-14:
@nkolban,

I had a look in the implementation of `heap_caps_check_integrity()` following the issue you reported.

The function calls internally a walker that jumps from free blocks to free blocks of memory and check for consistency and corruption. This walker does not return on first inconsistency/corruption found but decrease a status value and print diagnostics when the inconsistency/corruption was detected instead.

I agree that returning the address where the corruption was detected could be beneficial in you case but it also implies that the `heap_caps_check_integrity()` would return on first issue found and leave the rest of the heap unchecked which goes against the original idea that drove the implementation of this integrity check feature. 

For this reason, there won't be any change made to the API of the heap component for now .

closed 2022-11-16 by Alvin1Zhang