espressif/esp-idf #1317: Heap trace and WiFi / networking interference

nkolban, 2017-11-25:
I was debugging some heap problems and had my heap trace set to 10000 entries using a WROVER with 4MBytes.   I was running a heavy soak test against my ESP32 forming connections, sending data, receiving a response and closing the connections all in a tight loop.  What I found was that eventually my program would "lock up" and I would receive errors about watchdogs being fired in IDLE.  The strange thing was that when I slowed my program down by inserting delays in the network client running on my PC, I would still eventually lock up after about 70-80 connections.  As I dug and dug into this, I came to the conclusion that some resource must be being "used up".  After much fruitless testing, I disabled heap memory tracing and I was able to run indefinitely.  This makes me feel that there was something "odd" in that area.  My best bet is that something isn't quite right if we run out of storage space for heap trace records.   I will research more ... and try and create a stand-alone test.  But I wanted to raise this issue as a tracker in case others might see something or in case it might spark something in the minds of the heap trace developers.

projectgus, 2017-11-30:
Hi @nkolban ,

Were the heap trace records placed in external PSRAM? If you run tracing with an internal buffer (smaller number of entries), does this work? How about the smaller number of entries when placed in external PSRAM?

Angus

nkolban, 2017-12-14:
I'm afraid I don't have an immediate re-create for this.  Closing.

closed 2017-12-14 by nkolban