espressif/esp-idf #1322: Bluetooth Controller: host_recv_pkt_cb called from ISR context

mringwal, 2017-11-27:
Hi. The Bluetooth Controller still/again does callbacks from ISR context. 

The Bluedroid adapter in esp-idf/components/bt/bluedroid/hci/hci_hal_h4.c does not expect calls from ISR, which can lead to any kind of weird behaviour as it uses the non-ISR variants of the FreeRTOS functions like mutexes (in fixed_queue) and xQueueSend to notify another thread. 

BTstack has a check if it's called from ISR and asserts (by doing a printf in report_recv_called_from_isr), see here: https://gist.github.com/gtalusan/486677ac14cebbc16e75a6bd40306fbd

Snippet:
0x400e47e2: report_recv_called_from_isr at /Users/george/esp/esp-idf/components/btstack/./main.c:98
0x400e47f3: host_recv_pkt_cb at /Users/george/esp/esp-idf/components/btstack/./main.c:122
0x400f2b80: vhci_send at ??:?
0x400ed012: r_eif_send at ??:?
0x400ed591: hci_tx_start at hci_tl.c:?
0x400ed77a: r_hci_tl_send at ??:?
0x400838ca: r_assert_err at ??:?
0x40086407: r_rwbtdm_isr_wrapper at intc.c:?
0x40081e4a: _xt_lowint1 at /Users/george/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105
0x400f7384: avdtp_unpack_service_capabilities at /Users/george/esp/esp-idf/components/btstack/src/classic/avdtp_util.c:855

In this snippet, vhci_send is interrupted and the interrupt calls host_recv_pkt_cb from ISR context.

According to the user, this happens with the current esp-idf master. 

Please give high priority to this, as it can randomly cause other bugs that will be hard to analyze and fix.

TianHao-Espressif, 2017-12-08:
@mringwal , thank your for this bug report.

This problem is not due to "vhci_send is interrupted and the interrupt calls host_recv_pkt_cb from ISR context." An assert happened prior to vhci send. 

Because of "ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_acl.c at line 1900",  then r_assert_err happened, r_assert_err will call vhci_send to send the error msg to host. so vhci send called in ISR. But only r_assert_err may call vhci send in ISR, in normal scenario, vhci send never called in ISR. So this problem's root cause is  "ASSERT_ERR((EM_BT_RXDESC_OFFSET + ld_env.curr_rxdesc_index * REG_EM_BT_RXDESC_SIZE) == bt_et_currentrxdescptr_currentrxdescptr_getf()), in ld_acl.c at line 1900". 

We'll debug this problem, could you provide the demo code?

mringwal, 2017-12-08:
@TianHao-Espressif thanks for looking at the backtrace. Without the code for the assert, it's hard to guess what's causing it.

It was caused by a user testing BTstack on esp32 - let's assume latest versions of esp-idf and BTstack. It was from this bug report: https://github.com/bluekitchen/btstack/issues/91

The user was testing BTstack's a2dp_source_demo. You can follow the btstack esp32 setup (https://github.com/bluekitchen/btstack/tree/develop/port/esp32), and then compile & run the a2dp_source_demo.

Would it be safe for BTstack (and Bluedroid) to ignore callbacks on ISR context?
At the moment, BTstack tries a printf which causes a crash and backlog: https://github.com/bluekitchen/btstack/blob/develop/port/esp32/components/btstack/main.c#L122

mringwal, 2017-12-22:
Any idea what went wrong or if error events could be send from a thread? Thanks!

TianHao-Espressif, 2017-12-22:
@mringwal ，when error event sent  from controller to host, it means controller is in a crash status.
The crash may be caused by software or by hardware. Maybe it's hard to reset bluetooth for work.
Later, I will modify something on this problem. For now, resolve the crash(ASSERT) is prior to modify error event throw context.

mringwal, 2017-12-22:
Ok, so you suggest to treat "event from ISR" similar to a regular Hardware Error event.

How can I reset the Bluetooth Controller without a system reset? Would sending an HCI Reset work?

TianHao-Espressif, 2018-01-04:
@mringwal ， for now, due to the known issues, cannot send HCI RESET to controller when exception happened on controller. Later, we will do something different.

mringwal, 2018-01-04:
Thanks for confirming that sending HCI Reset would not work. Please let me know when there's a way to reset the Bluetooth Controller - I'll update BTstack then to treat this like a Hardware Error. Maybe the Bluetooth Controller could just emit a Hardware Error event even.

TianHao-Espressif, 2018-01-05:
@mringwal , I'm sorry for this problem. When I finish the fix, I'll tell you.

TianHao-Espressif, 2018-04-03:
@mringwal , I have remove "event from ISR" things on the latset master branch. And normally, you can do "send HCI reset" when something is going wrong.

mringwal, 2018-04-03:
@TianHao-Espressif thanks! does it mean that if there's a problem in the Controller, it will at least emit an Hardware Event error from thread context, which allows to send HCI Reset and start over without resetting the whole ESP32?

TianHao-Espressif, 2018-04-03:
@mringwal ， normally, it is.

vvhh2002, 2020-06-24:
This problem still exists in ESP-IDF v4.1-beta2-141-g84b51781c.
I use the btstack hfp_hs example code. when the SCO connection is established, the following error appears:

```

I (73122) BTS: hfp_hf.c.625: HFP: sending hci_accept_connection_request, sco_voice_setting 0x43
I (73252) BTS: hci.c.2247: Synchronous Connection Complete (status=0) XX:XX:XX:XX:XX:XX
I (73252) BTS: hfp.c.699: eSCO Connection established. 

I (73262) BTS: hfp.c.707: sco_handle 0x181, address E0:33:8E:A8:81:35, transmission_interval 12 slots, retransmission_interval 2 slots,  rx_packet_length 60 bytes, tx_packet_length 60 bytes, air_mode 0x 3 (0x02 == CVSD)

Audio connection established with SCO handle 0x0181.
Using mSBC codec.
I (732ASSERT_PARAM(1 0), in rwbt.c at line 314
82) sco_demo: data: size 63
I (73292) sco_demo: 81 01 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
I (73292) sco_demo: SCO CRC Error: 2 - data: 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x400859d0  PS      : 0x00060034  A0      : 0x8008a4f8  A1      : 0x3ffbe290  
0x400859d0: r_assert_param at ??:?

A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffbdbb4  A7      : 0x60008050  A8      : 0x800859cd  A9      : 0x3ffbe270  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40085905  LEND    : 0x4008590c  LCOUNT  : 0x00000000  
0x40085905: r_assert_param at ??:?

0x4008590c: r_assert_param at ??:?

Core 0 was running in ISR context:
EPC1    : 0x40132106  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400859d0
0x40132106: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:486

0x400859d0: r_assert_param at ??:?

ELF file SHA256: 630e8609643bb6a6

Backtrace: 0x400859cd:0x3ffbe290 0x4008a4f5:0x3ffbe2b0 0x4008a913:0x3ffbe2d0 0x40084df5:0x3ffbe2f0 0x40132103:0x3ffca9c0 0x400d668b:0x3ffca9e0 0x40095121:0x3ffcaa00 0x40092e81:0x3ffcaa20
0x400859cd: r_assert_param at ??:?

0x4008a4f5: r_rwbt_isr at ??:?

0x4008a913: r_rwbtdm_isr_wrapper at intc.c:?

0x40084df5: _xt_lowint1 at /esp/esp-idf/components/freertos/xtensa_vectors.S:1153

0x40132103: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d668b: esp_vApplicationIdleHook at /esp/esp-idf/components/esp_common/src/freertos_hooks.c:86

0x40095121: prvIdleTask at /esp/esp-idf/components/freertos/tasks.c:4691

0x40092e81: vPortTaskWrapper at /esp/esp-idf/components/freertos/port.c:436

Core 1 register dump:
PC      : 0x40132106  PS      : 0x00060234  A0      : 0x800d668e  A1      : 0x3ffcafc0  
0x40132106: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:486

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x80000001  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d60aa  A9      : 0x3ffcaf90  
A10     : 0x00000000  A11     : 0x00060623  A12     : 0x00060620  A13     : 0x3ffc2500  
A14     : 0x3ffbbdb0  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

ELF file SHA256: 630e8609643bb6a6

Backtrace: 0x40132103:0x3ffcafc0 0x400d668b:0x3ffcafe0 0x40095121:0x3ffcb000 0x40092e81:0x3ffcb020
0x40132103: esp_pm_impl_waiti at /esp/esp-idf/components/esp32/pm_esp32.c:484

0x400d668b: esp_vApplicationIdleHook at /esp/esp-idf/components/esp_common/src/freertos_hooks.c:86

0x40095121: prvIdleTask at /esp/esp-idf/components/freertos/tasks.c:4691

0x40092e81: vPortTaskWrapper at /esp/esp-idf/components/freertos/port.c:436

Rebooting...
ets Jun  8 2016 00:22:57

```