espressif/esp-idf #1325: I2C issue

chegewara, 2017-11-27:
Im not sure if this an idf issue or its my app, but first I2C init works, then i deinit I2C and then i init it again and get app crash. This is log output:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40088a7d  PS      : 0x00050031  A0      : 0x40083018  A1      : 0x3ffc05e0  
0x40088a7d: i2c_isr_handler_default at /home/esp32/esp/esp-idf/components/driver/./i2c.c:932

0x40083018: _xt_lowint1 at /home/esp32/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

A2      : 0x3ffbcb88  A3      : 0x00000000  A4      : 0x3ffc40d8  A5      : 0x00002a00  
A6      : 0x00000000  A7      : 0x1b34218e  A8      : 0x00000000  A9      : 0x3ffd5620  
A10     : 0x00000003  A11     : 0x00060023  A12     : 0x1b34218e  A13     : 0x3ffd56c0  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002c  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40088a7d:0x3ffc05e0 0x40083015:0x3ffc0610 0x400d29b1:0x00000000
0x40088a7d: i2c_isr_handler_default at /home/esp32/esp/esp-idf/components/driver/./i2c.c:932

0x40083015: _xt_lowint1 at /home/esp32/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

0x400d29b1: esp_vApplicationIdleHook at /home/esp32/esp/esp-idf/components/esp32/./freertos_hooks.c:85
```
and output spit by gdb after crash:
```
0x40088a7d in i2c_isr_handler_default (arg=0x3ffbcb88) at /home/esp32/esp/esp-idf/components/driver/./i2c.c:352
352	    uint32_t status = I2C[i2c_num]->int_status.val;
```

koobest, 2017-11-29:
Hi, chegewara. we are working on is  problen now. it caused by the issue of eventGroup. we will update the I2C driver soon.

chegewara, 2017-11-29:
There is few I2C issue reports so i will close this one.

closed 2017-11-29 by chegewara