espressif/esp-idf #1327: Error in ULP macros

No-0n3, 2017-11-28:
The global function "SOC_REG_TO_ULP_PERIPH_SEL" in "ulp.h" can't set the values for the instruction macros using it to set "periph_sel". I get the message "error: initializer element is not constant" when compiling. Log is added below.

> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:318:19: error: initializer element is not constant
>      .periph_sel = SOC_REG_TO_ULP_PERIPH_SEL(reg), \
>                    ^
> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:345:39: note: in expansion of macro 'I_WR_REG'
>  #define I_WR_REG_BIT(reg, shift, val) I_WR_REG(reg, shift, shift, val)
>                                        ^
> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:381:5: note: in expansion of macro 'I_WR_REG_BIT'
>      I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0)
>      ^
> C:/msys32/home/<user>/esp/projects/ulp/main/main.c:33:2: note: in expansion of macro 'I_END'
>   I_END(),
>   ^
> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:318:19: note: (near initialization for 'program[23].wr_reg.periph_sel')
>      .periph_sel = SOC_REG_TO_ULP_PERIPH_SEL(reg), \
>                    ^
> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:345:39: note: in expansion of macro 'I_WR_REG'
>  #define I_WR_REG_BIT(reg, shift, val) I_WR_REG(reg, shift, shift, val)
>                                        ^
> C:/msys32/home/<user>/esp/esp-idf/components/ulp/include/esp32/ulp.h:381:5: note: in expansion of macro 'I_WR_REG_BIT'
>      I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0)
>      ^
> C:/msys32/home/<user>/esp/projects/ulp/main/main.c:33:2: note: in expansion of macro 'I_END'
>   I_END(),
>   ^

Have tried to make the "periph_sel" member to an argument of the macro who uses it for example:

> #define I_END(periph_sel_) \
>     I_WR_REG_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN_S, 0, periph_sel_)

and using the defines
> #define RD_REG_PERIPH_RTC_CNTL 0    /*!< Identifier of RTC_CNTL peripheral for RD_REG and WR_REG instructions */
> #define RD_REG_PERIPH_RTC_IO   1    /*!< Identifier of RTC_IO peripheral for RD_REG and WR_REG instructions */
> #define RD_REG_PERIPH_SENS     2    /*!< Identifier of SARADC peripheral for RD_REG and WR_REG instructions */

directly as an argument and that worked. Using the function didn't work because it isn't a constant.
> I_END(RD_REG_PERIPH_RTC_CNTL),

ginkgm, 2017-12-01:
The macro works well and passes the build in my environment. Please refers to the usage of I_END() in example: `examples/system/deep_sleep/main/deep_sleep_example_main.c`.

If your code is different, please paste it here to help us know what happened.

No-0n3, 2017-12-01:
From ulp.h on my computer:  
> 
> /**
>  * Map SoC peripheral register to periph_sel field of RD_REG and WR_REG
>  * instructions.
>  *
>  * @param reg peripheral register in RTC_CNTL_, RTC_IO_, SENS_ peripherals.
>  * @return periph_sel value for the peripheral to which this register belongs.
>  */
> static inline uint32_t SOC_REG_TO_ULP_PERIPH_SEL(uint32_t reg) {
>     uint32_t ret = 3;
>     if (reg < DR_REG_RTCCNTL_BASE) {
>         assert(0 && "invalid register base");
>     } else if (reg < DR_REG_RTCIO_BASE) {
>         ret = RD_REG_PERIPH_RTC_CNTL;
>     } else if (reg < DR_REG_SENS_BASE) {
>         ret = RD_REG_PERIPH_RTC_IO;
>     } else if (reg < DR_REG_RTCMEM0_BASE){
>         ret = RD_REG_PERIPH_SENS;
>     } else {
>         assert(0 && "invalid register base");
>     }
>     return ret;
> }

The error I see is common when compiling in gnu99, I see that the newest toolchain i download is using it.  c99, c89 and gnu99 don't allow initialization of globals with non-constant data.  In c11 it was allowed.

igrr, 2017-12-01:
Could you please post the code snippet you are trying to compile? gist.github.com is a great place for code snippets.

No-0n3, 2017-12-01:
File i'm trying to compile with log output: (This is another file but the same error occurs)  
https://gist.github.com/No-0n3/83311cdf491bb850d3a9bf1d63bc359c

igrr, 2017-12-01:
You need to place definition of `program` into local function scope. It will not work in global scope.

Also there are other problems in your code: I_WR_REG takes 4 arguments, not 5. And bit arguments of I_WR_REG are bit numbers, not bit masks (e.g. `SENS_HALL_PHASE_FORCE_S` instead of `SENS_HALL_PHASE_FORCE`).

No-0n3, 2017-12-01:
Thank you, compilation succeed. Can you update the page http://esp-idf.readthedocs.io/en/latest/api-guides/ulp_macros.html with the information that the macros need to be local and can not be global. Does not say in the docs. It would be great if you update and mention it.

If you have time to answer, when ADC measurements are taken from ULP, is the data int16_t and stored in the first 16bit of uint32 or is it uint16_t? Just so I can typecast the values correctly in main cpu.

igrr, 2017-12-01:
Ok, the example will be updated to mention this. Thank you for pointing out this issue.

ULP registers are 16-bit, and ADC measurement is stored in the lower 12 bits of the 16-bit register. When writing from ULP register to RTC memory, 16 bits of the register are stored in the lower half of the 32-bit memory word. Higher 32 bits are written with a value which depends on the address of the ULP instruction being executed. See http://esp-idf.readthedocs.io/en/latest/api-guides/ulp_instruction_set.html#st-store-data-to-the-memory.

No-0n3, 2017-12-01:
That I know, my question is if the ADC value is signed 12bit value or unsigned. When reading the hall sensor i can get a negative value but how do i do the same calculation as in hall_sensor_get_value, and it uses adc_get_raw which returns an int, if the value is unsigned... the calculation is (sensor_vp1-sensor_vp0)-(sensor_vn1-sensor_vn0).

closed 2017-12-13 by igrr