espressif/esp-idf #1328: Constant Restart when trying to make flash monitor IoT code

lawlhwut, 2017-11-29:
Hi,

I'm working on a code that connects to wifi, gathers information from sensors, formats into a unit, compiles an MQTT message, and sends them to AWS IoT. However, when I try to `make flash monitor`, it constantly restarts/gets stuck in a reboot with this message:

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5732
load:0x40078000,len:0
load:0x40078000,len:13804
entry 0x40079030
I (68) boot: Detected ESP32
I (32) boot: ESP-IDF v3.0-dev-1289-g6e1453e8-dirty 2nd stage bootloader
I (32) boot: compile time 22:05:42
I (33) boot: Enabling RNG early entropy source...
I (39) boot: SPI Speed      : 40MHz
I (43) boot: SPI Mode       : DIO
I (47) boot: SPI Flash Size : 4MB
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x19098 (102
552) map
I (133) esp_image: segment 1: paddr=0x000290c0 vaddr=0x3ffb0000 size=0x031d4 ( 1
2756) load
I (138) esp_image: segment 2: paddr=0x0002c29c vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _iram_start at C:/Users/Trey/esp/esp-idf/components/freertos/xtensa_
vectors.S:1685

I (140) esp_image: segment 3: paddr=0x0002c6a4 vaddr=0x40080400 size=0x0396c ( 1
4700) load
I (155) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x73a14 (47
3620) map
0x400d0018: _stext at ??:?

I (322) esp_image: segment 5: paddr=0x000a3a34 vaddr=0x40083d6c size=0x0da5c ( 5
5900) load
0x40083d6c: spi_flash_disable_interrupts_caches_and_other_cpu at C:/Users/Trey/e
sp/esp-idf/components/spi_flash/cache_utils.c:95

I (345) esp_image: segment 6: paddr=0x000b1498 vaddr=0x400c0000 size=0x00000 (
   0) load
I (356) boot: Loaded app from partition at offset 0x10000
I (356) boot: Disabling RNG early entropy source...
I (357) cpu_start: Pro cpu up.
I (361) cpu_start: Starting app cpu, entry point is 0x40080fc8
0x40080fc8: call_start_cpu1 at C:/Users/Trey/esp/esp-idf/components/esp32/cpu_st
art.c:222

I (351) cpu_start: App cpu up.
I (371) heap_init: Initializing. RAM available for dynamic allocation:
I (378) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (384) heap_init: At 3FFB9018 len 00026FE8 (155 KiB): DRAM
I (390) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (397) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (403) heap_init: At 400917C8 len 0000E838 (58 KiB): IRAM
I (409) cpu_start: Pro cpu start user code
I (91) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
W (94) wifi: wifi nvs_open fail ret=4353

E (94) wifi: wifi_init 1418 ret=4353
ESP_ERROR_CHECK failed: esp_err_t 0x1101 at 0x4010ad71
0x4010ad71: initialise_wifi at C:/Users/Trey/esp/subscribe_publishTestingCode/ma
in/subscribe_publish_sample.c:612 (discriminator 1)

file: "C:/Users/Trey/esp/subscribe_publishTestingCode/main/subscribe_publish_sam
ple.c" line 612
func: initialise_wifi
expression: esp_wifi_init(&cfg)

Backtrace: 0x40088b84:0x3ffbac30 0x40088ef1:0x3ffbac50 0x4010ad71:0x3ffbac70 0x4
010b783:0x3ffbad50 0x400d0e46:0x3ffbad80
0x40088b84: invoke_abort at C:/Users/Trey/esp/esp-idf/components/esp32/panic.c:5
72

0x40088ef1: _esp_error_check_failed at C:/Users/Trey/esp/esp-idf/components/esp3
2/panic.c:584

0x4010ad71: initialise_wifi at C:/Users/Trey/esp/subscribe_publishTestingCode/ma
in/subscribe_publish_sample.c:612 (discriminator 1)

0x4010b783: app_main at C:/Users/Trey/esp/subscribe_publishTestingCode/main/subs
cribe_publish_sample.c:629

0x400d0e46: main_task at C:/Users/Trey/esp/esp-idf/components/esp32/cpu_start.c:
449

Rebooting...
```

I tried looking at other issues related to boot looping, and I downloaded esptool.py and did `erase_flash`, but it's still happening. I am a bit new to this and I don't know where to go from here.

igrr, 2017-11-29:
You may need to add a call to nvs_flash_init before initializing wifi. See
ESP-IDF examples/wifi directory for reference.

lawlhwut, 2017-11-29:
Thanks so much! It was related to the NVS.

closed 2017-11-29 by lawlhwut