espressif/esp-idf #1329: SPI transfers limited to approximately 4KiB

MrSurly, 2017-11-29:
Attempting to transfer more than 4092 bytes (using DMA) results in `E (63662) spi_master: spi_device_queue_trans(620): txdata transfer > host maximum`.

Solutions, in order of most desirable to least:
* `spi_device_transmit` handles larger blocks transparently
* The SPI code provides a mechanism for user code to query the maximum transfer size
* The SPI documentation is updated to reflect the limitation

MrSurly, 2017-11-29:
I'd be willing to submit a PR, if there was a reasonable chance it would be accepted.

negativekelvin, 2017-11-29:
https://github.com/espressif/esp-idf/blob/master/components/driver/spi_master.c#L87

negativekelvin, 2017-11-29:
https://github.com/espressif/esp-idf/blob/63e1e4e502178ced9d29999e45050c952be29c26/components/driver/include/driver/spi_common.h#L60

FayeY, 2017-12-01:
Hi @MrSurly , did the above suggestion help you resolve the issue?

ginkgm, 2017-12-01:
@MrSurly 
As @negativekelvin said, there's a configuration field to set the max transfer size, this is mainly because we have to allocate space for the DMA descriptors at initialization. 
If this field is not set, we only use 1 descriptor which support up to 4094 bytes buffer length.

MrSurly, 2017-12-01:
@FayeY Yes, that works.  Thanks.

closed 2017-12-01 by MrSurly

bonmotwang, 2019-07-10:
I am having the same issue in using SPI to update a LCD module. I need 38400 bytes of DMA buffer in calling spi_device_transmit. It is only around 4000 bytes could be sent at once.

ginkgm, 2019-07-11:
Hi @bonmotwang ,
Please set the parameter here, and also prepare enough size of *Internal* buffer for RAM

https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#_CPPv4N16spi_bus_config_t15max_transfer_szE

https://espressif-docs.readthedocs-hosted.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html?highlight=heap_caps#_CPPv416heap_caps_malloc6size_t8uint32_t

If the internal RAM is not enough, please split the data into pieces and send by several times.