espressif/esp-idf #1330: linker failing on macOS

matthewberryman, 2017-11-29:
I've followed the instructions here https://esp-idf.readthedocs.io/en/latest/get-started/macos-setup.html and http://esp-idf.readthedocs.io/en/latest/get-started/index.html#get-esp-idf making sure to recursively clone and also setting PATH and IDF_PATH variables (and restarting my shell to pick them up), but I get the following linker errors building the hello_world project. I've repeated the same steps (but with the linux build tools) on Ubuntu and not encountered this issue.
```
/Users/matthew/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Users/matthew/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/Users/matthew/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/Users/matthew/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/Users/matthew/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
make[1]: *** [/Users/matthew/esp/hello_world/build/bootloader/bootloader.elf] Error 1
make: *** [/Users/matthew/esp/hello_world/build/bootloader/bootloader.bin] Error 2
```

igrr, 2017-12-08:
It looks like some compiler flags are being overridden from the environment. Could you please check if `printenv | grep -i FLAGS` contains any of `LDFLAGS`, `CFLAGS`, `CPPFLAGS`, `CXXFLAGS`?

matthewberryman, 2017-12-12:
Ah, that's it. I had LDFLAGS being set separately through a really old shell startup script line I'd forgotten about. Problem solved, thanks!

closed 2017-12-12 by matthewberryman

HarterHorst, 2017-12-28:
exact same error here. No results on `printenv | grep -i FLAGS`. 

Any other ideas?

`andi@strech:~/eclipse-workspace/template/Debug$ make
Building file: ../main/main.c
Invoking: Cross GCC Compiler
xtensa-esp32-elf-gcc -I/home/andi/esp/esp-idf/components/esp32/include -I/home/andi/esp/esp-idf/components/freertos/include -I/home/andi/esp/esp-idf/components/driver/include -I/home/andi/esp/esp-idf/components/nvs_flash/include -I/home/andi/esp/esp-idf/components/wpa_supplicant/include -I/home/andi/eclipse-workspace/template/build/include -I/home/andi/esp/esp-idf/components/soc/esp32/include -I/home/andi/esp/esp-idf/components/heap/include -I/home/andi/esp/esp-idf/components/soc/include -I/home/andi/esp/esp-idf/components/tcpip_adapter/include -I/home/andi/esp/esp-idf/components/lwip/include/lwip -I/home/andi/esp/esp-idf/components/lwip/include/lwip/port -I/home/andi/esp/esp-idf/components/vfs/include -O0 -g3 -Wall -c -fmessage-length=0 -MMD -MP -MF"main/main.d" -MT"main/main.o" -o "main/main.o" "../main/main.c"
Finished building: ../main/main.c
 
Building target: template
Invoking: Cross G++ Linker
xtensa-esp32-elf-g++  -o "template"  ./main/main.o   
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
makefile:165: recipe for target 'template' failed
make: *** [template] Error 1
`

matthewberryman, 2017-12-28:
Maybe you’ve got a LD_LIBRARY_PATH or other compiler / linker env var set?

HarterHorst, 2017-12-28:
Can't find anything in the environment. I have added the -v options to get more information from the compiler and structured the file a little bit. I have to idea what these files are and if the belong to the build environment of the esp-idf or not. 

```
COLLECT_GCC_OPTIONS='-v' '-o' 'template'
 /home/andi/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/collect2 
 -plugin /home/andi/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/liblto_plugin.so 
 -plugin-opt=/home/andi/esp/xtensa-esp32-elf/bin/../libexec/gcc/xtensa-esp32-elf/5.2.0/lto-wrapper 
 -plugin-opt=-fresolution=/tmp/cc928UcH.res 
 -plugin-opt=-pass-through=-lgcc 
 -plugin-opt=-pass-through=-lc 
 -plugin-opt=-pass-through=-lsim 
 -plugin-opt=-pass-through=-lc 
 -plugin-opt=-pass-through=-lhandlers-sim 
 -plugin-opt=-pass-through=-lhal 
 -plugin-opt=-pass-through=-lgcc 
 
 --sysroot=/home/andi/esp/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot -o template crt1-sim.o 
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib/crt0.o 
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/crti.o 
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/crtbegin.o _vectors.o -L
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0 -L
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc -L
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/lib -L
/home/andi/esp/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/lib -L
/home/andi/esp/xtensa-esp32-elf/bin/../xtensa-esp32-elf/sysroot/usr/lib ./main/main.o -lstdc++ -lm -lgcc -lc -lsim -lc -lhandlers-sim -lhal -lgcc 
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/crtend.o 
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/crtn.o

/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find _vectors.o: No such file or directory
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lsim
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhandlers-sim
/home/andi/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lhal
collect2: error: ld returned 1 exit status
makefile:165: recipe for target 'template' failed
make: *** [template] Error 1

```

rodrigoords, 2019-08-03:
Can you explain what exactly to do with this error? I have the same.

`/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -ldriver
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lwpa_supplicant
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lbt
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lc_nano
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lc
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lmbedtls
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -llwip
/Users/rodrigosene/Documents/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.
`