espressif/esp-idf #1332: esp_wifi_80211_tx function overwrites frame counter

seppestas, 2017-11-29:
I originally reported this reported as a comment on issue #677.

The [`esp_wifi_80211_tx` function](https://github.com/espressif/esp-idf/blob/master/docs/api-guides/wifi.rst#wi-fi-80211-packet-send) seems to overwrite the frame counter of the frames sent using it, even if `en_sys_seq` is set to false.

When setting `en_sys_seq` to false the frame counter is always overwritten to 0. When setting `en_sys_seq` to true the frame counter follows the frame counter of frames sent by other components acting on WiFi (e.g AP beacons), but the frames sent by `esp_wifi_80211_tx` do not themselves cause the frame counter to increment. This happens both when WiFi was configured in AP and Station mode. Tested on ESP-IDF v2.1 and on the master branch (094e47f).

The old [`ieee80211_freedom_output` function](https://github.com/Jeija/esp32free80211/blob/master/components/free80211/free80211.c) did not have this behaviour, allowing to send frames with arbitrary frame counter values.

liuzfesp, 2018-01-22:
@seppestas esp_wifi_80211_tx will be optimized in v3.1

seppestas, 2018-01-22:
@liuzfesp Any idea when v3.1 would become available? [The milestones on GH](https://github.com/espressif/esp-idf/milestones) do not seem to be very up to date...

We currently have 2 different firmwares, one based on ESP-IDF v2.0 to support the low level IEEE802.11 functionality we need and one where we have a BLE PoC for configuration. Merging both to a newer version of the ESP-IDF would be a lot easier (and future proof) than trying to backport the BLE functionality.

closed 2018-05-18 by igrr