espressif/esp-idf #1333: ADC1 does not work in Multiple-channel scanning mode.

phonec, 2017-11-29:
As stated in the "ESP32 Technical Reference Manual", in the "27.3.5 DIG SAR ADC Controllers" can perform high-speed scanning by pattern table. This is true only for the ADC2, because their functionality is already exists in bootloader_random_enable(). But this is not what we need, when you turn on the WiFi, the scan is broken because wifi driver uses some ADC2 lines. ADC1 does not work in single scanning mode?

phonec, 2017-12-07:
It seems that ADC1 is working in scanning mode, if you return back to the "[feature(I2S-ADC): add ADC mode for I2S](https://github.com/espressif/esp-idf/commit/2fceec4d858c62a489496678c765ae1a2d68645c)" but after the "[Merge branch 'feature/i2s_built_in_adc' into 'master'](https://github.com/espressif/esp-idf/commit/b6f37bda9715cfd026cd3203a0fa3b4faf475324#diff-27688c6b3c29373d2a2b142b8471981c)"  it ceased.

admiralbub, 2017-12-08:
We checkout to the earlier commit of esp-idf: "2fceec4" and we now have a microphone really records the sound. Will this be fixed in the near future?
What should I do to record two microphones for the left and right stereo channels at the same time?

phonec, 2017-12-11:
up

costaud, 2018-01-29:
We already have a branch fix i2s-adc function under merge reviewing. Will fix this soon.

phonec, 2018-02-05:
@costaud  Thank you for the good news, we are looking forward to it.

phonec, 2018-02-12:
Hi! , good job! The example i2s_adc_dac work! I need some time to check this with together with Wifi  in the project that I suspended 2 months ago.

phonec, 2018-02-12:
Hi, please see simple code is that I tested this issue, here is an example: **[main.c](https://gist.github.com/phonec/42d685580b4c5b93c8b38d5fa68e0e58)**

- ADC-I2S Scanning does not normal start (**more badly**) after the reset of ESP32. It is necessary to add a delay of 5 seconds before switching on the ADC for a reliable start, this is issue.

- The raw ADC data is written in DMA in inverted form. I have to manualy add an inversion:
```SET_PERI_REG_MASK(SYSCON_SARADC_CTRL2_REG, SYSCON_SARADC_SAR1_INV);```
 before starting the ADC, please correct this.

- Please allow global access to the template table customization function adc_set_i2s_data_pattern(). In order to set several ADC lines for scanning, Now I have to manually fill this table before starting the ADC, for example 1 and 4 ADC line simultaneous:
```SET_PERI_REG_BITS(SYSCON_SARADC_CTRL_REG, SYSCON_SARADC_SAR1_PATT_LEN, 1, SYSCON_SARADC_SAR1_PATT_LEN_S); WRITE_PERI_REG(SYSCON_SARADC_SAR1_PATT_TAB1_REG, 0x0E4E0000); ```
and it works for me, what prevents you from doing it?

- After running **WiFi** on **```esp_wifi_set_mode()```**, the ADC-I2S scanning is stops. **This is the biggest issue.**

phonec, 2018-02-14:
@costaud Please make feedback to my comment. We really have such issues, can you tell how to get around them? I also failed to start ADC->I2S together with the enabled WiFi.

FayeY, 2018-02-23:
Hi phonec, sorry that these days are the Chinese Spring Festival Days, so all our engineers are on holiday. They will back to work in the next couple days. Sorry for the inconvenience.

costaud, 2018-02-24:
@phonec , I can't access to your main.c. 
We will have a test on those issues.

phonec, 2018-02-24:
Hi @costaud, the [main.c](https://gist.github.com/phonec/42d685580b4c5b93c8b38d5fa68e0e58) already exists in public **gist**: [https://gist.github.com/phonec/42d685580b4c5b93c8b38d5fa68e0e58](https://gist.github.com/phonec/42d685580b4c5b93c8b38d5fa68e0e58)
there must be public access without restrictionons

igrr, 2018-02-25:
FWIW, gist.github.com is not accessible from China. I am attaching the file here for reference.
[42d685580b4c5b93c8b38d5fa68e0e58-eb519e6929be7a5abe20df28effdbbdc8d7919ae.zip](https://github.com/espressif/esp-idf/files/1756579/42d685580b4c5b93c8b38d5fa68e0e58-eb519e6929be7a5abe20df28effdbbdc8d7919ae.zip)

phonec, 2018-02-25:
@iggr thank you for the help, did not know this. Obviously this is the strange issue of GFW.

koobest, 2018-02-26:
Hi, @phonec. I have a test with your code and the results as you said, but when I turn on wifi befor init i2s, it woks. we will try to find the cause of the problem.  thanks for your feedback.

phonec, 2018-02-26:
OK, this is already something, but there may be cases when you need to initialize the Wi-Fi after the adc-i2s. Need to enter a delay before *i2s_adc_enable()*, for good starting?

koobest, 2018-03-12:
Hi, @phonec 
    After some testing we found that there are indeed some configurations in the wifi initialization which will lead to the problems you describe. Our engineers are solving this bug. If there are any updates, we will reply to you immediately.
 thanks !!!

closed 2018-05-11 by FayeY

phonec, 2018-05-11:
Hi @koobest, works fine, thanks !!
adc_set_i2s_data_pattern() and adc_set_i2s_data_len() in rtc_module.c
now do not must have to be static, we can call them to expand the scan table.