espressif/esp-idf #1336: Struct in queue has strange behavior

jstar88, 2017-11-30:
I have ESP-IDF 2.1 and I'm using a queue  to reiceve data from multiple task.
For the porpuse, my code is pretty same of the example in freertos open book.
[https://www.freertos.org/Documentation/161204_Mastering_the_FreeRTOS_Real_Time_Kernel-A_Hands-On_Tutorial_Guide.pdf](url)
at page 145.

```
typedef enum
{
	eSender1 = 1, eSender2, eSender3
} DataSource_t;

typedef struct
{
	DataSource_t eDataSource;
	uint16_t ucValue;
}Data_t;
```

When i dequeue an Data_t item as above, the fields have a correct value.
However, if the order of eDataSource and ucValue are switches, eDataSource is always empty.

```
typedef struct
{
        uint16_t ucValue;
	DataSource_t eDataSource; //always empty
}Data_t;
```

projectgus, 2017-11-30:
Hi @jstar88,

Check that you are passing `sizeof(Data_t)` correctly as the item size when calling xQueueCreate().

If you're still having problems, can you please post the exact code you are running? https://gist.github.com/ is a good place to post code snippets.

Angus

FayeY, 2017-12-05:
Hi @jstar88 , did the above suggestion help you resolve the issue?
If not, could you provide your test code?
Thanks.

jstar88, 2017-12-05:
hi @projectgus @FayeY 
no. I'm correctly initializing queue with sizeof(Data_t).
I also tried using __attribute__((packed))  without benefits.

My code has lot of things, i'm going to try the example in the book to test if the bug persist

projectgus, 2017-12-05:
Hi @jstar88,

Thanks for the update. Github issues are not the best place for this kind of discussion.

Please consider posting this as a question on the http://esp32.com/ forum, along with the sample code you are running. We'll be able to help you better over there.

If you believe strongly there's a bug in ESP-IDF and you have some test code, feel free to reopen.

Angus

closed 2017-12-05 by projectgus