espressif/esp-idf #1338: Warning "-s option given but default rule can be matched"

tdy91, 2017-11-30:
Building esp-idf hello-world template, after a "make clean", the first "make all" generates the 
warning "-s option given but default rule can be matched" .

This warning appears as an error in Eclipse, even though the compilation succeed.

This warning is generated by flex :

flex -L -P zconf -o zconf.lex.c zconf.l
zconf.l:256: warning, -s option given but default rule can be matched.

I think it is because in **esp-idf/tools/kconfig/zconf.l** file, there is option **nodefault** on line 2 (which is the same as specifying "-s" flex command line option) :

**Line 2 :** %option 8bit **nodefault** perf-report perf-report

but there is also a **defaut rule "."** specified on line 97 :
.	{
	unput(yytext[0]);
	BEGIN(COMMAND);
}

The solution seems to be to suppress nodefault option on line 2 if a default rule is defined.

Line 2 : %option 8bit perf-report perf-report

This suppress the warning.

krzychb, 2017-12-03:
Hi @tdy91,

This warning was bugging me for several months on both Linux and Windows installation.
I also saw it in logs by some other people. Now I checked it on Linux and after applying your fix the warning does not show anymore. Later I will check my Windows installation.

Thank you for taking your time to troubleshoot it and  providing a fix :+1: 

I am wondering if there is anybody else who can comment in this issue? I do not have experience with flex and can not comment on possible side effects on suppressing this warning. Depending on feedback I think we should apply it in master.

tdy91, 2017-12-04:
Hi @krzychb

I am just beginning on esp-idf and don't like to see an icon error in Eclipse, mostly on very simple Hello-World project; it was bugging me too !

I do not have experience with flex too, but with ANTLR with is another parser generator using a similar lexical analyzer.

Trying to find a solution, I found this answer to the same warning on another project : 

******************
Source : https://stackoverflow.com/questions/1622493/bison-build-warning-s-option-given-but-default-rule-can-be-matched
Answer : date : Oct 26 '09 at 4:37
Author : UncleO

The "flex" command on your system could have been aliased to be the flex command with an -s option. That could occur in any of several startup scripts.

The warning is telling you that there are some inputs that aren't matched by any of the rules in your input file. The warning is there to get you to make sure you have considered all cases in your regular expressions.

If you truly want the default rule to be used, then add it to the end of the rules in your input file. Try adding the rules

. ECHO;
\n ECHO;
Update: The input file you provide includes the %option nodefault, which is the same as specifying "-s" on the command line, so that mystery is solved. Also, the file does not have a rule for ".". There are some characters not covered, '@' for example.

I'm also suspicious of the "-" symbol in the character class. It is supposed to be a range operator, but it may work as a character in the first position as you have it. I don't know. It might be a good idea to use "\-" instead.

You could change the last rule to

. { printf("lex Unknown character = '%s'", yytext); yyerror("lex Unknown character"); }
to get rid of the warning (and to prevent the resulting scanner from crashing).

******************

My own analysis is following one :
 

1. Help of flex V2.6.4 used in esp-idf project gives :
flex --help
  -s, --nodefault         suppress default rule to ECHO unmatched text
info flex
'-s, --nodefault, '%option nodefault''
     causes the _default rule_ (that unmatched scanner input is echoed
     to 'stdout)' to be suppressed.  If the scanner encounters input
     that does not match any of its rules, it aborts with an error.
     This option is useful for finding holes in a scanner's rule set.

2. The flex default rule matches a single character and prints it on standard output.

3. The zconf.l file contains tokens and rules to produce sp-idf\tools\kconfig\mconf.exe, used to produce sdkconfig on "make menuconfig" command.
   Today, in this file there is :
   * line 2  : nodefault option defined
   * line 97 : a default rule is explicitly defined by kconfig developper, so the flex default rule is not used. 

So, from my point of view, to use flex, you have :
 * either write an *.l file which uses flex default rule (i.e. without defining in it an "." explicit default rule) and 
   use flex -s (or nodefault option) command line for syntax debugging purposes.
 * either write an *.l which defines an explicit default rule, and in this case, there can be no syntax hole in scanner rule set,  so -s (or nodefault option) command line is not useful, and this is the meaning of the warning..

In fact, this is just a flex warning, not an error, and as we are not debugging kconfig, but developping esp-idf projects, my own opinion is that suppressing "nodefault" option from zconf.l will have no side effects, but another feedback may be preferable before applying this solution in master.

krzychb, 2017-12-04:
Very nice explanation, thanks @tdy91 !

Sure enough Windows system is also cured with your fix :medal_sports: 

For other people reading, this issue looks as below:

[image]

I found it very distractive to see an icon error in Eclipse, while actually there are no any errors! You can stop the build from Eclipse and start it again to remove it. The fix identified by @tdy91 does not require any artificial treatment like that.

closed 2017-12-13 by igrr