espressif/esp-idf #1339: Using SDCard in SPI mode & spi_master at the same time

loboris, 2017-12-01:
It looks like **spi_master** and sd card in spi mode (**sdpi_host**) cannot be used at the same time.

I've tested combining the spi master example and sd card example and I cannot get  both working.
Test configuration:
- SDcard and display use the **different pins** for spi interface.
- Display uses **HSPI_HOST** and **DMA channel 1**.
- SDCard uses **VSPI_HOST** an **DMA channel 2**.

**_Depending on which one is accessed first, either sdcard or display does not work._**

- 
When **the sdcard is mounted first**, sd card is detected, card info is correct, file can be opened, read, written...
Then the display is initialized.
After that, trying to open the file fails with:
```
E (2800) sdspi_host: sdspi_host_start_command: cmd=24 error=0x107
D (2800) sdmmc_cmd: sdmmc_req_run returned 0x107
E (2800) sdmmc_cmd: sdmmc_write_sectors_dma: sdmmc_send_cmd returned 0x107
E (2810) diskio_sdmmc: sdmmc_write_blocks failed (263)
```
Display work with no problem.

- 
When **the display is intialized first**, after the initialization the first display test works as expected, display is initialized and the pattern drawn.
Then the sdcard is mounted and works as expected, card info is correct, file can be opened, read, written...
After the sdcard intialization **the display no longer works**.

ginkgm, 2017-12-07:
Okay, we'll check this.

ginkgm, 2018-01-23:
I've tried the following code in ``spi_master_example_main.c``, but cannot reproduce your issue. 
Can you show me your code?

```
void app_main()
{
    esp_err_t ret;

    spi_device_handle_t spi;
    spi_bus_config_t buscfg={
        .miso_io_num=PIN_NUM_MISO,
        .mosi_io_num=PIN_NUM_MOSI,
        .sclk_io_num=PIN_NUM_CLK,
        .quadwp_io_num=-1,
        .quadhd_io_num=-1
    };
    spi_device_interface_config_t devcfg={
        .clock_speed_hz=10*1000*1000,               //Clock out at 10 MHz
        .mode=0,                                //SPI mode 0
        .spics_io_num=PIN_NUM_CS,               //CS pin
        .queue_size=7,                          //We want to be able to queue 7 transactions at a time
        .pre_cb=lcd_spi_pre_transfer_callback,  //Specify pre-transfer callback to handle D/C line
    };
    //Initialize the SPI bus
    ret=spi_bus_initialize(HSPI_HOST, &buscfg, 1);
    assert(ret==ESP_OK);
    //Attach the LCD to the SPI bus
    ret=spi_bus_add_device(HSPI_HOST, &devcfg, &spi);
    assert(ret==ESP_OK);

    ESP_LOGI(TAG, "Initializing SD card");

    ESP_LOGI(TAG, "Using SPI peripheral");

    sdmmc_host_t host = SDSPI_HOST_DEFAULT();
    host.slot = VSPI_HOST;
    sdspi_slot_config_t slot_config = SDSPI_SLOT_CONFIG_DEFAULT();
    slot_config.gpio_miso = 2;
    slot_config.gpio_mosi = 15;
    slot_config.gpio_sck  = 14;
    slot_config.gpio_cs   = 13;
    slot_config.dma_channel = 2;
    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.

    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and
    // formatted in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false,
        .max_files = 5
    };

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc_mount is an all-in-one convenience function.
    // Please check its source code and implement error recovery when developing
    // production applications.
    sdmmc_card_t* card;
    ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);

    if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. "
                "If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%d). "
                "Make sure SD card lines have pull-up resistors in place.", ret);
        }
        return;
    }

    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

    // Use POSIX and C standard library functions to work with files.
    // First create a file.
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen("/sdcard/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello %s!\n", card->cid.name);
    fclose(f);
    ESP_LOGI(TAG, "File written");

    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen("/sdcard/hello.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    //Go do nice stuff.
    display_pretty_colors(spi);
}
```

Abol-Wang, 2018-10-23:
@loboris I recently encountered the same problem with M5stack, do you have a suitable solution?

FayeY, 2018-10-25:
Hi @loboris , I'm going to close this issue due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.
Hi @Abol-Wang , is your question #2616 which the same as #1597? If it is, please follow the #1597 for further update. If it is not, what you mentioned is a new, different problem from #1597, please feel free to let us know. Thanks.

closed 2018-10-25 by FayeY