espressif/esp-idf #1341: BLE Joystick disconnected

Sagigamil, 2017-12-01:
Hi!
I'm succesfully connected my esp32 to BLE joystick by changing the gatt secured client example to register the hid events. I get two bytes for every joystick button click and this is great.
I noticed that after a minute or two, after not pusing any button, the connection is disconnectins and the error I'm get is:
E (166964) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0022
I'm do not know what cause it, maybe there is a way to make the gatt protocol wait forever to the event?

Yulong-espressif, 2017-12-15:
@Sagigamil The reason is because the controller response timeout. So your joystick is used BLE 4.0 instead of BLE 4.2, Yeah? I think you can try to update the IDF version, we have fix a similar bug in the new version of SDK. Thanks.

FayeY, 2017-12-21:
Please feel free to reopen it, if this issue is still unsolved on the latest IDF. Thanks.

closed 2017-12-21 by FayeY