espressif/esp-idf #1343: Unexpected fail

Sagigamil, 2017-12-01:
I'm running the gatt secured client with a few changes, The program crash after an unexpected with error:
ASSERT_PARAM(0 0), in llc.c at line 746ets
someone know what is the issue? I cannot found the file llc.c

rdsantos18, 2017-12-01:
I have a similar error using socket tcp. when socket is connected I have several reboots

I (732) wifi: mode : softAP (30:ae:a4:02:78:ed)
Access point started
Bus I2C init.
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Register dump:
PC      : 0x40084b35  PS      : 0x00060034  A0      : 0x8008647e  A1      : 0x3ffc0520  
0x40084b35: xQueueGenericSendFromISR at /home/rinaldo/esp/esp-idf/components/freertos/./queue.c:2037

A2      : 0x3ffcf9a0  A3      : 0x00060021  A4      : 0x3ffc0580  A5      : 0x00000000  
A6      : 0x3ffcf9a0  A7      : 0x3ffcf9e8  A8      : 0x80084bb3  A9      : 0x3ffc0510  
A10     : 0x00000003  A11     : 0x3ffc0550  A12     : 0x00000484  A13     : 0x3f406499  
A14     : 0x00000031  A15     : 0x00000000  SAR     : 0x00000014  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x40084b35:0x3ffc0520 0x4008647b:0x3ffc0550 0x40083d90:0x3ffc0580 0x400841b5:0x3ffc05c0 0x40082601:0x3ffc0600 0x4000bfed:0x00000000
0x40084b35: xQueueGenericSendFromISR at /home/rinaldo/esp/esp-idf/components/freertos/./queue.c:2037

0x4008647b: xTimerPendFunctionCallFromISR at /home/rinaldo/esp/esp-idf/components/freertos/./timers.c:1000

0x40083d90: i2c_master_cmd_begin_static at /home/rinaldo/esp/esp-idf/components/driver/./i2c.c:973

0x400841b5: i2c_isr_handler_default at /home/rinaldo/esp/esp-idf/components/driver/./i2c.c:973

0x40082601: _xt_lowint1 at /home/rinaldo/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

Rebooting...
ets Jun  8 2016 00:22:57

Sagigamil, 2017-12-02:
Any solution?

koobest, 2017-12-25:
Hi, @Sagigamil  please update your IDF to the latest version,  we have made changes to the I2C driver, and maybe can solve your problem.
see commit  bf3c41bb929410493e2a9203534bdc8f21683209

FayeY, 2018-01-05:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-01-05 by FayeY