espressif/esp-idf #1344: BLE multi-device application problem

Arthedian, 2017-12-02:
Hi,
In my application my ESP32 connect to 2 devices. I succesfully connect a register notify event. When I send notify event from device to my ESP suddenly GATTC handler has connection ID 22, but my devices has ID 0 a 1? I know this because I print my Connection everytime. How should I check than what device is it for?

xiewenxiang, 2017-12-04:
You can check it by the remote address.

Arthedian, 2017-12-04:
Most of the mac supposed remote mac address is bad. For the first time I call the handler (ESP_GATTC_CONNECT_EVT happen) the mac address is good, but than I am searching for characteristics and register for notify event the mac address goes nuts and its bad. I am not sure why (I am 100% sure, that I dont register any other device)-

xiewenxiang, 2017-12-04:
the demo was one client connect to three server
[multi_connection.tar.gz](https://github.com/espressif/esp-idf/files/1526531/multi_connection.tar.gz)

Arthedian, 2017-12-04:
So If I am looking right, I should check againt gattc_if number?

Arthedian, 2017-12-04:
How should I check which device created the event? Because when I call `esp_ble_gattc_open` with specifig gattc_if, the GATTC event handler is called for every gattc_if registered?

Arthedian, 2017-12-04:
So I am checking my device against gattc_if. I successfully register notify event, but when I am connecting to my second device and I call `esp_ble_gattc_send_mtu_req` or `esp_ble_gattc_search_service` nothing happends. Why? Is it normal?

Arthedian, 2017-12-04:
I am doing almost the same code as in the example above from @xiewenxiang

xiewenxiang, 2017-12-06:
could you send your test code to me?

Arthedian, 2017-12-06:
[https://1drv.ms/u/s!AtQMJ50-g_ldh_lHohSTFFrmvEGeaA](url) Here is the link. I am periodically sending all visible devices to a the server gives me list of mac address which I should connect to.

FayeY, 2017-12-06:
Hi @Arthedian , sorry that the link you provided is blocked by the GFW, we cannot visit it. 
Could you share it by another way?

Arthedian, 2017-12-06:
[https://owncloud.cesnet.cz/index.php/s/PiBwu903IExJ9fg](url) is this ok?

Arthedian, 2017-12-06:
And I am trying to test it on your multiconnect client, and it register my device to all application handler. Is this normal? I only changed name of the device.

xiewenxiang, 2017-12-07:
@Arthedian  ,sorry that the link you provided is also blocked by the GFW.

Do you test the BLE multiconection use the client demo provided by us and the server is your code? If  so, it maybe cann't work well.

In the client , you need do as these:

1. regeister two app profile.
    `esp_ble_gattc_app_register(PROFILE_APP_ID_A);`
    `esp_ble_gattc_app_register(PROFILE_APP_ID_B);`
2. set the scan param and start scan.
    `esp_ble_gap_set_scan_params(&ble_scan_params);`
    `esp_ble_gap_start_scanning(SCAN_ALL_THE_TIME);`
3. get the scan result and find the device which you want to connect.
4. stop scan
5. connect to the first device.
    `esp_ble_gattc_open(gl_profile_tab[PROFILE_APP_ID_A].gattc_if, scan_result->scan_rst.bda, true);`
6. do the service search for the first device.
    `esp_ble_gattc_search_service`
7. exchange MTU size
    `esp_ble_gattc_send_mtu_req`
8. register the notification
    `esp_ble_gattc_register_for_notify`
9. write the characteristic description which prop is notify or indicate. 
10. start scan and ready connect to the second device

Arthedian, 2017-12-07:
What do you mean by the 9? Do I need to do it? I want to send data to my server, when I Get notification from BLE server device.

xiewenxiang, 2017-12-07:
@Arthedian  Yes, If you only send data to server, you just need write the characteristic, If you want receive the notification from server, you need to write the characteristic description.

xiewenxiang, 2017-12-07:
you can find the detail in BLE <Core_v4.1 Part G 3.3.3.3>

xiewenxiang, 2017-12-07:
[https://www.bluetooth.org/DocMan/handlers/DownloadDoc.ashx?doc_id=282159&_ga=2.205366736.1560893716.1512634918-1549765612.1511763818](url)

Arthedian, 2017-12-07:
Oh sorry for my bad explanation. When notification from BLE server come I want to send request to my Web server. 

1) Why If I dont write characteristic desription it is working for 1 connected device, but when I am trying to connect second one its not working? Its still working only for 1 device. I will try to add writing characteristic description.

2) Why I Need to start scanning again after registering for notification, not immediately after connecting?

xiewenxiang, 2017-12-07:
Hi Arthedian,

1. Why If I dont write characteristic desription it is working for 1 connected device, but when I am trying to connect second one its not working? Its still working only for 1 device. I will try to add writing characteristic description.

because the bluedroid didn't check it before send notify or indicate, so, we suggested the client write the characteristic description (UUID:0x2902) before receive nofity/indicate(notify:write 0x01, indicate:write 0x02). In server, check the description value in application code before send notify/indicate. 

2. Why I Need to start scanning again after registering for notification, not immediately after connecting?

you can start scan immediately after connecting

closed 2017-12-19 by FayeY