espressif/esp-idf #1346: spi_slave.c

reklof, 2017-12-03:
The code for setting the hardware according to the 'mode' parameter is incorrect. mode == 2 and mode == 3 should swap (see spi_master.c)

`if (slave_config->mode == 0) {`
`spihost[host]->hw->pin.ck_idle_edge = 0;`
`        spihost[host]->hw->user.ck_i_edge = 1;`
`        spihost[host]->hw->ctrl2.miso_delay_mode = nodelay ? 0 : 2;`
`    } else if (slave_config->mode == 1) {`
`        spihost[host]->hw->pin.ck_idle_edge = 0;`
`        spihost[host]->hw->user.ck_i_edge = 0;`
`        spihost[host]->hw->ctrl2.miso_delay_mode = nodelay ? 0 : 1;`
`    } else if (slave_config->mode == 2) {`
`        spihost[host]->hw->pin.ck_idle_edge = 1;`
`        spihost[host]->hw->user.ck_i_edge = 0;`
`        spihost[host]->hw->ctrl2.miso_delay_mode = nodelay ? 0 : 1;`
`    } else if (slave_config->mode == 3) {`
`        spihost[host]->hw->pin.ck_idle_edge = 1;`
`        spihost[host]->hw->user.ck_i_edge = 1;`
`        spihost[host]->hw->ctrl2.miso_delay_mode = nodelay ? 0 : 2;`
`    }`

Spritetm, 2017-12-09:
Can I ask how you figured out this is incorrect? Does it not match other documentation, or do you have issues using mode 2 or 3 in practice?

jlujan-na, 2018-01-05:
The driver code does not match `Table 27: Clock Polarity and Phase, and Corresponding SPI Register Values for SPI Slave` from the technical reference. 

Registers | mode0 | mode1 | mode2 | mode3
-- | -- | -- | -- | --
SPI_CK_IDLE_EDGE | 0 | 0 | 1 | 1
SPI_CK_I_EDGE | 0 | 1 | 1 | 0
SPI_MISO_DELAY_MODE | 0 | 0 | 0 | 0
SPI_MISO_DELAY_NUM | 0 | 0 | 0 | 0
SPI_MOSI_DELAY_MODE | 2 | 1 | 1 | 2
SPI_MOSI_DELAY_NUM | 0 | 0 | 0 | 0

It also has hardcoded nodelay=true; https://github.com/espressif/esp-idf/blob/eb659e401d812801c024258a1c927143b4861124/components/driver/spi_slave.c#L148

jlujan-na, 2018-01-05:
There is also an ambiguous assignment https://github.com/espressif/esp-idf/blob/eb659e401d812801c024258a1c927143b4861124/components/driver/spi_slave.c#L144

negativekelvin, 2018-01-05:
It seems more likely the table row SPI_CK_I_EDGE is wrong and values should be inverted bc that would match all code.  But nodelay should be false to match table.

jlujan-na, 2018-01-05:
The `mode` configuration in spi_master.c matches the table.

> sync_reset is not ambiguous, it is set and then cleared intentionally.
I see

negativekelvin, 2018-01-05:
Well if table is right then all values of ck_i_edge should be inverted in code.  But not swap 2&3.

jlujan-na, 2018-01-05:
I think the code should also be setting `mosi_delay_mode` instead of `miso_delay_mode`. Would make the code to match the tech ref:

```
if (slave_config->mode == 0) {
        spihost[host]->hw->pin.ck_idle_edge = 0;
        spihost[host]->hw->user.ck_i_edge = 0;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 2;
    } else if (slave_config->mode == 1) {
        spihost[host]->hw->pin.ck_idle_edge = 0;
        spihost[host]->hw->user.ck_i_edge = 1;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 1;
    } else if (slave_config->mode == 2) {
        spihost[host]->hw->pin.ck_idle_edge = 1;
        spihost[host]->hw->user.ck_i_edge = 1;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 1;
    } else if (slave_config->mode == 3) {
        spihost[host]->hw->pin.ck_idle_edge = 1;
        spihost[host]->hw->user.ck_i_edge = 0;
        spihost[host]->hw->ctrl2.mosi_delay_mode = 2;
    }
```

WolfWalter, 2018-01-09:
I checked the MISO for a mode3 spi-slave with an oscilloscope. To fix the timing I needed to change spihost[host]->hw->ctrl2.mosi_delay_mode to 1.

ginkgm, 2018-02-27:
As far as I know, the table given by the Technical Reference Manual can produce correct timing. But sometimes it does not work well with DMA, so there're 2 modes with different values with the TRM.

Please try:
1. If the current settings work for your design, suggest ignore the timing. Due to the output delay, the SPI still works well for most cases.
2. If you want to fix the timing, first set the register as the TRM described. If everything is well, skip step 3 and 4.
3. Please note the slave readings may get corrupted due to a DMA issue. If unfortunately your reading is not correct, to see if this is the reason, initialize without DMA and retry. 
4. You have 2 ways to get around this issue: 
    1. use other SPI modes, currently mode 0 and 2 (or maybe 1 and 3, I can't remember the detail.) cannot work with DMA well, but the other 2 modes do.
    2. turn off the DMA.

I prepare to refactor the code to make the workaround more obviously and influence smaller region.

closed 2019-01-29 by igrr