espressif/esp-idf #1347: Add assembly output support to build system

OtherCrashOverride, 2017-12-04:
There needs to be an option to enable the output of ".s" (assembly) files in addition to ".o" (object) files when "make" is run.  This is important to understanding the performance implications of code being compiled.

OtherCrashOverride, 2017-12-04:
The following produces the expected output:
```xtensa-esp32-elf-objdump -d -M xtensa -S path/to/file.o```

(Thanks to IRC user dickon for the suggestion.)

projectgus, 2017-12-05:
There is a subtle difference between running objdump to disassemble the object file, and dumping the assembly output from gcc itself. Sometimes the former produces incorrect results via disassembler bugs/limitations.

I have a technique for this which I use sometimes manually, which is to run "make V=1", look for the (long) line calling xtensa-esp32-elf-gcc on the particular source file I want, ie:

```
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DESP_PLATFORM -D IDF_VER=\"v3.0-dev-1354-g09be90bef-dirty\" -MMD -MP   -DWITH_POSIX -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I /home/gus/esp/32/idf/components/freertos/include/freertos -I /home/gus/esp/32/idf/components/freertos/include -I /home/gus/esp/32/idf/components/app_trace/include -I /home/gus/esp/32/idf/components/app_update/include -I /home/gus/esp/32/idf/components/bootloader_support/include -I /home/gus/esp/32/idf/components/bt/include -I /home/gus/esp/32/idf/components/coap/port/include -I /home/gus/esp/32/idf/components/coap/port/include/coap -I /home/gus/esp/32/idf/components/coap/libcoap/include -I /home/gus/esp/32/idf/components/coap/libcoap/include/coap -I /home/gus/esp/32/idf/components/console/. -I /home/gus/esp/32/idf/components/cxx/include -I /home/gus/esp/32/idf/components/driver/include -I /home/gus/esp/32/idf/components/esp32/include -I /home/gus/esp/32/idf/components/esp_adc_cal/include -I /home/gus/esp/32/idf/components/ethernet/include -I /home/gus/esp/32/idf/components/expat/port/include -I /home/gus/esp/32/idf/components/expat/include/expat -I /home/gus/esp/32/idf/components/fatfs/src -I /home/gus/esp/32/idf/components/heap/include -I /home/gus/esp/32/idf/components/idf_test/include -I /home/gus/esp/32/idf/components/jsmn/include/ -I /home/gus/esp/32/idf/components/json/include -I /home/gus/esp/32/idf/components/json/port/include -I /home/gus/esp/32/idf/components/libsodium/libsodium/src/libsodium/include -I /home/gus/esp/32/idf/components/libsodium/port_include -I /home/gus/esp/32/idf/components/log/include -I /home/gus/esp/32/idf/components/lwip/include/lwip -I /home/gus/esp/32/idf/components/lwip/include/lwip/port -I /home/gus/esp/32/idf/components/lwip/include/lwip/posix -I /home/gus/esp/32/idf/components/lwip/apps/ping -I /home/gus/esp/32/idf/examples/protocols/https_request/main/include -I /home/gus/esp/32/idf/components/mbedtls/port/include -I /home/gus/esp/32/idf/components/mbedtls/include -I /home/gus/esp/32/idf/components/mdns/include -I /home/gus/esp/32/idf/components/micro-ecc/micro-ecc -I /home/gus/esp/32/idf/components/newlib/platform_include -I /home/gus/esp/32/idf/components/newlib/include -I /home/gus/esp/32/idf/components/nghttp/port/include -I /home/gus/esp/32/idf/components/nghttp/nghttp2/lib/includes -I /home/gus/esp/32/idf/components/nvs_flash/include -I /home/gus/esp/32/idf/components/openssl/include -I /home/gus/esp/32/idf/components/pthread/include -I /home/gus/esp/32/idf/components/sdmmc/include -I /home/gus/esp/32/idf/components/soc/esp32/include -I /home/gus/esp/32/idf/components/soc/include -I /home/gus/esp/32/idf/components/spi_flash/include -I /home/gus/esp/32/idf/components/spiffs/include -I /home/gus/esp/32/idf/components/tcpip_adapter/include -I /home/gus/esp/32/idf/components/ulp/include -I /home/gus/esp/32/idf/components/vfs/include -I /home/gus/esp/32/idf/components/wear_levelling/include -I /home/gus/esp/32/idf/components/wpa_supplicant/include -I /home/gus/esp/32/idf/components/wpa_supplicant/port/include -I /home/gus/esp/32/idf/components/wpa_supplicant/../esp32/include -I /home/gus/esp/32/idf/components/xtensa-debug-module/include -I /home/gus/esp/32/idf/examples/protocols/https_request/build/include  -I. -c /home/gus/esp/32/idf/components/freertos/./timers.c -o timers.o
```

(Yeah, the compiler command lines are a bit ridiculously long)

It's important to find the line for the exact source file you care about, because the command line args differ between components and files. The easiest way is to update just that file's timestamp and then run `make`, so only that one file is compiled.

Copy everything on the command line except the last two args, ie `-o timer.o` and then paste that line and append either `-S` for assembler output or `-E` for C preprocessor output, instead. The assembler .s file is dropped in the working directory, IIRC.

I think the suggestion of supporting this properly in the build system is a valid one, though.

projectgus, 2020-08-19:
Since we've moved to CMake this feature is both easier and harder to implement. We don't have an easy way to add assembly output feature in the CMake side of the build system, but there is a file `build/compile_commands.json` which contains all of the source files in the project and compiler command line used for them.
For example, one entry in the hello world example for me looks like this:

```
{
  "directory": "/home/angus/esp/32/idf/examples/get-started/hello_world/build",
  "command": "/home/angus/.espressif/tools/xtensa-esp32s2-elf/esp-2020r2-8.2.0/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-gcc -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\\\"mbedtls/esp_config.h\\\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../../../../components/newlib/platform_include -I../../../../components/freertos/include -I../../../../components/freertos/xtensa/include -I../../../../components/heap/include -I../../../../components/log/include -I../../../../components/lwip/include/apps -I../../../../components/lwip/include/apps/sntp -I../../../../components/lwip/lwip/src/include -I../../../../components/lwip/port/esp32/include -I../../../../components/lwip/port/esp32/include/arch -I../../../../components/soc/src/esp32s2/. -I../../../../components/soc/src/esp32s2/include -I../../../../components/soc/include -I../../../../components/esp_rom/include -I../../../../components/esp_common/include -I../../../../components/esp_system/include -I../../../../components/xtensa/include -I../../../../components/xtensa/esp32s2/include -I../../../../components/esp32s2/include -I../../../../components/driver/include -I../../../../components/driver/esp32s2/include -I../../../../components/esp_ringbuf/include -I../../../../components/efuse/include -I../../../../components/efuse/esp32s2/include -I../../../../components/espcoredump/include -I../../../../components/esp_timer/include -I../../../../components/esp_ipc/include -I../../../../components/soc/soc/esp32s2/. -I../../../../components/soc/soc/esp32s2/include -I../../../../components/soc/soc/esp32s2/../include -I../../../../components/vfs/include -I../../../../components/esp_wifi/include -I../../../../components/esp_wifi/esp32s2/include -I../../../../components/esp_event/include -I../../../../components/esp_netif/include -I../../../../components/esp_eth/include -I../../../../components/tcpip_adapter/include -I../../../../components/app_trace/include -I../../../../components/mbedtls/port/include -I../../../../components/mbedtls/mbedtls/include -I../../../../components/mbedtls/esp_crt_bundle/include -I../../../../components/bootloader_support/include -I../../../../components/app_update/include -I../../../../components/spi_flash/include -I../../../../components/nvs_flash/include -I../../../../components/pthread/include -I../../../../components/wpa_supplicant/include -I../../../../components/wpa_supplicant/port/include -I../../../../components/wpa_supplicant/include/esp_supplicant -I../../../../components/asio/asio/asio/include -I../../../../components/asio/port/include -I../../../../components/cbor/port/include -I../../../../components/coap/port/include -I../../../../components/coap/port/include/coap -I../../../../components/coap/libcoap/include -I../../../../components/coap/libcoap/include/coap2 -I../../../../components/console -I../../../../components/nghttp/port/include -I../../../../components/nghttp/nghttp2/lib/includes -I../../../../components/esp-tls -I../../../../components/esp_gdbstub/include -I../../../../components/esp_hid/include -I../../../../components/tcp_transport/include -I../../../../components/esp_http_client/include -I../../../../components/esp_http_server/include -I../../../../components/esp_https_ota/include -I../../../../components/protobuf-c/protobuf-c -I../../../../components/protocomm/include/common -I../../../../components/protocomm/include/security -I../../../../components/protocomm/include/transports -I../../../../components/mdns/include -I../../../../components/esp_local_ctrl/include -I../../../../components/sdmmc/include -I../../../../components/esp_serial_slave_link/include -I../../../../components/esp_websocket_client/include -I../../../../components/expat/expat/expat/lib -I../../../../components/expat/port/include -I../../../../components/wear_levelling/include -I../../../../components/fatfs/diskio -I../../../../components/fatfs/vfs -I../../../../components/fatfs/src -I../../../../components/freemodbus/common/include -I../../../../components/idf_test/include -I../../../../components/idf_test/include/esp32s2 -I../../../../components/jsmn/include -I../../../../components/json/cJSON -I../../../../components/libsodium/libsodium/src/libsodium/include -I../../../../components/libsodium/port_include -I../../../../components/mqtt/esp-mqtt/include -I../../../../components/openssl/include -I../../../../components/perfmon/include -I../../../../components/spiffs/include -I../../../../components/ulp/include -I../../../../components/unity/include -I../../../../components/unity/unity/src -I../../../../components/wifi_provisioning/include  -mlongcalls   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\\\"v4.3-dev-865-gf0d53bd097-dirty\\\" -DESP_PLATFORM -o esp-idf/main/CMakeFiles/__idf_main.dir/hello [... cut ...] /32/idf/examples/get-started/hello_world/main/hello_world_main.c
```

> There is a subtle difference between running objdump to disassemble the object file, and dumping the assembly output from gcc itself. Sometimes the former produces incorrect results via disassembler bugs/limitations.

Thankfully this hasn't been true any more for a while, this bug in binutils was fixed a little while back. Objdump still occasionally gets confused dumping a full ELF file but the unlinked .o file should be OK.

closed 2020-08-19 by projectgus

jdoubleu, 2021-08-08:
Another solution, which I find is way more accessible, is to use the [`-save-temps`](https://gcc.gnu.org/onlinedocs/gcc/Developer-Options.html#Developer-Options) flag with gcc. This flag can be set per component through CMake. Add the following line to your compoenent's `CMakeLists.txt`:
```CMake
set_target_properties(${COMPONENT_LIB} PROPERTIES COMPILE_FLAGS "-save-temps=obj")
```

Don't forget to reload CMake. From now an all intermediate files (including the `.s` assembly listing) will be kept in the same directory as the `.obj` files. That is, differing based on the component path:
```
build/esp-idf/main/CMakeFiles/__idf_main.dir/main.c.s
```