espressif/esp-idf #1351: Heltec ESP32 doesn't work with ESP-IDF

chebukin, 2017-12-05:
Hi all,

I have 2 ESP32 boards: 

1. Heltec Automation ESP32 Development Board With 0.96inch OLED

2. WEMOS WiFi & Bluetooth Battery ESP32

here is my simple code:

```
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"

void app_main()
{
    printf("hello world\n");
}

```

it works as expected for the second board, but not the first one.
I have following console output for Heltec board:

```
Flashing binaries to serial port COM5 (app at offset 0x10000)...
esptool.py v2.1
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0220
Compressed 17568 bytes to 10507...
Wrote 17568 bytes (10507 compressed) at 0x00001000 in 0.9 seconds (effective 150.0 kbit/s)...
Hash of data verified.
Compressed 143088 bytes to 68405...
Wrote 143088 bytes (68405 compressed) at 0x00010000 in 6.1 seconds (effective 188.7 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 82...
Wrote 3072 bytes (82 compressed) at 0x00008000 in 0.0 seconds (effective 1521.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...
MONITOR
--- idf_monitor on COM5 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON__RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:4480
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:12988
entry 0x40078d8c

Traceback (most recent call last):
  File "C:/msys32/home/chu/esp/esp-idf/tools/idf_monitor.py", line 575, in <module>
    main()
  File "C:/msys32/home/chu/esp/esp-idf/tools/idf_monitor.py", line 504, in main
    monitor.main_loop()
  File "C:/msys32/home/chu/esp/esp-idf/tools/idf_monitor.py", line 269, in main_loop
    self.handle_serial_input(data)
  File "C:/msys32/home/chu/esp/esp-idf/tools/idf_monitor.py", line 302, in handle_serial_input
    self.console.write_bytes(b)
  File "C:/msys32/mingw32/lib/python2.7/site-packages/serial/tools/miniterm.py", line 63, in write_bytes
    self.byte_output.write(byte_string)
  File "C:/msys32/home/chu/esp/esp-idf/tools/idf_monitor.py", line 567, in write
    self.output.write(b)
IOError: [Errno 22] Invalid argument
make: *** [/home/chu/esp/esp-idf/components/esptool_py/Makefile.projbuild:96: monitor] Error 1
```
Any ideas why and how to solve it?
PS Heltec board works perfect from arduino IDE, so it's not broken.
BR, Igor

chebukin, 2017-12-05:
just found the reason... (as usually after issue described) 
printf sends data on different speed. Looks like on 74880...

negativekelvin, 2017-12-05:
That board probably uses 26mhz crystal so you need to select it in menuconfig

chebukin, 2017-12-06:
No, this is not the root. 
When I used another console at 115200 a can see loading logs and then noice, 
at 74880 I can see noice insead of loading logs and than my output. 
So it starts at 115200 and then swith to 74880 somehow. The same code for different boards works different - this is what I didn't expect. 
Looking now how to change baudrate for output...

negativekelvin, 2017-12-06:
Did you try it?

https://github.com/espressif/arduino-esp32/blob/79e5d4c40da01306bfc03aa7150440d1fcb65f41/tools/sdk/sdkconfig#L236

https://github.com/espressif/esp-idf/blob/f32fa0c1e9843565f95a632baf27df04e8b636fd/components/esp32/Kconfig#L716

igrr, 2017-12-07:
@chebukin as @negativekelvin has hinted above, you need to modify your project configuration, setting XTAL frequency from the default value of 40MHz to either "Autodetect" or 26MHz.

You can change this setting by running `make menuconfig`, navigating into Component config > ESP32-specific, and changing "Main XTAL frequency" option. This option is documented here:
http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#esp32-xtal-freq-sel

The same board works in Arduino because, unlike IDF, in Arduino "Autodetect" option is used as a default.

closed 2017-12-07 by igrr

chebukin, 2017-12-07:
Yes, that really helps. Thanks a lot!