espressif/esp-idf #1357: Failing to make blink example

jcherry4, 2017-12-06:
Hi,
I get the following error when I make the blink example:

WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/main/bootloader_start.o
/home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_start.c: In function 'bootloader_main':
/home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_start.c:458:5: error: missing braces around initializer [-Werror=missing-braces]
     bootloader_state_t bs = { 0 };
     ^
/home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_start.c:458:5: error: (near initialization for 'bs.ota_info') [-Werror=missing-braces]
/home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_start.c:458:5: warning: missing initializer for field 'factory' of 'bootloader_state_t' [-Wmissing-field-initializers]
In file included from /home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_start.c:50:0:
/home/jcherry/esp32/esp-idf/components/bootloader/subproject/main/./bootloader_config.h:35:25: note: 'factory' declared here
     esp_partition_pos_t factory;
                         ^
cc1: some warnings being treated as errors
/home/jcherry/esp32/esp-idf/make/component_wrapper.mk:242: recipe for target 'bootloader_start.o' failed
make[2]: *** [bootloader_start.o] Error 1
/home/jcherry/esp32/esp-idf/make/project.mk:448: recipe for target 'component-main-build' failed
make[1]: *** [component-main-build] Error 2
/home/jcherry/esp32/esp-idf/components/bootloader/Makefile.projbuild:40: recipe for target '/home/jcherry/esp32/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin' failed
make: *** [/home/jcherry/esp32/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin] Error 2

I am running Ubuntu 64 bit in a virtual box and have the following paths set up:

echo $PATH
/home/jcherry/bin:/home/jcherry/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/jcherry/esp32/xtensa-esp32-elf/bin

echo $IDF_PATH
/home/jcherry/esp32/esp-idf

Any help would be much appreciated.
Thanks,
John

chegewara, 2017-12-06:
There may be few reasons i can think of. Try to restart ubuntu if you fresh installed toolchain. Did you install 64-bit toolchain?

http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup.html

jcherry4, 2017-12-06:
It turns out I loaded a previous version if the xtensa tools. I tried the version in the path you sent and it worked.

Thanks so much

John

projectgus, 2017-12-07:
Glad to know you got this working, @jcherry4 !

closed 2017-12-07 by projectgus