espressif/esp-idf #1360: Crash in BTDM

rojer, 2017-12-06:
Was running a test with sending BLE notifications, when suddenly, this happens:
```
[Dec  6 23:45:19.165] Tock uptime: 2090.589886 {"free_ram":101044,"total_ram":188760} 
[Dec  6 23:45:19.172] esp32_bt_gatts_ev    CONF cid 0 st 0
[Dec  6 23:45:20.165] Tick uptime: 2091.589923 {"free_ram":101044,"total_ram":188760} 
[Dec  6 23:45:20.172] esp32_bt_gatts_ev    CONF cid 0 st 0
[Dec  6 23:45:20.257] Guru Meditation Error: Core  0 panic'ed (LoadStoreError)
[Dec  6 23:45:20.257] . Exception was unhandled.
[Dec  6 23:45:20.257] Register dump:
[Dec  6 23:45:20.257] PC      : 0x4001742d  PS      : 0x00060a33  A0      : 0x8004a7b4  A1      : 0x3ffbca40  
[Dec  6 23:45:20.258] A2      : 0x3ffbb87c  A3      : 0x4000d146  A4      : 0x3ffc0834  A5      : 0x00000000  
[Dec  6 23:45:20.279] A6      : 0x3ffb8360  A7      : 0x40013fb8  A8      : 0x80086f3c  A9      : 0x3ffbca20  
[Dec  6 23:45:20.280] A10     : 0xffffffff  A11     : 0x00000064  A12     : 0x00000069  A13     : 0x00000021  
[Dec  6 23:45:20.280] A14     : 0x00000010  A15     : 0x3ffb8360  SAR     : 0x0000001d  EXCCAUSE: 0x00000003  
[Dec  6 23:45:20.301] EXCVADDR: 0x4000d146  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
[Dec  6 23:45:20.301] 
[Dec  6 23:45:20.301] Backtrace: 0x4001742d 0x4004a7b1 0x4004b466 0x401b31c2 0x400469b6 0x40019d11 0x40055b4d 0x401b0c4b 0x401b0c9b

...
(gdb) bt
#0  0x4001742d in r_em_buf_tx_free ()
#1  0x4004a7b4 in r_lld_ral_renew_req ()
#2  0x4004b469 in r_lld_pdu_tx_flush ()
#3  0x401b31c5 in r_lld_evt_elt_delete ()
#4  0x400469b9 in r_lld_evt_delete_elt_handler ()
#5  0x40019d14 in r_ke_event_schedule ()
#6  0x40055b50 in r_rwip_schedule ()
#7  0x401b0c4e in r_rw_schedule ()
#8  0x401b0c9e in btdm_controller_task ()
```

so, crash happened soon after we got confirmation after sending a notification to the peer (one is sent every second).

xiewenxiang, 2017-12-07:
Hi rojer,
which the commit of esp-idf did you used?
In your test, do you just send notify to the peer every second?
I want to reproduce this issue.

rojer, 2017-12-07:
https://github.com/espressif/esp-idf/commit/582b731c23061d140cf6354fa7cbefad279537ff + some patches.
you can see the branch i used here: https://github.com/cesanta/esp-idf/commits/2.1-r9

yes, the application sends a notification every second - actually, it sends the exact string that is printed to stdout, so in this case `Tick uptime: 2091.589923 {"free_ram":101044,"total_ram":188760}` was sent (albeit truncated to MTU, which is 23 bytes in this case).

xiewenxiang, 2017-12-08:
I use the [branch](https://github.com/cesanta/esp-idf/commits/2.1-r9) and do a test, The ESP-32 send notify(20 Bytes) to the peer every second, the peer is iPod. Until now, it has been tested for 14 hours,and it still work. 
Are there any other operations?

rojer, 2017-12-08:
yes, unfortunately, i've only seen this once so far. there is no other acivity. maybe it was related to the other bug that i filed and that's already fixed.
let's close this one for now, i will report if i see this happen again with updated IDF.
thanks for looking into it!

closed 2017-12-08 by rojer