espressif/esp-idf #1361: Heap corruption in BT

rojer, 2017-12-06:
Seen this twice already, happens when client is connecting to GATTS

```
[Dec  7 02:22:47.586] esp32_bt_gatts_ev    CONNECT cid 0 addr a4:5e:60:b8:65:14
[Dec  7 02:22:47.594] esp32_bt_gattc_ev    CONNECT cid 0 addr a4:5e:60:b8:65:14
[Dec  7 02:22:47.603] esp32_bt_gap_ev      ADV_START_COMPLETE st 0
[Dec  7 02:22:47.609] esp32_bt_gap_ev      BLE advertising started
[Dec  7 02:22:47.615] mgos_bt_dbg_svc_ev   MTU 23
[Dec  7 02:22:47.621] mg_rpc_add_channel_i 0x3ffda034 '' GATTS
[Dec  7 02:22:47.630] mg_rpc_ev_handler    0x3ffda034 CHAN OPEN (GATTS if 4 conn 0 peer a4:5e:60:b8:65:14)
[Dec  7 02:22:48.006] led_timer_cb         Tick uptime: 4164.64, RAM: 189868, 118364 free
[Dec  7 02:22:49.006] led_timer_cb         Tock uptime: 4165.64, RAM: 189856, 118340 free
[Dec  7 02:22:50.006] led_timer_cb         Tick uptime: 4166.64, RAM: 189860, 118344 free
[Dec  7 02:22:50.606] esp32_bt_gap_ev      UPDATE_CONN_PARAMS st 16 addr a4:5e:60:b8:65:14 int 48-80 lat 0 conn_int 78 tout 400
[Dec  7 02:22:51.006] led_timer_cb         Tock uptime: 4167.64, RAM: 189856, 118340 free
[Dec  7 02:22:51.580] E (4169575) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0008
[Dec  7 02:22:51.605] E (4169595) BT: p_conn is NULL in bta_gattc_wait4_service_change_ccc_cback
[Dec  7 02:22:51.605] 
[Dec  7 02:22:51.605] CORRUPT HEAP: multi_heap.c:415 detected at 0x3ffda4f4
[Dec  7 02:22:51.605] abort() was called at PC 0x40089b4f on core 0
[Dec  7 02:22:51.605] 
[Dec  7 02:22:51.605] Backtrace: 0x4008a30f 0x4008a40f 0x40089b4f 0x40089f21 0x40082029 0x40082489 0x4000bec7 0x401a240d 0x40174a51
```

backtrace:

```
(gdb) bt
#0  0x4008a30f in invoke_abort ()
    at /opt/Espressif/esp-idf/components/esp32/./panic.c:139
#1  0x4008a412 in abort ()
    at /opt/Espressif/esp-idf/components/esp32/./panic.c:148
#2  0x40089b52 in multi_heap_assert (condition=false, line=415, 
    address=1073587444, 
    format=0x3ffc27c5 "CORRUPT HEAP: multi_heap.c:%d detected at 0x%08x\n")
    at /opt/Espressif/esp-idf/components/heap/./multi_heap_platform.h:55
#3  0x40089f24 in multi_heap_free_impl (heap=0x3ffd8508, p=0x3ffda4f8)
    at /opt/Espressif/esp-idf/components/heap/./multi_heap.c:415
#4  0x4008202c in heap_caps_free (ptr=<optimized out>)
    at /opt/Espressif/esp-idf/components/heap/./heap_caps.c:275
#5  0x4008248c in _free_r (r=0x3ffb6e4c, ptr=0x3ffda4f8)
    at /opt/Espressif/esp-idf/components/newlib/./syscalls.c:42
#6  0x4000beca in free ()
#7  0x401a2410 in bta_gattc_wait4_service_change_ccc_cback (
```

Yulong-espressif, 2017-12-07:
@rojer We have fix this bug now, please update the IDF version, thanks.

rojer, 2017-12-07:
ok, thanks!

closed 2017-12-07 by rojer