espressif/esp-idf #1363: cJSON library

chegewara, 2017-12-07:
I see there was cJSON library refactored lately. Is it bug or createDouble method is missing by purpose?

projectgus, 2017-12-07:
cJSON was updated to the latest 1.6.0 release as part of #1163. It looks like we missed that an IDF-specific CreateDouble function had been added in 5c1c6d2c32245e017646b1cb17e6da8a603dc553.

I'll check with my colleagues about the purpose of this function (compared to cJSON_CreateNumber), we can look at reinserting it and/or submitting a PR to upstream.

projectgus, 2017-12-10:
Confirmed with the original developer that cJSON_CreateNumber() is the preferred option now, so we're probably not going to re-add support for cJSON_CreateDouble().

It looks like the upstream esp32-snippets has fixed this issue by changing to CreateNumber().

I'm going to close this issue, but if anyone is reading this and experiencing unresolveable issues because this function has been removed, please reopen.

closed 2017-12-10 by projectgus