espressif/esp-idf #1364: ble connection disconnected

Sagigamil, 2017-12-07:
Hi!
I'm using the secured gat client with a few changes:
The esp32 connects to a ble gamepad and register for a hid event.
For every button click i get a uint16_t number that represents the button pressed as bitwise.
After a minute or two without clicking the gamepad, the connection is over due to:
> E (81393) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0022
Maybe I'm do not know bluetooth as well and I'm missing something need to be done. 
If anyone have a suggestion, I would be happy to know.
Thanks!

negativekelvin, 2017-12-08:
Possibly because gamepad go to sleep to save battery?

Sagigamil, 2017-12-08:
@negativekelvin I opened a new thread that read a characteristics every 10s in order to keep the gamepad alive and it seems like it's solve the problem.
Do you have another solution in order to keep the gamepad alive?

shmuelzon, 2017-12-12:
Having the exact same issue with my own peripheral. Seems to only happen on authenticated (paired) connections. If I modify the peripheral to allow unauthenticated read/writes then the connection stays alive for days. Otherwise, it disconnects after ~80 seconds. My code is based on the gatt_security_client example.
I've tried contacting [TI's forums](https://e2e.ti.com/support/wireless_connectivity/bluetooth_low_energy/f/538/p/642525/2383584) (my peripheral is based on TI's SoC) and after capturing the BLE packets, they said that the master stops transmitting leading to the supervision timeout (LSTO).

Do you have any pointers as to how I can debug/fix this issue? Where should I start looking?

Yulong-espressif, 2017-12-15:
@shmuelzon Can you show the Capture or the debug log?

shmuelzon, 2017-12-15:
@Yulong-espressif I don't have the debug log anymore and will take me a while to recreate it. The app I ran was the GATTC secure example (modified for my device name and services). It connected and paired correctly and after ~80 seconds I got a disconnected event with a 0x22 reason.
[Attached](https://github.com/espressif/esp-idf/files/1562772/sniff.zip) the BLE capture. It was taken with [TI's packet sniffer](http://www.ti.com/tool/packet-sniffer).

shmuelzon, 2018-01-09:
@Yulong-espressif Are there any updates on this issue? Anything I can do to help debug and / or fix it?

At first I thought this might be because my devices didn't make use of BT v4.2 Secure Connections because I saw the following error in the serial console:
`BT: lmp_version_below LMP version 7 < 8`
But even after adding support for the updated security profile to my peripheral I still see the same disconnection issues (though the error above now states `8 < 8`...

chegewara, 2018-01-15:
@shmuelzon This is just logging info that states:
`BT: lmp_version_below LMP version 7 < 8`

- on the left is 7 which mean peer device which is connecting to esp32 is BLE v4.1 device
- 8 means that esp32 is BLE v4.2 device
Thats all, nothing more and nothing less

shmuelzon, 2018-01-15:
@chegewara, right. But since BT 4.2 introduced LE Secure Connections and my issue only surfaces on secured connections, I though it might be related. Alas, it isn't...

chegewara, 2018-01-15:
Ok, just wanted to give you some info.

Yulong-espressif, 2018-01-15:
@shmuelzon What is your issues? It is means that just secure connection will cause the problem, yet?

chegewara, 2018-01-15:
@Yulong-espressif  im strugling right now with similar situation. When i connect with secure connection esp32 with my android phone all is ok. But when i reset esp32 and esp32 is connecting with stored keys (bonded devices) my phone is disconnecting after some time (25 to 90 seconds i think).
ATM i dont know what is the cause, will do some more test. 

- esp32 app is my own app based on Neil's Kolban library,
- android phone with BLEv4.0 and nRF connect to tests
- error: GATT CONN TIMEOUT (logged in nRF connect)

shmuelzon, 2018-01-15:
@Yulong-espressif as I described above, whenever I try to make a secured connection (via pin code pairing) my connections are dropped after ~80 seconds with a disconnection reason of 0x22. My peripherals are based on TI's CC2640 SoC. According to a BT dump and TI's forum, the master stops (i.e. ESP32) transmitting leading to a supervision timeout.

When my peripheral is modified to work without pairing, the connections last for days.
If it helps, [this](https://github.com/shmuelzon/esp32-ble2mqtt) is my app.

chegewara, 2018-01-15:
I didnt notice this before, but from this log we can see that some peer device keys are exchanged after writting keys to nvs. This is long shot and maybe those keys arent important and dont need to be stored. Stored is only ESP_LE_KEY_PENC, and other two (ESP_LE_KEY_PID and ESP_LE_KEY_PCSRK) arent not (at least we can read it from logs):

```
D (7484) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7484) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7484) BLEDevice: key type = ESP_LE_KEY_LENC
D (7489) BLEServer: BLEServer ... handling GAP event!
D (7494) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7500) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7505) BLEDevice: key type = ESP_LE_KEY_LID
D (7510) BLEServer: BLEServer ... handling GAP event!
D (7516) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7521) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7527) BLEDevice: key type = ESP_LE_KEY_LCSRK
D (7532) BLEServer: BLEServer ... handling GAP event!
D (7775) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7775) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7775) BLEDevice: key type = ESP_LE_KEY_PENC
D (7780) BLEServer: BLEServer ... handling GAP event!
D (7824) nvs: nvs_open_from_partition bt_config.conf 1
D (7824) nvs: nvs_set_blob bt_cfg_key 250
D (7827) nvs: nvs_close 4
D (7827) nvs: nvs_open_from_partition bt_config.conf 1
D (7830) nvs: nvs_set_blob bt_cfg_key 333
D (7836) nvs: nvs_close 5
D (7837) nvs: nvs_open_from_partition bt_config.conf 1
D (7842) nvs: nvs_set_blob bt_cfg_key 393
D (7849) nvs: nvs_close 6
D (7849) nvs: nvs_open_from_partition bt_config.conf 1
D (7854) nvs: nvs_set_blob bt_cfg_key 457
D (7861) nvs: nvs_close 7
D (7861) nvs: nvs_open_from_partition bt_config.conf 1
D (7866) nvs: nvs_set_blob bt_cfg_key 512
D (7874) nvs: nvs_close 8
D (7874) nvs: nvs_open_from_partition bt_config.conf 1
D (7878) nvs: nvs_set_blob bt_cfg_key 576
D (7885) nvs: nvs_close 9
D (7885) nvs: nvs_open_from_partition bt_config.conf 1
D (7890) nvs: nvs_set_blob bt_cfg_key 590
D (7897) nvs: nvs_close 10
D (7898) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7903) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7908) BLEDevice: key type = ESP_LE_KEY_PID
D (7913) BLEServer: BLEServer ... handling GAP event!
D (7918) BLEUtils: Received a GAP event: ESP_GAP_BLE_KEY_EVT
D (7924) BLEUtils: *** dumpGapEvent: Logger not coded ***
I (7929) BLEDevice: key type = ESP_LE_KEY_PCSRK

```

Yulong-espressif, 2018-01-15:
@shmuelzon I think this is because the esp32 device is set to the default 4.0 device, resulting in the use of secure connections can not be encrypted.

shmuelzon, 2018-01-15:
@Yulong-espressif Really? There are references in the code to v4.2 and secure connections. Do you mean that the firmware the stack communicates with is set to 4.0? In theory interoperability shouldn't be an issue as both devices should revert to what they both support.
Do you think there is a mismatch between the stack configuration and Bluetooth firmware?

chegewara, 2018-01-15:
This is log when i just disconnect and re-connect:
```
E (2823193) BT: lmp_version_below LMP version 6 < 8
D (2823510) nvs: nvs_open_from_partition bt_config.conf 1
D (2823511) nvs: nvs_set_blob bt_cfg_key 590
D (2823514) nvs: nvs_close 11
D (2823515) nvs: nvs_open_from_partition bt_config.conf 1
D (2823518) nvs: nvs_set_blob bt_cfg_key 590
D (2823525) nvs: nvs_close 12
D (2823526) nvs: nvs_open_from_partition bt_config.conf 1
D (2823531) nvs: nvs_set_blob bt_cfg_key 590
D (2823539) nvs: nvs_close 13
D (2823539) nvs: nvs_open_from_partition bt_config.conf 1
D (2823544) nvs: nvs_set_blob bt_cfg_key 590
D (2823551) nvs: nvs_close 14
D (2823551) nvs: nvs_open_from_partition bt_config.conf 1
D (2823556) nvs: nvs_set_blob bt_cfg_key 590
D (2823564) nvs: nvs_close 15
D (2823564) nvs: nvs_open_from_partition bt_config.conf 1
D (2823569) nvs: nvs_set_blob bt_cfg_key 590
D (2823577) nvs: nvs_close 16
D (2823577) BLEUtils: Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
```
As we can see we have reading operation from nvs. And this is re-connect log after hard reseting esp32:
```
E (13356) BT: lmp_version_below LMP version 6 < 8
D (13675) BLEUtils: Received a GAP event: ESP_GAP_BLE_AUTH_CMPL_EVT
```
There are no nvs operations.
Another difference is two versus one ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT event:
```
D (2823602) BLEServer: BLEServer ... handling GAP event!
D (2825199) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (2825200) BLEUtils: [status: 16, bd_addr: 78:1f:db:c0:c7:f3, min_int: 6, max_int: 6, latency: 0, conn_int: 39, timeout: 2000]
D (2825207) BLEServer: BLEServer ... handling GAP event!
D (2826292) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (2826292) BLEUtils: [status: 0, bd_addr: 78:1f:db:c0:c7:f3, min_int: 0, max_int: 0, latency: 0, conn_int: 6, timeout: 600]
```
and connection after hard reset esp32 and auto disconnecting because GATT CONN TIMEOUT:
```
D (264477) BLEUtils: Received a GAP event: ESP_GAP_BLE_UPDATE_CONN_PARAMS_EVT
D (264477) BLEUtils: [status: 16, bd_addr: 78:1f:db:c0:c7:f3, min_int: 6, max_int: 6, latency: 0, conn_int: 39, timeout: 2000]
```

Yulong-espressif, 2018-01-16:
@shmuelzon It is because the TI boad act as a master but didn't exchange the version to the ESP32. The ESP32 will think of each other as a 4.0 device, resulting in a failure to pair using a secure connection.

Yulong-espressif, 2018-01-16:
@shmuelzon So, can you sent the version exchange by the application layer in the TI board?

shmuelzon, 2018-01-16:
@Yulong-espressif I’m not sure I follow. In my usecase the ESP32 is the master and the TI board is the slave/peripheral. Furthermore, it does get some indication of the TI BLE version since according to the log it detected it as a 4.1 device and then as a 4.2 device when I enabled LE Secured Connections. So it should be aware of what’s supported.
The pairing process itself does work. I can read the relevant characteristics only after pairing and I get notifications on. The only problem is that the connection is dropped shortly after.

Can you please provide some reference to the version exchange message you believe is missing? I didn’t find something specific on that.

pa7lim, 2018-01-19:
Same issue here. I am connecting to a HM-17. 
I sent every 100ms three bytes.  After a few seconds it stops with this message.
bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0022

shmuelzon, 2018-01-21:
@pa7lim Are you using a secure connection?

pa7lim, 2018-01-22:
@shmuelzon Not that I know. Can I check that somewhere?

Yulong-espressif, 2018-01-22:
@shmuelzon So means that your TI board support 4.1 instead of 4.2, yeah? For the 4.1 device, It can't used secure connection, so I don't think it have any problem in this case, thanks.

shmuelzon, 2018-01-22:
@pa7lim It really depends on your application and BLE peripheral. If you didn't need to do anything special, you're probably not pairing

@Yulong-espressif My board can support both and I've tried both. I've initially used 4.1 and then added support for 4.2 Secure Connections in the hopes it would help.
In both cases it gets disconnected after ~80 seconds. Presumably since the master (ESP32) stopped transmitting. I noticed that there is a binary library in the BT component. Is there anything I can do (enable some logs, produce more BLE captures) that might help getting to the bottom of this issue?

Yulong-espressif, 2018-01-22:
@shmuelzon First, When your TI board exchange the version with 4.1 to the ESP32 then it shouldn't used secure connection feature, even if it support the secure connection feature. 
Second, you said that 80 seconds will disconnect, so what is the error code with the disconnect reason? Does the application have received the ESP_GAP_BLE_AUTH_CMPL_EVT event?

shmuelzon, 2018-01-22:
@Yulong-espressif 4.1 also supported pairing with a pin code and MITM protection but the key exchange process was less secure than the new standard with 4.2.
The pairing process itself works as expected and authentication is completed successfully and I receive the ESP_GAP_BLE_AUTH_CMPL_EVT event. After pairing, I can successfully read the protected characteristics so it seems to work as expected.
When the peripheral is connected, I can read it get notifications. After ~80 seconds I receive a disconnection event and the reason is 0x22.

shmuelzon, 2018-01-22:
@Yulong-espressif Thanks! I've updated my IDF and it seems to be working a lot better. The ESP is connected to my peripheral for an hour already without any issues.

I don't know if @Sagigamil original problem was also solved, but as far as I'm concerned, this issue can be closed.

closed 2018-03-22 by FayeY