espressif/esp-idf #1365: esp_ble_remove_bond_device doesn't remove a device if it is connected

rojer, 2017-12-08:
connection is dropped, but device remains on the bonded list. `ESP_GAP_BLE_REMOVE_BOND_DEV_COMPLETE_EVT` is raised with status 1.

the call should either have no effect or (preferably) disconnect and remove the device.

TianHao-Espressif, 2017-12-13:
@rojer, bonded device have no certain relationship to connection. When A reconnect to B, if the B  have bond A, then it will use the bonded information to connect with SMP. If B have removed A's boned information, then A and B will do a clean connect.

rojer, 2017-12-13:
@TianHao-Espressif yeah, i get it. but if the device happens to be connected at the time when `esp_ble_remove_bond_device` is called, it doesn't work properly: connection is dropped and device is not removed.
either of the following would be acceptable:
* close connection _and_ remove
* remove but leave connection intact (as you say, connection and bonding are not really related)

Weijian-Espressif, 2018-01-08:
@rojer thanks for your reminder, it is really a bug, we have fixed the bug now, it will merge the latest master as soon as possible.

rojer, 2018-01-08:
@Weijian-Espressif cool. so which is it now: close and remove or remove but leave open?

FayeY, 2018-01-11:
Hi, this issue should have been solved in the latest IDF, please have a try. If it is not, please feel free to reopen it. Thanks.

closed 2018-01-11 by FayeY