espressif/esp-idf #1368: BLE Whitelisting not working properly

bmetman, 2017-12-09:
The esp_ble_gap_update_whitelist function in the esp_gap_ble.h library does not add Bluetooth MAC address properly. Combined with advertising BLE parameters such as the adv_filter_policy = ADV_FILTER_ALLOW_SCAN_WLST_CON_WLST the devices are still shown, and can still be connected to. 

Also, the esp_ble_gap_get_whitelist_size function always returns 11, even if no addresses have been added to the whitelist. Why is this?

Finally, it would be nice to have functionality to clear the BLE whitelist instead of manually removing separate BLE addresses.

bmetman, 2017-12-15:
Do you know when you have time to look at this issue?

Yulong-espressif, 2017-12-15:
@bmetman What is your commit ID? The esp_ble_gap_get_whitelist_size function is indicate the remaining size. Thanks.

bmetman, 2017-12-15:
Ah thank you, that helps. I don't know what my commit ID is, I only logged this issue with id: [TW#16937]

Yulong-espressif, 2017-12-15:
@bmetman So can you update your IDF version? I think we have fix this issues in the new version of SDK.

bmetman, 2017-12-18:
I have cloned the latest IDF and pushed it to our branch. It still won't work for me. When I add a MAC address to my whitelist, change my advertising filter policy to only advertise to whitelisted device, and start advertising, the ESP32 chip is visible to either all devices or no devices at all. Do you have any idea what causes this behavior?

Weijian-Espressif, 2018-02-12:
[sacan_whitelist.c.zip](https://github.com/espressif/esp-idf/files/1715518/sacan_whitelist.c.zip)
@bmetman There is a demo that only scan adv in  white list. it works well.  you can change whitelist_addr at line 59 and try. if any question, let me know.

closed 2018-03-15 by FayeY

stmariej, 2020-08-17:
This demo is not working.

I (473) heap_init: At 3FFC9080 len 00016F80 (91 KiB): DRAM
I (479) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (486) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (492) heap_init: At 40092E04 len 0000D1FC (52 KiB): IRAM
I (498) cpu_start: Pro cpu start user code
I (516) spi_flash: detected chip: generic
I (517) spi_flash: flash io: dio
W (517) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (528) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (614) BTDM_INIT: BT controller compile version [c1cbe45]
I (614) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (704) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0
I (1074) GATTC_DEMO: REG_EVT
I (1094) GATTC_DEMO: scan start success