espressif/esp-idf #1369: fatfs stat() times can be off by an hour

lllucius, 2017-12-10:
Because the "tm" struct isn't completely cleared, the tm_isdst field can have random values.  This can cause mktime() to adjust the time incorrectly depending on the passed in value (see mktime man page).

The attached patch fixed the problem.

[vfs_fat_stat.diff.gz](https://github.com/espressif/esp-idf/files/1545416/vfs_fat_stat.diff.gz)

lllucius, 2017-12-10:
While you're at it, you might consider converting the times to UTC(GMT) since fat stores them in local time.

Updated patch attached:

[vfs_fat_stat_with_gmtime.diff.gz](https://github.com/espressif/esp-idf/files/1545452/vfs_fat_stat_with_gmtime.diff.gz)

igrr, 2017-12-10:
Hi @lllucius, thanks for reporting this. We happen to have a fix for this currently in review, should be merged in the next few days.

closed 2017-12-13 by igrr