espressif/esp-idf #1373: ESP32 SPI slave receive interrupt

dipankarpanda, 2017-12-11:
Hello,
I am trying to connect ESP32 as a SPI Slave mode with a sensor (the senor is not just a sensor, it is a sensor device, equipped microcontroller in it, which is the Master). 

**I want to enable the receive interrupt of SPI slave**, so that, whenever any byte I receive from Sensor control goes to receive ISR and do some task.
****
 In the driver provided by esp-idf, I couldn't find the interrupt enable option and also interrupt handler/ISR. 

So, kindly help me in finding, how I can enable the receive interrupt for SPI communication while it is in slave more. Any information will be highly appreciated.

Thanks in advance
Dipankar Panda

AmandaYaoYao, 2017-12-13:
the structure "spi_slave_interface_config_t" has two member ".post_setup_cb" and ".post_trans_cb". theis can help I think

dipankarpanda, 2017-12-13:
Hi, @AmandaYaoYao thanks for the reply. 

Ya I have seen that, but the problem with these 2 call back function is, we have to first call the function `spi_slave_transmit(HSPI_HOST, &t, portMAX_DELAY); `first and then if anything it receives in rx_buffer, that function gets called automatically. so whenever we want to receive something we have to call the function `spi_slave_transmit(HSPI_HOST, &t, portMAX_DELAY)` and then when it receives it goes to the call back function. 
But this is not the interrupt functionality right? if SPI receive interrupt is enabled, we don't need to call any function to receive or send something, automatically it receives and whenever it receives it goes to ISR automatically. I just wanted this functionality. Is this possible in ESP32 using esp-idf driver?

Thanks
Dipankar

ginkgm, 2017-12-14:
@dipankarpanda 
The process in a spi slave transaction is:
1. call the ``queue_trans`` to push data of a transaction into ``trans_queue``
2. enable the interrupt, so that the ISR is invoked
3.
    1. after that ``queue_trans`` returns, you can do whatever you want without waiting for the transaction.
    1. **(in ISR)** retrieve the transaction from the queue
4. **(in ISR)** configure data to send in the ISR, and enable new transaction, so that the slave will respond to the master's communication.
5. ``post_setup_cb`` is called.
6. **(in ISR)** transaction is done on the bus
7. **(in ISR)** the ISR is invoked by the hardware, retrieves data, ``post_trans_cb`` is called.
8. **(in ISR)** push the transaction back to the ``ret_queue``
7. when you called ``get_trans_result``, it tries to retrieve a transaction from ``ret_queue``, this holds the data.

NOTE: 
1. if you don't call ``get_trans_result``, you can send without waiting it to be sent; or set the ``ticks_to_wait`` to 0 perform a polling to the receiving state.
2. if you don't call ``queue_trans``, you cannot get the received data, because the hardware is not ready (you have to tell the hardware you have already dealt with the old data and prepared for the new transactions).
3. ``transmit`` is a wrap of ``queue_trans`` and ``get_trans_result``, if you don't want to wait, just call ``queue_trans``.

In your case, you can use ``queue_trans`` to send a lot of transactions (maybe without any data) into the ``trans_queue``, and make sure to queue new transactions in time (before the queue is run out). Then the received data will be send back to the ``ret_queue`` automatically by the ISR. Just poll the ``get_trans_result`` to see whether new transactions are done.
If you want a function to be called, you can use the ``post_trans_cb``, but please note that this is called in the ISR, you can not loop, wait, or do any big stuffs (maybe including queue new trans). 

Sorry that the IDF now do not support event loop open to users (event handlers are called outside ISR, we are thinking about this). But you can do this yourself: 

1. initialize the SPI slave, and maybe queue the very begining transaction, so that the hardware will work.
2. create a new task
3. inside the task, ``queue_trans`` without waiting to make sure the hardware is always working, and ``get_trans_result`` without waiting, and call handlers if a new transaction is returned.

The task works just like a event loop, and will call the handler at safe position.

dipankarpanda, 2017-12-15:
Dear @ginkgm, Thanks for the detailed information. But few things are unclear with respect to my case,

What I want is simple --

1. Configure SPI in slave mode with interrupt enable 
2. Whenever master sends clock to salve, slave will send a data to master`(0xAB)`
3. Then Master will also send a data`(0xAB)` to slave.
4. Whenever slaves receives this, in slave's ISR first it checks data (whether received data is `0xAB` or not), if matches it will send "**0**" to master.
5. upon reception master will send it's data (_1 byte_).
6. upon reception Slave will send "**1**" to master
7. upon reception master again will send it's data(_1 byte_).

and this process continues ...

So, from slaves ISR itself, I need to trigger the next data to be transmitted. and every-time I need to  do the same whenever slave receives something from master.

I want make this process flow. is this possible in esp32 to implement?
I have implemented this process in Renesas micro controllers simply using rx interrupt. But here getting difficulty in managing the process flow. 

Thanks
Dipankar

ginkgm, 2017-12-15:
@dipankarpanda do you mean in the same CS?

dipankarpanda, 2017-12-18:
@ginkgm sorry for the delay in reply, coz I was confirming with the Sensor(master)  vendor regarding the functionality.

In my case, **master doesn't provide any chip select pin**, so slave needs to put its chip select line always low to receive signals from the master. Also slave needs to make its `rx `interrupt enable, so that it can receive the signal from master, otherwise master is not going to send any separate signal that it is going to send data now.

Only the way to receive data is, to enable the `rx `interrupt (According to manufacturer). Also from `rx `ISR itself we need to transmit data to master also for synchronization (like I said in my previous comment). So, this implementation I need to do.. Kindly suggest..

Thanks
Dipankar

ginkgm, 2017-12-19:
@dipankarpanda 
Sorry, but our hardware is designed to start and end a transaction by the CS line, rather than in unit of bytes. So we're not going to provide such driver probably.

Here're some tricks to help you to do this:
1. controll the CS line by yourself (the slave, connect a gpio to the CS line internally or by a jump wire)
2. send transactions with length of 1 byte.
3. pull low the CS at the beginning
4. when the master sends one byte, the slave is likely to get trans_done, ``SPI_OUT_EOF_INT`` or ``SPI_IN_SUC_EOF_INT`` interrupts (we haven't use later two interrupts in the IDF implementation).
5. pull high the CS to reset the slave hardware, and send a new transaction. and pull low CS again.

note: you have to control the speed not to send too fast (no sending when CS is inactive).

WolfWalter, 2018-01-09:
I also needed to communicate with a master (mode 3) without cs-line. With the tips from @ginkgm I was able to get it working. For my application I first needed set he transaction length to 6 bytes ( t.length = 6*8; ) to receive the master command. In the post_setup_cb callback I set my fake-CS line low and in the post_trans_cb callback I set the fake-CS line high. After the transaction I set the length depending on the received command.
To get In sync with the SPI-clk I also have a soft-timer that toggles the CS-line after a certain time of inactivity of the SPI-clk line.

ginkgm, 2018-01-10:
To get interrupted, you may have to work byte-by-byte...

closed 2018-01-26 by FayeY

aasick, 2019-12-18:
I'm not using idf.
To receive slave data in esp32 (MISO), how to program slave interrupt method using Arduino IDE