espressif/esp-idf #1377: WiFi auto connect at boot time doesn't seem to work

huibo-shi, 2017-12-12:
My goal is to configure once wifi SSID and PASSWORD, then ESP32 can automatically connect to target access point after start up. 

I set first esp wifi storage type to save settings into flash by `esp_wifi_set_storage(WIFI_STORAGE_FLASH)`  , then activate auto connect by `esp_wifi_set_auto_connect(true)`. Finally configure settings by `esp_wifi_set_config()`. It can connect correctly to the target AP when testing my settings. However after reboot, the auto connect is not triggered! 

I searched in ESP32 forum and people guide me to have a look at kolban-ESP32 (https://leanpub.com/kolban-ESP32), but in his book, he exactly use the same idea (To see the section WiFi at boot time). But from my test,  it doesn't work. 

Additionally I understood this concept from ESP8266 and it works very well for ESP8266 with the same approach. So I'm thinking there may be something missing in SDK. Could you have a look at this issue?

krzychb, 2017-12-12:
Hi @huibo-shi,
Could you post a code snippet (using e.g. [gist](https://gist.github.com/)) to reproduce this issue? 
This will make it easier to check it out.

chegewara, 2017-12-12:
@huibo-shi You can check this code, it should work
https://github.com/nkolban/esp32-snippets/tree/master/networking/bootwifi

FayeY, 2017-12-14:
Hi @huibo-shi , did the above suggestion help you resolve the issue?

huibo-shi, 2017-12-19:
Hi @krzychb Thanks for your reply. I will try to do it as soon as possible.

huibo-shi, 2017-12-19:
Hi @chegewara and @FayeY  Thanks for the suggestion. The bootwifi example stores wifi ssid and password into flash and call connect with this credentials to wifi manually after start up. This is a solution I also used to temporarily make my project moving on. But it doesn't use the SDK APIs to solve this scenario. 
If I understand correctly, combination usage of `esp_wifi_set_storage(WIFI_STORAGE_FLASH)` `esp_wifi_set_auto_connect(true)` and `esp_wifi_set_config()` should do the trick. Those APIs should be designed to do a such mission. What I wanted is to connect to wifi AP automatically after boot by using SDK APIs.

In fact after these days' tests, I finally find out a solution by using SDK APIs. When initialize the main program after boot, I call first `esp_wifi_get_auto_connect` to check whether auto connect is activated, if yes, then call `esp_wifi_get_config` to retrieve wifi configuration data. If retrieved SSID and password are not NULL. then I enable station mode and call `esp_wifi_start` and `esp_wifi_connect`. 
I think maybe you can improve bootloader to automatize this process.

zhangyanjiaoesp, 2018-01-16:
@huibo-shi  Thanks for your suggestion. We will discuss this problem and try to automatize this process

heyinling, 2018-01-16:
@huibo-shi adding the following code during startup should meet your requirement: 

```c
    bool auto_connect;
    esp_wifi_start();
    esp_wifi_get_auto_connect(&auto_connect);
    if (auto_connect == true){
        esp_wifi_connect();
    }
```

I think the current design is reasonable. User might want to define their own connect function, for example, do scan and connect to the AP with best RSSI, or prepare certificate for WPA2 enterprise AP (not sure about this). If change to auto connect directly, then user can't do customization on this part.

liuzfesp, 2018-02-23:
@huibo-shi  currently esp_wifi_set_auto_connect has no effect to the connection, we will mark this API s obsolete API. Please following @heyinling 's suggestion to auto connect.

FayeY, 2018-02-23:
Closing due to lack of response, please feel free to reopen if this is still an issue.

closed 2018-02-23 by FayeY

my-abousamra, 2018-05-13:
@liuzfesp what is the alternative solution to make auto connect as esp_wifi_set_auto_connect is deprecated and @heyinling solution is based on esp_wifi_get_auto_connect which is deprecated as well?

liuzfesp, 2018-05-14:
HI @my-abousamra, call esp_wifi_connect after received event SYSTEM_EVENT_STA_DISCONNECTED

my-abousamra, 2018-05-14:
@liuzfesp I mean after start up, how to make a wifi connection based on configuration stored in flash using esp_wifi_set_storage(WIFI_STORAGE_FLASH) where esp_wifi_get_auto_connect is deprecated now

liuzfesp, 2018-05-16:
@my-abousamra , we mark esp_wifi_get/set_auto_connect as obsolete because they lead to some misunderstanding, most of the users think they don't need to call esp_wifi_connect once esp_wifi_set_auto_connect(true) after the system startup, but actually the esp_wifi_set_auto_connect() only set the auto_connect flag, that's why we mark it as obsolete. If the application want to store the auto connect flag into flash, they need to handle it themselves.

liuzfesp, 2018-05-16:
HI @my-abousamra, do you think removing esp_wifi_set_auto_connect makes application programming more hard?

Reiner1210, 2018-05-16:
I use the function to switch off auto connect! At the moment I get a warning during compilation...what should I use instead?

Reiner1210, 2018-05-16:
By the way: I think you should remove the auto connection stuff. I do not see any reason for it. If auto connect is requured it can be done with some lines of code in app_main

liuzfesp, 2018-05-16:
@Reiner1210 yes, I think so. For the documents, I will create another MR to update it.

my-abousamra, 2018-05-16:
Hi @liuzfesp, For easy application programming esp-idf should have a clear and easy way to auto connect to wifi after start up based on saved credentials (I'm not talking about reconnect to wifi if the connection lost as it easy to do) so if you removed esp_wifi_set_auto_connect you should give an alternative for it and I don't find it. If there is an alternative function or method to it please inform us. Updating documents is a great idea.

my-abousamra, 2018-05-16:
@Reiner1210 If you have a way to auto connect after start up based on saved credentials please inform me

Reiner1210, 2018-05-16:
@my-abousamra It's so easy !!!
The user gives SSIP / password via terminal / Webform or whatever you have designed .... values are stored in nvs Flash. in app_main I read this values - and if not empty I use to connect - voila .....autoconnect on boot

my-abousamra, 2018-05-16:
@Reiner1210 I was thinking a way using wifi functions (like the esp_wifi_set_auto_connect, .. etc), Any way thank you!

chschnell, 2019-04-18:
It appears that auto-connect is active by default, and the function to deactivate it has been marked deprecated. That's actually quite a good practical joke, because I actually don't want it, and now ended up with a deprecated warning that I cannot even patch away because it sits in libnet80211.a. So I kindly ask to change the default auto-connect state to inactive. Thanks!