espressif/esp-idf #1379: DTLS implemented?

lancegerday, 2017-12-13:
After using make menuconfig and component config->mbedTLS->support DTLS protocol (all versions).

[image]

I see no way to use DTLS with CoAP using the example client-server code no additional .h files to be included appear.

My final goal is to implement simple a client/server using CoAP with DTLS in PreSharedKey mode which means using cipher-suite AEAD_AES_128_CCM_8 which should be able to use the built-in AES hardware acceleration.

negativekelvin, 2017-12-13:
The libcoap submodule is not using the dtls branch

lancegerday, 2017-12-13:
But then how can I use DTLS with CoAP for esp32? DTLS is part of the CoAP standard as stated in RFC 7252.

projectgus, 2017-12-14:
ESP-IDF has network protocol level DTLS support via mbedTLS. However the version of upstream libcoap we ship does not include DTLS support.

It appears that DTLS support has since been added to the main branch of libcoap and there was a pre-4.1.2 release that uses a library called "tinydtls". There appears to also be openssl DTLS support, but no mbedTLS support yet.

I agree that it's desirable for us to support CoAP over DTLS, as soon as possible.

negativekelvin, 2018-04-08:
Any progress?

JcBernack, 2018-10-15:
Bump. We will need this feature and would really appreciate some advice on how to get this going - or an estimate on when we can expect progress on this issue.

daveismith, 2018-10-27:
@JcBernack / @projectgus - it seems like there's support for OpenSSL in the upstream project now. Is that sufficient, or does there need to be support for mbedtls?

negativekelvin, 2018-10-27:
@daveismith no because esp-idf openssl shim does not support required functions

H-LK, 2018-10-27:
I would also like to have this function soon.

TorstenL, 2019-02-03:
Bump - It would be really great to have DTLS support

cgawron, 2019-04-08:
According to the [Changelog](https://github.com/obgm/libcoap/blob/develop/ChangeLog), libcoap 4.2.0 supports DTLS. Has anybody successfully tried it?

daveismith, 2019-04-08:
I've used it with TinyDTLS on the desktop. I'd prefer an mbed implementation as TinyDTLS doesn't support x509 certificates and says they don't want to (even though the patches aren't too significant to make it do so)

jitin17, 2019-04-08:
@cgawron I have used TinyDTLS with `libcoap` and it works as expected, and support for DTLS in esp-idf's `libcoap` is in the pipeline. For your reference https://github.com/obgm/libcoap/issues/313#issuecomment-477050314
@daveismith It is true that TinyDTLS doesn't support PKI. But if pre-shared keys suffice to the need then one can use TinyDTLS.

daveismith, 2019-04-08:
@jitin17 - agreed that PSK or RPK may be sufficient for some things, but IoT standards are starting to require PKI. An example would be dotdot, which requires the use of PKI with operational certificates to play on the application layer. It's important to have this capability in devices like the ESP for developers.

jitin17, 2019-06-18:
@JcBernack @lancegerday @cgawron @daveismith We have worked on adding DTLS support in libcoap with the help of mbedtls, which will help us support both PSK and PKI based authentication. Please take a look at this PR: https://github.com/obgm/libcoap/pull/353
We will try to add it to ESP-IDF as soon as possible.

closed 2019-08-21 by igrr