espressif/esp-idf #1380: example_ble_sec_gatts_demo typo

chegewara, 2017-12-13:
There is typo in advertising uuid (most significant octets):
https://github.com/espressif/esp-idf/blob/bd6ee752fcf074f2fc681c001a329300c8a180d1/examples/bluetooth/gatt_security_server/main/example_ble_sec_gatts_demo.c#L60
should be:
`0xfb, 0x34, 0x9b, 0x5f, 0x80, 0x00, 0x00, 0x80, 0x00, 0x10, 0x00, 0x00, 0x0D, 0x18, 0x00, 0x00,`

PS i have question, why in this example service uuid found by nRF connect (not the one advertised) is 0x1809. I cant even find where it is setup. Is it issue with encoding or is it me?

Yulong-espressif, 2017-12-15:
@chegewara，The uuid in the ADV will only be carried in the broadcast data packet, and connect, the client found service UUID is in the service attribute database, so the corresponding value is ESP_GATT_UUID_HEALTH_THERMOM_SVC (0x1809)

chegewara, 2017-12-15:
Thanks, i have found it already

closed 2017-12-19 by FayeY